synthetos/g2 #230: Active low step pins

executivul, 2017-01-13:
I'm trying to use g2core on an Arduino Due with level converters and externals TB6560 drivers which require active low step signals. How can that be accomplished?
Thank you!

aldenhart, 2017-01-13:
This is on the immediate roadmap. I expect sometime in the next few weeks to edge. Alternately, an external inverter can be used.

executivul, 2017-01-14:
Thanks. 
I've tried changing false->true in motate step pin defs for due and recompiling everything but didn't work.
I shall wait for the changes frome someone more skilled than me and run on grbl until then.
I'm already using txb0108 level converters I've tried some mosfets (irlz44n) which I had on hand but the signal was bad on the scope and I gave up, adding more circuitry is prone to missing steps and problems I'm affraid.

giseburt, 2017-01-14:
As a quick fix, to reverse the step pin (only) for ALL motors, you can do
the following:

In g2core/device/step_dir_driver/step_dir_driver.h line 55: (
https://github.com/synthetos/g2/blob/edge/g2core/device/step_dir_driver/step_dir_driver.h#L55
)

Change:
OutputPin<step_num> _step;

To:
OutputPin<step_num> _step{kStartHigh};

That will make the step pins high at boot up.

Now to make the pins invert, at lines 128-130 of the same file (
https://github.com/synthetos/g2/blob/edge/g2core/device/step_dir_driver/step_dir_driver.h#L128-L130
) invert set() which sets the pin high and clear() which sets it low:

So, this:
void stepStart() override { _step.set(); };

void stepEnd() override { _step.clear(); };

Becomes:
void stepStart() override { _step.clear(); };

void stepEnd() override { _step.set(); };

With those changes, all step pins will start HIGH, and then will be pulsed
LOW and returned to HIGH in order to step the motor.

Hopefully that helps.

-Rob

executivul, 2017-01-14:
Thank you!
That did the trick!

grbd, 2017-01-29:
Just to add to this
would it be possible to add in a feature to invert the enable pin for the motor at some point?
the RAPS128 Pololu driver has an inverted enable pin for activation
I can already work around this by hacking the source code but it'd be nice if it was something that could be set within the settings

Also I should be doing a pull request in a short while for the RADDS board from ooznest