synthetos/g2 #232: Zmin/Probe always triggers a Shutdown State

executivul, 2017-01-20:
I have a machine with no limit switches but with touchplate so I've configured all limit inputs as:

// Xmin on v9 board
#define DI1_MODE                    IO_MODE_DISABLED
#define DI1_ACTION                  INPUT_ACTION_NONE
#define DI1_FUNCTION                INPUT_FUNCTION_NONE

// Xmax
#define DI2_MODE                    IO_MODE_DISABLED
#define DI2_ACTION                  INPUT_ACTION_NONE
#define DI2_FUNCTION                INPUT_FUNCTION_NONE

// Ymin
#define DI3_MODE                    IO_MODE_DISABLED
#define DI3_ACTION                  INPUT_ACTION_NONE
#define DI3_FUNCTION                INPUT_FUNCTION_NONE

// Ymax
#define DI4_MODE                    IO_MODE_DISABLED
#define DI4_ACTION                  INPUT_ACTION_NONE
#define DI4_FUNCTION                INPUT_FUNCTION_NONE

// Zmin
#define DI5_MODE                    IO_ACTIVE_LOW  // Z probe
#define DI5_ACTION                  INPUT_ACTION_NONE
#define DI5_FUNCTION                INPUT_FUNCTION_NONE

// Zmax
#define DI6_MODE                    IO_MODE_DISABLED
#define DI6_ACTION                  INPUT_ACTION_STOP
#define DI6_FUNCTION                INPUT_FUNCTION_NONE

// Amin
#define DI7_MODE                    IO_MODE_DISABLED
#define DI7_ACTION                  INPUT_ACTION_NONE
#define DI7_FUNCTION                INPUT_FUNCTION_NONE

// Amax
#define DI8_MODE                    IO_MODE_DISABLED
#define DI8_ACTION                  INPUT_ACTION_NONE
#define DI8_FUNCTION                INPUT_FUNCTION_NONE

Any time I try a touchplate probing in Chilipeppr I get immediate hang with no Z movement and shutdown state. 
The Due pin 18 is working ok in a normal sketch.
I've tried setting it as ACTIVE_HIGH even though my plate is set as NO (pull-up).
I've even tried using a piece of wire with the Arduino disconnected from the machine and pulling it to 3.3V or GND. Always get a shutdown state! I thought strapping it to 3.3V should keep it indefinitely in "searching" mode since it waits for that H->L edge, but all I get is instant shutdown.

Any help please?

aldenhart, 2017-01-21:
What g2 branch and build number are you running?

executivul, 2017-01-21:
I am using:
{"r":{"fv":0.99,"fb":100.19,"fbs":"100.19-1-g6f95-dirty","fbc":"settings_3040.h","hp":3,"hv":0,"id":"0084-d639-18c6-7bd","msg":"SYSTEM READY"},"f":[1,0,1]}

And the exact probing sequence is:
G21 G91 (Use mm and rel coords)
 G38.2 Z-20 F50
{"r":{},"f":[1,0,32]}
{"r":{},"f":[1,0,15]}
{"qr":47}
{"er":{"fb":100.19,"st":250,"msg":"Probe cycle failed - Probing error - invalid probe destination"}}
{"qr":46}
{"qr":48}
{"sr":{"stat":2,"feed":50,"dist":1}}

executivul, 2017-01-21:
I've traced it down to the cycle_probing.cpp
It needs z homing input even though I don't use it for homing.

executivul, 2017-01-21:
Now probing works for setting Z height. But when trying autolevel in Chilipeppr after a few probes I get: Probe cycle failed - probe failed to change.

executivul, 2017-01-21:
It's the bug with the probe state not changing on probe release. 
I've tried changing INPUT_LOCKOUT_MS to 0 in gpio.h but it still does the same.

executivul, 2017-01-22:
I've commented out lines 120-122 and 132 in gpio.c which handles lockout timer as I have external debouncing cap. It is working ok now. 
BUT I get a delay after moving in position and before the actual probing start which is very annoying when probing several hundreds points. Any idea how to reduce that lag?

xpix, 2017-01-22:
Please check this link of long delay in probing:

https://plus.google.com/+FrankHerrmann1967/posts/TqK3g1RwYsy

aldenhart, 2017-01-22:
Thanks for the investigation on this. We'll incorporate in an upcoming push. 

Also make sure the soft limits are not interfering with the probe. You may want to turn them off for the duration. We are looking into this as well. It looks like in some cases the probes can extend beyond the z min, which makes them fail, but if you are really going to stop on Z then it's not a problem. We are looking at the cases.

executivul, 2017-01-22:
After working ok for a few hundred points (about 7-8 complete board tests) now I get the "probe failed to change" again... and Cloud9 asked me for credit card to sign up to modify the sillypepper autolevel widget.. I'm afraid I'm giving up for the moment. I'll go back to grbl on Arduino Nano and try bCNC or at least keep pressing "~" whenever the stupid serial-json stops seeing the grbl ok response.
I haven't thrown the Due at the bin yet and I'll wait for your push. Unfortunately I've spend two weeks already focusing on this and other things are lagging behind.
Thanks again for all your help.

hurr1k4ne, 2017-01-28:
I can confirm the problem in 100.19 and aldenhart's update from today.

Homing will lead to shutdown until you define Z probe port. it is getting better when setting Zmin and Zmax to not 0 (-60/60).

When using auto level widget in chilipeppr i can probe 4-5 points until immediate shutdown when next probing starts.

aldenhart, 2017-01-28:
Have you tried turning off soft limits?

hurr1k4ne, 2017-01-28:
Yes,
i disabled all of them, then i can do some probing until it stops.

[sl]  soft limit enable           0 [0=disable,1=enable]
[lim] limit switch enable         0 [0=disable,1=enable]
[saf] safety interlock enable     0 [0=disable,1=enable]

[image]

$z:
[zam] z axis mode                 1 [standard]
[zvm] z velocity maximum        600 mm/min
[zfr] z feedrate maximum        300 mm/min
[ztn] z travel minimum          -60.000 mm
[ztm] z travel maximum            0.000 mm
[zjm] z jerk maximum           3000 mm/min^3 * 1 million
[zjh] z jerk homing            5000 mm/min^3 * 1 million
[zhi] z homing input              6 [input 1-N or 0 to disable homing this axis]
[zhd] z homing direction          0 [0=search-to-negative, 1=search-to-positive]
[zsv] z search velocity         250 mm/min
[zlv] z latch velocity        25.00 mm/min
[zlb] z latch backoff             4.000 mm
[zzb] z zero backoff              2.000 mm
g2core[mm] ok> 

Serial Console without filter:
 G38.2 Z-1.5 F25
{"r":{},"f":[1,0,16]}
{"qr":47}
{"qr":47}
{"qr":48}
{"prb":{"e":0, "z":0.500}}
{"er":{"fb":100.19,"st":250,"msg":"Probe cycle failed - Probing error - probe failed to change."}}
{"qr":47}
{"qr":48}
{"sr":{"momo":4,"stat":2}}
 G0 Z1.5
 G0 X0 Y20
 G0 Z0.5
{"r":{},"f":[1,204,8]}
{"r":{},"f":[1,204,10]}
{"r":{},"f":[1,204,8]}
 G38.2 Z-1.5 F25
{"r":{},"f":[1,204,16]}

http://pastebin.com/vpMFc1X4

hurr1k4ne, 2017-01-28:
I added a small capacitor to the probe setup and tried the hack in gpio.cpp it works kind of. but not reliable...
[image]

hurr1k4ne, 2017-01-29:
I did a complete rewiring of probe setup with RG174 coax-cable, 4.7k pullup, 100nF capacitor.
Still no better result. So it should be a config or program error.

I changed the probing port to DO6 (Pin18) also not sucessful.

aldenhart, 2017-01-31:
I've opened up a new issue (#237) to address this issue and some others with probing.

aldenhart, 2017-02-03:
This should be handled in Issue #237, PR #240. I'll leave the issue open until it's confirmed.

closed 2017-03-13 by aldenhart

benjamind, 2017-06-04:
Sadly @aldenhart this is not actually fixed. Same behavior with the input latching high when it should have released.