synthetos/g2 #233: Motors always powered, POWER_MODE ignored?

FutureSharks, 2017-01-22:
Hi!
I have a C-Beam setup with an Arduino Due, gShield and 2A motors.

When I run [this old version of the firmware](http://synthetos.github.io/g2/), the power management seems to behave correct. The motors a powered when an axis is moving and they timeout after 2s and there is no power to them and they do not hum.

But I wanted a newer version of G2 and after compiling the latest edge branch I can't seem to get the motors to timeout. They are just always powered up as soon as the Due is turned on. The green lights are constantly lit on the gShield and they hum.

I've tried setting `Mx_POWER_MODE` setting in my config but it never seems to make any difference.

What am I doing wrong? Or is there something I don't understand?

My compile config is (copied from settings_shapeoko2.h and adjusted) [here](https://gist.github.com/FutureSharks/ca454a0fdfd9ab806f720942f42ab9b8).
My make command was: `user@iMac:g2/g2core$ make BOARD=gShield SETTINGS_FILE=settings_cbeam.h`
And here is the output of ``{1:n}/{2:n}/{3:n}` querying the config via serial connection:
```json
{"r":{"fv":0.99,"fb":100.19,"fbs":"100.19-1-g6f95-dirty","fbc":"settings_cbeam.h","hp":3,"hv":0,"id":"0084-d639-0084-9ce","msg":"SYSTEM READY"},"f":[1,0,1]}

{"r":{"1":{"ma":0,"sa":1.8,"tr":8,"mi":8,"su":200,"po":1,"pm":2,"pl":0.75}},"f":[1,0,8]}
{"sr":{"stat":3}}

{"r":{"2":{"ma":1,"sa":1.8,"tr":8,"mi":8,"su":200,"po":0,"pm":2,"pl":0.75}},"f":[1,0,8]}
{"sr":{"stat":3}}

{"r":{"3":{"ma":2,"sa":1.8,"tr":8,"mi":8,"su":200,"po":1,"pm":2,"pl":0.75}},"f":[1,0,8]}
{"sr":{"stat":3}}
```

Shouldn't there be an `mt` key in there?

ewidance, 2017-01-22:
Identified and workaroud described in #212

FutureSharks, 2017-01-23:
OK thanks. I'll close this issue.

closed 2017-01-23 by FutureSharks