synthetos/g2 #235: Incredible flash erasing

alessandrolecce, 2017-01-27:
I can understand how is it possible that happen something at the end of a cycle that the flash completely erase! And I have to reload .bin with bossa.exe .

I use Arduino due....Anyone have news about this?

ril3y, 2017-01-28:
@alessandrolecce 

I have not seen this issue.  

Typically to get any support on issues you need to open them in a way that is helpful.

Hardware Platform
Build Number, branch, etc.
Configuration dumped
and the gcode file/s

Also to reproduce this.

IE:
I send this file with this config and this build over XYZ sender then at the end.

alessandrolecce, 2017-01-29:
I use Arduino due platform... giny g2 100.19 version.
On native USB arduino due has an erase command on 1200 serial bps. There is a way to remove it? Or control the event more accurately, because on Industrial environment can happen some interference....I think this command can erase flash unawares.

ewidance, 2017-01-30:
Maybe you should check my post about chinese arduino DUE clones and erase signal : http://translate.google.com/translate?hl=en&sl=auto&tl=en&u=http%3A%2F%2Fwww.civade.com%2Fpost%2F2016%2F12%2F18%2FCorriger-les-probl%C3%A8mes-de-Reset-sur-Arduino-DUE-R3-et-clones-chinois&sandbox=1

alessandrolecce, 2017-01-31:
Great...I buy it with a sign: "product in EU"....resolved with your suggest!

closed 2017-01-31 by alessandrolecce

hidoba, 2019-10-11:
+1 resolved. I believe it's worth adding this issue to Troubleshooting page.

justinclift, 2019-10-15:
@hidoba Not a bad idea.  The wiki is open for anyone (on GitHub) to edit.  Any interest in adding this? :smile: