synthetos/g2 #242: G28.2 not setting axis position to 0 after homing cycle

wasper1, 2017-02-15:
Hello,

I've encountered an issue after migration from master version to edge version:

After homing cycle (in direction to minimum) position is set to Y_TRAVEL_MIN instead of 0

For example :
(#define Y_TRAVEL_MIN                -20.0)
'G28.2 Y0' 
after homing cycle response is:
"posy":-20.0

I've created fix setting pos to 0, but @aldenhart please clarify if such behaviour is demanded and if i can create PR

Thanks

aldenhart, 2017-02-16:
This change was made to accommodate setups where the max travel for Z (top) is set to zero and the min is a negative number, say -100, and the homing occurs on the minimum switch. Similarly for homing to a switch in the max position that is set to some positive number.

I'm curious about your case, where homing to the Y minimum switch would not have Y_TRAVEL_MIN not equal to zero. I'd like to see how this case works, and how it can be accommodated.

hurr1k4ne, 2017-02-17:
i remember situations like this using the super-touch-plate widget in chilipeppr with 100.19.
I saw the -20 value after probing too...

Can not reproduce now, as i got more "low level" problems to solve with my Due/gShield setup.

wasper1, 2017-03-22:
Ok I understand intentions, but in the settings_*.h there is something like Y_ZERO_BACKOFF constant which in my opinion says that after back off position is set to 0.

I think there are two solutions:
1. Rename constants *_ZERO_BACKOFF to *_TRAVEL_MIN_MAX_BACKOFF because it is misleading now. In my case I can perform G0 Y0 command after homing.
2. After back off set position to 0. In this case when homing is on min switch with negative min position (ie.  -100), You can correct manually configuration by adding 100 to ZERO_BACKOFF constant 

In my solution I've chosen option 2)