synthetos/g2 #243: g2core-api is 404

ctag, 2017-02-22:
Hi,

Just trying to get up to speed with the wiki. Links to https://github.com/synthetos/node-g2core-api return a 404 page, and my next best guess of https://github.com/synthetos/node-g2core-server looks like an empty repo.

Where is the connection software?

aldenhart, 2017-02-22:
Try the API link again - I just tried it and they both worked for me. SERVER is a blank repo for now.

ctag, 2017-02-22:
Thanks @aldenhart, but the link is still a 404 for me. I've also tried opening it on a coworker's computer, same result.

If you can open the link and I can't, perhaps the repo is set to [private](https://developer.github.com/v3/troubleshooting/)?

Mgilbride, 2017-02-25:
I can confirm. Not there. Also Motate directory is empty when I clone. Throws errors on compile.

giseburt, 2017-02-25:
Hi guys. The node-g2core-api is currently private. It's not ready for use
at the moment. Time permitting it will be up this week and fully public, as
well as available on npm.

As for the Motate directory being empty, it's a submodule and you have to
tell git to pull those. See
https://github.com/synthetos/g2/wiki/Troubleshooting#problem-the-motate-directory-is-empty-the-motate-project-is-missing
for full details and explanation.

I'll update and close this issue once the API is public.

Thank you!
-Rob

ctag, 2017-02-25:
Awesome, thank you for the clarification!

Mgilbride, 2017-02-25:
> problem-the-motate-directory-is-empty-the-motate-project-is-missing

Yes. Many thanks.

Mgilbride, 2017-02-26:
Sorry for follow on question

So stepped back to edge-100. Cloned repo and submodules as instructed.
Worked great.

Attempt a build with following failure. Unclear if I have made an error in
specifying make parameters. Appears there is a file missing.

make SETTINGS_FILE=settings_Makeoko2.h BOARD=Due
"Found that we're in Linux"
Compiling cpp ./canonical_machine.cpp
    -> build/Due/./canonical_machine.o
In file included from ./stepper.h:251:0,
                 from ./canonical_machine.cpp:99:
./board/ArduinoDue/board_stepper.h:32:29: fatal error: step_dir_driver.h:
No such file or directory
compilation terminated.
../Motate/MotateProject/motate/Motate.mk:469: recipe for target
'build/Due/./canonical_machine.o' failed
make: *** [build/Due/./canonical_machine.o] Error 1

giseburt, 2017-02-26:
I didn't realize edge-100 was still around. Hmm.

Well, try edge. Make sure you do a
git submodule update
after you switch branches.

Also, the settings file, did you mean M is settings_Makeoko2.h ? Or did you
mean Shapeoko2? Just checking.

Otherwise I don't see anything obvious that could cause that problem.

-Rob

Mgilbride, 2017-02-26:
Wiped repo and set it up again in case I made an error. Same result.

g2core$ make SETTINGS_FILE=settings_Makeoko2.h BOARD=Due
"Found that we're in Linux"
Installing the necessary tools...
cd ../Motate/MotateProject/motate/../../Tools && make "ARCH=gcc-arm-none-eabi"
make[1]: Entering directory '/home/matt/src/g2-edge-100/g2/Motate/Tools'
The gcc-arm-none-eabi tools need installed. Downloading linux/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2 and installing now.

mkdir -p linux && \
cd linux && \
wget "https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2"
--2017-02-25 21:06:08--  https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
Resolving launchpad.net (launchpad.net)... 91.189.89.222, 91.189.89.223
Connecting to launchpad.net (launchpad.net)|91.189.89.222|:443... connected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: https://launchpadlibrarian.net/268330503/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2 [following]
--2017-02-25 21:06:08--  https://launchpadlibrarian.net/268330503/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
Resolving launchpadlibrarian.net (launchpadlibrarian.net)... 91.189.89.228, 91.189.89.229
Connecting to launchpadlibrarian.net (launchpadlibrarian.net)|91.189.89.228|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 92600244 (88M) [application/x-tar]
Saving to: ‘gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2’

gcc-arm-none-eabi-5_4-201 100%[=======================================>]  88.31M  2.53MB/s   in 54s    

2017-02-25 21:07:03 (1.64 MB/s) - ‘gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2’ saved [92600244/92600244]

cd linux && \
rm -f "gcc-arm-none-eabi" && \
tar xjf gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
cd linux && \
ln -s "gcc-arm-none-eabi-5_4-2016q2" "gcc-arm-none-eabi"
make[1]: Leaving directory '/home/matt/src/g2-edge-100/g2/Motate/Tools'
Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.c
    -> build/Due/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o 
Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/system_sam3x.c
    -> build/Due/motate/cmsis/TARGET_Atmel/sam3x/source/templates/system_sam3x.o 
Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/gcc/startup_sam3x.c
    -> build/Due/motate/cmsis/TARGET_Atmel/sam3x/source/templates/gcc/startup_sam3x.o 
Compiling c ../Motate/MotateProject/motate/platform/atmel_sam/hooks.c
    -> build/Due/motate/platform/atmel_sam/hooks.o 
Compiling c ../Motate/MotateProject/motate/platform/atmel_sam/cortex_handlers.c
    -> build/Due/motate/platform/atmel_sam/cortex_handlers.o 
Compiling cpp ../Motate/MotateProject/motate/MotateUtilities.cpp
    -> build/Due/motate/MotateUtilities.o 
Compiling cpp ../Motate/MotateProject/motate/main.cpp
    -> build/Due/motate/main.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamTimers.cpp
    -> build/Due/motate/Atmel_sam_common/SamTimers.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamUART.cpp
    -> build/Due/motate/Atmel_sam_common/SamUART.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamDMA.cpp
    -> build/Due/motate/Atmel_sam_common/SamDMA.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamPower.cpp
    -> build/Due/motate/Atmel_sam_common/SamPower.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamPins.cpp
    -> build/Due/motate/Atmel_sam_common/SamPins.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamUniqueID.cpp
    -> build/Due/motate/Atmel_sam_common/SamUniqueID.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamSPI.cpp
    -> build/Due/motate/Atmel_sam_common/SamSPI.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamServiceCall.cpp
    -> build/Due/motate/Atmel_sam_common/SamServiceCall.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam3x/SamUSB.cpp
    -> build/Due/motate/Atmel_sam3x/SamUSB.o 
Compiling cpp ../Motate/MotateProject/motate/platform/atmel_sam/syscalls.cpp
    -> build/Due/motate/platform/atmel_sam/syscalls.o 
Compiling cpp ./canonical_machine.cpp
    -> build/Due/./canonical_machine.o 
In file included from ./stepper.h:251:0,
                 from ./canonical_machine.cpp:99:
./board/ArduinoDue/board_stepper.h:32:29: fatal error: step_dir_driver.h: No such file or directory
compilation terminated.
../Motate/MotateProject/motate/Motate.mk:469: recipe for target 'build/Due/./canonical_machine.o' failed
make: *** [build/Due/./canonical_machine.o] Error 1

Mgilbride, 2017-02-26:
It is M. I have  a home brew machine generally based on that design. So just copied the Shapeoko2 settings and started from there. Thanks. I'll try your suggestions.

Mgilbride, 2017-02-26:
Same deal. Though the device directory has more sub directories than edge-100. The missing file is there. Trying to figure out why the include is not working. 

[image]

giseburt, 2017-02-26:
> Found it. Same as issue 123 towards the bottom.
>
> "Hi @stg <https://github.com/stg>,
>
> You've run into an interesting issue. The BOARD=Due isn't supposed to
> work, but once I tested it I found that it in fact does. It's Motate that
> has a BOARD=Due target, and I will likely remove that shortly to keep this
> from happening again.
>
> Use BOARD=gShield in your CONFIG settings.
>
> I'm sorry for the Due confusion, and thank you for testing this.
>
> -Rob"
>
> Board=Due is not a valid compile option though it works due to its
> presence in Motate. Unless you have since implemented it as an option. I
> guessed at Due as a board option since I do not fully understand the
> options available at this time. Using Board=gShield seems to get the job
> done.

Oh yes, I missed that. Good catch. I'll rename Due in Motate so it stops
tripping people up. Sorry about that!

justinclift, 2018-04-12:
As a general data point (1 year later :wink:) the [node-g2core-api](https://github.com/synthetos/node-g2core-api) repo is public these days.

Tried it after logging out of GitHub too, just to make sure, and yep... still able to access it.  So, closing this issue now.

If there's something in this issue that still needs doing (eg the rename in Motate), then people are welcome to re-open this. :smile:

closed 2018-04-12 by justinclift