synthetos/g2 #247: Flow Control

Mgilbride, 2017-02-26:
A question rather than an issue. Some of the settings files in the edge branch lack FLOW_CONTROL_XON as an option listed in the settings options. However, the default settings file still has it. Just want to verify that the three options are still the same.

aldenhart, 2017-03-05:
When using the USB none of the flow controls matter, as the USB protocols take care of all the flow control. It's kind of a hold-over from the v8 where the FTDI chip needed to be drive by flow control. 

On g2 the USB is native to the code base so it doesn't matter.

That said, there is an option to talk to g2 with raw serial, assuming the host can drive that. In that case it's best to use RTS/CTS.

Mgilbride, 2017-03-06:
Got it. Many thanks for the explanation.

closed 2017-03-12 by aldenhart