synthetos/g2 #248: Status Reports

Mgilbride, 2017-02-27:
Status reports after ? give machine position only and g2core (mm) ok echo. I have it set to verbose in my config settings. Currently, this is the only way I can see an Alarm or other states. But no status is actually given.

 ?
 {"ej":""}
Line number:       945
X position:         40.687 mm
Y position:         62.914 mm
Z position:          5.000 mm
g2core[mm] ok> 

Settings:

[fb]  firmware build             100.21
[fbs] firmware build "100.21-dirty"
[fbc] firmware config "settings_makeoko2.h"
[fv]  firmware version            0.99
[hp]  hardware platform           3.00
[hv]  hardware version            0.00
[id]  g2core ID   0084-d639-1842-6bd
[jt]  junction integrgation time  0.75
[ct]  chordal tolerance           0.0100 mm
[sl]  soft limit enable           1 [0=disable,1=enable]
[lim] limit switch enable         1 [0=disable,1=enable]
[saf] safety interlock enable     0 [0=disable,1=enable]
[m48e] overrides enabled          1 [0=disable,1=enable]
[mfoe] manual feed override enab  0 [0=disable,1=enable]
[mfo]  manual feedrate override   1.000 [0.05 < mfo < 2.00]
[mtoe] manual traverse over enab  0 [0=disable,1=enable]
[mto]  manual traverse override   1.000 [0.05 < mto < 1.00]
[mt]  motor idle timeout          2.00 seconds
[spep] spindle enable polarity    1 [0=active_low,1=active_high]
[spdp] spindle direction polarity 0 [0=CW_low,1=CW_high]
[spph] spindle pause on hold      1 [0=no,1=pause_on_hold]
[spdw] spindle dwell time         1.0 seconds
[ssoe] spindle speed override ena 0 [0=disable,1=enable]
[sso] spindle speed override      1.000 [0.050 < sso < 2.000]
[cofp] coolant flood polarity     1 [0=low is ON,1=high is ON]
[comp] coolant mist polarity      1 [0=low is ON,1=high is ON]
[coph] coolant pause on hold      0 [0=no,1=pause_on_hold]
[tv]  text verbosity              1 [0=silent,1=verbose]
[ej]  enable json mode            2 [0=text,1=JSON,2=auto]
[jv]  json verbosity              5 [0=silent,1=footer,2=messages,3=configs,4=linenum,5=verbose]
[qv]  queue report verbosity      2 [0=off,1=single,2=triple]
[sv]  status report verbosity     2 [0=off,1=filtered,2=verbose]
[si]  status interval           250 ms
[gpl] default gcode plane         0 [0=G17,1=G18,2=G19]
[gun] default gcode units mode    1 [0=G20,1=G21]
... etc., etc.

aldenhart, 2017-02-27:
Heres how to set up the variable you want to see in status reports:
https://github.com/synthetos/g2/wiki/Status-Reports

Mgilbride, 2017-02-27:
Thanks Alden. Got that. And can just set up the reports from command line
after each restart or set up a Chilipeppr startup . But the settings file I
am using has the following default set:

#define STATUS_REPORT_DEFAULTS
"line","posx","posy","posz","bcr","feed","vel","momo","stat"

Should that not output all of those values by default for a firmware
compiled using those settings?

aldenhart, 2017-02-27:
The status reports in the settings file are just the defaults. Whatever you write in the SR line will override that. If you want those variables include them in your SR settings line. Or build a defaults line that's got what you want in it.

Mgilbride, 2017-02-27:
I think I've done that. All of the settings in my defsults line sre not
being displayed. If there is an sr line overriding that, I don't know where
it is coming from.But I can write one to the serial console to get what I
want.

aldenhart, 2017-03-05:
I think CP outputs an SR line in it's board setup.

mennodegraaf, 2017-03-05:
Many examples in the settings folder (like settings_shapeoko2.h) have "bcr" in the status report. This does not exist (anymore?) though and causes the status fields after it to be ignored. Also the comments in settings_default.h still mentions "bcr" as one of the possible fields. Makes sense to update this to avoid unexpected behaviour in CP.

aldenhart, 2017-03-12:
Closed in PR #254

closed 2017-03-12 by aldenhart