synthetos/g2 #249: G2/G3 radius mode resulting in Arc specification error - missing offset(s)

Mephistofeles, 2017-03-04:
When I try to use for example G2 X10 Y10 R10 F50 I get an error saying that offsets are missing. To make it work I can add I1 J1 or I100 J100 and it works the same - offsets doesn't seem to be used and that's fine.
I think the issue was introduced in commit [Updated arcs to agree with G18 motion in v8 edge-0.97](https://github.com/synthetos/g2/commit/aa2135ecad899b10225fd87bd6d6fc6475d4813f)
in file [plan_arc.cpp](https://github.com/synthetos/g2/blob/aa2135ecad899b10225fd87bd6d6fc6475d4813f/g2core/plan_arc.cpp#L231
)
This check (line 231) should probably occur only when in offsets mode.

aldenhart, 2017-03-05:
Thanks for the heads up. I'll take care of this in a subsequent push

aldenhart, 2017-03-12:
Closed in PR #254

closed 2017-03-12 by aldenhart