synthetos/g2 #252: Windows10 / Atmel Studio 7 compilation

shczm, 2017-03-10:
Hi,

Followed the instructions carefully, however, tried many times did not succeed. 

1. recipe for target '../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin' failed	g2core	C:\github\g2\Motate\MotateProject\motate\Motate.mk	404

2. recipe for target 'win32/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip' failed	g2core	C:\github\g2\g2core\Makefile	143

if the ATMEL ICE needed when compiling?

Thankd

Dan211, 2017-03-27:
This looks like youre missing Motate, did you clone the repository using the --recusive flag?

geosmall, 2017-04-01:
I'm having the same issue on Win 10 with Atmel Studio 7.

Motate present, --recursive was used.  Builds fine for me on my Linux machine.

[image]

geosmall, 2017-04-01:
Ok it was my fault for not having git-scm installed (per instructions).  A manual install into Tools/win32/ of gcc-arm-none-eabi compiles correctly.

geosmall, 2017-04-01:
Looking further this is an issue within Motate repo as Motate.mk is root cause.

closed 2017-04-22 by shczm