synthetos/g2 #255: Planner assertion failure caused by feedhold

gilbertoverg, 2017-03-23:
Hi, sometimes i'm getting "Planner assertion failure - mp_exec_aline() zero length move" after sending the feedhold character. This is a log from my test application:
```
2017-03-23T08:28:12.786Z Opening: /dev/tty.usbmodem1411
2017-03-23T08:28:12.841Z Port opened
2017-03-23T08:28:12.851Z RX: {"r":{"fv":0.99,"fb":100.23,"fbs":"","fbc":"settings_ebe.h","hp":3,"hv":0,"id":"0084-d639-18c6-8ce","msg":"SYSTEM READY"},"f":[1,0,1]}
2017-03-23T08:28:13.841Z TX: G90
2017-03-23T08:28:13.841Z TX: F6000
2017-03-23T08:28:13.842Z RX: {"r":{},"f":[1,0,4]}
2017-03-23T08:28:13.842Z RX: {"r":{},"f":[1,0,6]}
2017-03-23T08:28:13.967Z RX: {"sr":{"line":0,"posx":0,"posy":0,"posz":0,"posa":0,"ofsx":0,"ofsy":0,"ofsz":0,"ofsa":0,"feed":6000,"vel":0,"unit":1,"coor":1,"dist":0,"path":0,"plan":0,"stat":1,"homx":0,"homy":0,"homz":0}}
2017-03-23T08:28:14.343Z TX: G1 X100
2017-03-23T08:28:14.344Z RX: {"r":{},"f":[1,0,8]}
2017-03-23T08:28:14.344Z RX: {"qr":47}
2017-03-23T08:28:14.470Z RX: {"sr":{"posx":0.563,"vel":1221.13,"stat":5}}
2017-03-23T08:28:14.596Z RX: {"sr":{"posx":7.852,"vel":5465.64}}
2017-03-23T08:28:14.721Z RX: {"sr":{"posx":20.26,"vel":6000}}
2017-03-23T08:28:14.848Z RX: {"sr":{"posx":32.845}}
2017-03-23T08:28:14.973Z RX: {"sr":{"posx":45.43}}
2017-03-23T08:28:15.097Z RX: {"sr":{"posx":57.866}}
2017-03-23T08:28:15.222Z RX: {"sr":{"posx":70.451}}
2017-03-23T08:28:15.349Z RX: {"sr":{"posx":83.036}}
2017-03-23T08:28:15.474Z RX: {"sr":{"posx":94.932,"vel":4639.3}}
2017-03-23T08:28:15.599Z RX: {"sr":{"posx":99.866,"vel":447.68}}
2017-03-23T08:28:15.662Z RX: {"qr":48}
2017-03-23T08:28:15.724Z RX: {"sr":{"posx":100,"vel":0,"stat":3}}
2017-03-23T08:28:16.848Z TX: G1 X125
2017-03-23T08:28:16.850Z RX: {"r":{},"f":[1,0,8]}
2017-03-23T08:28:16.850Z RX: {"qr":47}
2017-03-23T08:28:16.976Z RX: {"sr":{"posx":100.577,"vel":1241.83,"stat":5}}
2017-03-23T08:28:17.050Z TX: !
2017-03-23T08:28:17.101Z RX: {"sr":{"posx":107.68,"vel":5146.4,"stat":6}}
2017-03-23T08:28:17.227Z RX: {"sr":{"posx":118.815,"vel":4842.95}}
2017-03-23T08:28:17.352Z RX: {"sr":{"posx":124.724,"vel":786.8}}
2017-03-23T08:28:17.432Z RX: {"er":{"fb":100.23,"st":94,"msg":"Planner assertion failure - mp_exec_aline() zero length move"}}
2017-03-23T08:28:17.477Z RX: {"sr":{"posx":125,"vel":0}}
2017-03-23T08:28:19.553Z Done
```

aldenhart, 2017-03-24:
Thanks for the heads up.

gilbertoverg, 2017-03-24:
If the feedhold is sent during the acceleration period of a short movement (acceleration and deceleration only) the error is generated. If the movement is long (acceleration, constant speed and deceleration) no errors are generated. 
Short movement:
```
2017-03-24T13:14:15.918Z Opening: /dev/tty.usbmodem1411
2017-03-24T13:14:15.975Z Port opened
2017-03-24T13:14:15.984Z RX: {"r":{"fv":0.99,"fb":100.23,"fbs":"","fbc":"settings_ebe.h","hp":3,"hv":0,"id":"0084-d639-18c6-8ce","msg":"SYSTEM READY"},"f":[1,0,1]}
2017-03-24T13:14:16.979Z TX: G90
2017-03-24T13:14:16.979Z TX: F6000
2017-03-24T13:14:16.981Z RX: {"r":{},"f":[1,0,4]}
2017-03-24T13:14:16.981Z RX: {"r":{},"f":[1,0,6]}
2017-03-24T13:14:17.106Z RX: {"sr":{"line":0,"posx":0,"posy":0,"posz":0,"posa":0,"ofsx":0,"ofsy":0,"ofsz":0,"ofsa":0,"feed":6000,"vel":0,"unit":1,"coor":1,"dist":0,"path":0,"plan":0,"stat":1,"homx":0,"homy":0,"homz":0}}
2017-03-24T13:14:18.983Z TX: G1 X25
2017-03-24T13:14:18.985Z RX: {"r":{},"f":[1,0,7]}
2017-03-24T13:14:18.985Z RX: {"qr":47}
2017-03-24T13:14:19.110Z RX: {"sr":{"posx":0.577,"vel":1241.83,"stat":5}}
2017-03-24T13:14:19.184Z TX: !
2017-03-24T13:14:19.236Z RX: {"sr":{"posx":7.68,"vel":5146.4,"stat":6}}
2017-03-24T13:14:19.361Z RX: {"sr":{"posx":18.815,"vel":4842.95}}
2017-03-24T13:14:19.487Z RX: {"sr":{"posx":24.724,"vel":786.8}}
2017-03-24T13:14:19.567Z RX: {"er":{"fb":100.23,"st":94,"msg":"Planner assertion failure - mp_exec_aline() zero length move"}}
2017-03-24T13:14:19.612Z RX: {"sr":{"posx":25,"vel":0}}
2017-03-24T13:14:21.687Z Port closed
```

Long movement (no problems):
```
2017-03-24T13:14:57.362Z Opening: /dev/tty.usbmodem1411
2017-03-24T13:14:57.418Z Port opened
2017-03-24T13:14:57.427Z RX: {"r":{"fv":0.99,"fb":100.23,"fbs":"","fbc":"settings_ebe.h","hp":3,"hv":0,"id":"0084-d639-18c6-8ce","msg":"SYSTEM READY"},"f":[1,0,1]}
2017-03-24T13:14:58.420Z TX: G90
2017-03-24T13:14:58.421Z TX: F6000
2017-03-24T13:14:58.422Z RX: {"r":{},"f":[1,0,4]}
2017-03-24T13:14:58.422Z RX: {"r":{},"f":[1,0,6]}
2017-03-24T13:14:58.546Z RX: {"sr":{"line":0,"posx":0,"posy":0,"posz":0,"posa":0,"ofsx":0,"ofsy":0,"ofsz":0,"ofsa":0,"feed":6000,"vel":0,"unit":1,"coor":1,"dist":0,"path":0,"plan":0,"stat":1,"homx":0,"homy":0,"homz":0}}
2017-03-24T13:15:00.426Z TX: G1 X100
2017-03-24T13:15:00.429Z RX: {"r":{},"f":[1,0,8]}
2017-03-24T13:15:00.429Z RX: {"qr":47}
2017-03-24T13:15:00.553Z RX: {"sr":{"posx":0.563,"vel":1221.13,"stat":5}}
2017-03-24T13:15:00.631Z TX: !
2017-03-24T13:15:00.678Z RX: {"sr":{"posx":7.852,"vel":5465.64,"stat":6}}
2017-03-24T13:15:00.804Z RX: {"sr":{"posx":20.15,"vel":5654.86}}
2017-03-24T13:15:00.929Z RX: {"sr":{"posx":28.064,"vel":1556.2}}
2017-03-24T13:15:01.055Z RX: {"sr":{"posx":28.834,"vel":0}}
2017-03-24T13:15:03.138Z Port closed
```

aldenhart, 2017-03-24:
Thanks for the detailed diagnosis.

aldenhart, 2017-05-01:
Is this issue still a problem with the newer branches?

gilbertoverg, 2017-05-01:
In edge version 100.26 problem still exists

gilbertoverg, 2017-05-14:
This issue does not occur in master branch

aldenhart, 2017-05-14:
I see the moves you are using to produce the error. I'm trying to understand the timing a little better. What are you using the send the Gcode? I'd like to be able to reproduce your exact timing if I can.

Also, can you post some more information about your setup and settings? The settings file would be useful if that's where you set things, as would the settings dump.

gilbertoverg, 2017-05-14:
This is my settings file: [settings_ebe.txt](https://github.com/synthetos/g2/files/999264/settings_ebe.txt), i'm compiling with BOARD=gShield and i'm using an arduino due. To control my cnc machine i'm using a custom java application.
I can always generate the error with this timing:
Reset the arduino and open serial port
send: G90
send: F6000
send: G1 X25
pause 200 milliseconds
send: !
I've coded this java application to test this issue: [Test.zip](https://github.com/synthetos/g2/files/999280/Test.zip), simply run with java -jar Test.jar