synthetos/g2 #256: Feedhold ignored during high speed transmission

gilbertoverg, 2017-03-24:
Hi,
I'm using g2core with 2 serial ports exposed and 2 windows of CoolTerm (my operating system is macOS Sierra) to send gcode. I've noticed that during high speed transmission of short movements to the data channel the feedhold character sent to the control channel is ignored.
This is my test file:
[littleMovements.txt](https://github.com/synthetos/g2/files/868668/littleMovements.txt)

Thank you for your assistance

aldenhart, 2017-03-24:
More recent coolterm has some issues keeping up. I don;t know what they did. I'm still using 1.4.3 on OSX. No problems.

gilbertoverg, 2017-03-24:
I've tried with the latest version 1.4.7 and the 1.4.0b5 downloaded from master branch, for me result is the same.

rkoe, 2018-04-26:
On some of my tests, the feedhold was not ignored, but delayed for several seconds, especially during G2/G3-movements. Is this intentional or a bug?

Example:
```
...
!
{'sr': {'vel': 182.74, 'posy': -0.082, 'stat': 6, 'posx': 0.75}}
{'sr': {'vel': 141.9, 'posy': 0.335, 'posx': 1.287}}
{'sr': {'vel': 136.08, 'posy': 0.666, 'posx': 1.817}}
{'sr': {'vel': 146.95, 'posy': 0.929, 'posx': 2.337}}
{'sr': {'vel': 127.03, 'posy': 1.135, 'posx': 2.856}}
{'sr': {'vel': 111.26, 'posy': 1.289, 'posx': 3.361}}
{'sr': {'vel': 137.69, 'posy': 1.401, 'posx': 3.869}}
{'sr': {'vel': 131.9, 'posy': 1.47, 'posx': 4.383}}
{'sr': {'vel': 122.37, 'posy': 1.498, 'posx': 4.889}}
{'sr': {'vel': 637.7, 'posy': 1.5, 'posx': 7.389, 'momo': 1}}
{'sr': {'vel': 70.71, 'posx': 8.802}}
{'sr': {'vel': 0, 'posx': 8.844}}
```

giseburt, 2018-04-26:
CoolTerm appears to not handle multiple windows properly. I have tested with the sender in node-g2core-api the dual-port mode extensively, and the machine responds within a few milliseconds of receiving the !.

That's part of the reason for making that utility. Unfortunately it's a little outdated and needs some love. It works great if you use node 6.x, but needs updated to work with newer node and newer node-serialport. (Pull requests welcome, of course. :) )

-Rob

rkoe, 2018-04-26:
Oh, maybe I shouldn't have posted it in this issue.
For my tests I did not use CoolTerm but minicom or Python-serial (so I have full control and can check what is sent/received when). And there, I sometimes got such a delay.

closed 2023-02-16 by ril3y