synthetos/g2 #257: Probe issues - probe already engaged/cycle fail

Dan211, 2017-03-27:
I am having issues using the probe function on g2Core 100.23, it gives an error saying that the probe is engaged (when its clearly not touching) or throwing a probe cycle failed error. My setup is a Shapeoko 2 mill hooked up to a due with a gshield. Input 5 on pin 18 is configured to be the probe, configured normally open and it is also externally pulled high by a 10k resistor. The probe pin is connected to the bit in the spindle by an alligator clip and another alligator clip connects ground to a blank pcb placed under the bit. I use the tinyg chilipeppr enviroment, both logs are from the terminal and I enter the gcode manually. The machine registers chages when i send {in5:n} to check the state, true when I touch the probe to the pcb and false when they are not touching. I have use my oscilloscope to measure noise in the signal and it does not appear to be significant, I even put 2 * 0.1μF multilayer caps (to reducre noise) from ground to the probe and from 3.3v to the probe, with no success. 
>
First test. I check the pin configuration and state, then run a probe. (Bold is my input)
>**{di5:n}**
{"r":{"di5":{"mo":0,"ac":0,"fn":4}},"f":[1,0,9]}
{"sr":{"stat":3}}
**{in5:n}**
{"r":{"in5":false},"f":[1,0,9]}
{"sr":{"stat":3}}
**G0 X0 Y0 Z0.5**
{"r":{},"f":[1,0,14]}
{"sr":{"posz":0.5,"momo":0,"stat":3}}
{"sr":{"stat":3}}
 **G38.2 Z-10 F25**
{"r":{},"f":[1,0,15]}
{"er":{"fb":100.23,"st":254,"msg":"Probe is already tripped - probe error"}}
{"sr":{"momo":4,"stat":2}}

Sometimes this happens and I get probe cycle failed instead, this happens exactly when the probe touches the plate (never reaching -2 given by G38.2 Z-2 F25) and then the machine goes into shutdown. 
>
Second test. I check pin state (configuration is the same) then run a probe
> **{in5:n}**
{"r":{"in5":false},"f":[1,0,9]}
{"sr":{"stat":3}}
**G38.2 Z-2 F25**
{"r":{},"f":[1,0,14]}
{"sr":{"posz":-0.087,"vel":25,"momo":1,"stat":7}}
{"sr":{"posz":-0.192}}
{"sr":{"posz":-0.296}}
{"sr":{"posz":-0.4}}
{"sr":{"posz":-0.505}}
{"sr":{"posz":-0.609}}
{"sr":{"posz":-0.713}}
{"sr":{"posz":-0.817}}
{"sr":{"posz":-0.922}}
{"sr":{"posz":-1.026}}
{"sr":{"posz":-1.13}}
{"er":{"fb":100.23,"st":250,"msg":"Probe cycle failed - probing failed"}}
{"prb":{"e":0, "z":-2.000}}
{"sr":{"posz":-2,"vel":0,"momo":4,"stat":2}}

From my settings file
>// Zmin
#define DI5_MODE                    NORMALLY_OPEN  // Z probe
#define DI5_ACTION                  INPUT_ACTION_NONE
#define DI5_FUNCTION                INPUT_FUNCTION_PROBE

Thanks for the help

giseburt, 2017-03-28:
Hi Dan. I just wanted to check and see what firmware you're running. Could you please tell us the output from `{"fbs":n}` and `{"fbc":n}`?

On thing to look for is do any of the other inputs have `"fn":4`? Checking with `{"di1fn":n}` etc. I believe it will silently use the first input with the function set to probe.

-Rob

Dan211, 2017-03-28:
{"fbs":n} returns {"r":{"fbs":"100.23-dirty"},"f":[1,0,11]} and 
{"fbc":n} returns {"r":{"fbc":"settings_shapeoko2Custom.h"},"f":[1,0,11]}

heres my settings file, I dont know how to format it properly so this is just a paste:

 ***> NOTE: The init message must be a single line with no CRs or LFs
#define INIT_MESSAGE "Initializing configs to Shapeoko2 500mm profile"

//**** GLOBAL / GENERAL SETTINGS ******************************************************

// Machine configuration settings

#define JUNCTION_INTEGRATION_TIME   0.75                    // cornering - between 0.10 and 2.00 (higher is faster)
#define CHORDAL_TOLERANCE           0.01                    // chordal tolerance for arcs (in mm)

#define SOFT_LIMIT_ENABLE           0                       // 0=off, 1=on
#define HARD_LIMIT_ENABLE           1                       // 0=off, 1=on
#define SAFETY_INTERLOCK_ENABLE     1                       // 0=off, 1=on

#define SPINDLE_ENABLE_POLARITY     1                       // 0=active low, 1=active high
#define SPINDLE_DIR_POLARITY        0                       // 0=clockwise is low, 1=clockwise is high
#define SPINDLE_PAUSE_ON_HOLD       true
#define SPINDLE_DWELL_TIME          1.0

#define COOLANT_MIST_POLARITY       1                       // 0=active low, 1=active high
#define COOLANT_FLOOD_POLARITY      1                       // 0=active low, 1=active high
#define COOLANT_PAUSE_ON_HOLD       false

// Communications and reporting settings

#define USB_SERIAL_PORTS_EXPOSED	2						// 1=single endpoint usb, 2=dual endpoint usb
#define COMM_MODE                   JSON_MODE               // one of: TEXT_MODE, JSON_MODE
#define XIO_ENABLE_FLOW_CONTROL FLOW_CONTROL_RTS            // FLOW_CONTROL_OFF, FLOW_CONTROL_RTS

#define TEXT_VERBOSITY              TV_VERBOSE              // one of: TV_SILENT, TV_VERBOSE
#define JSON_VERBOSITY              JV_MESSAGES             // one of: JV_SILENT, JV_FOOTER, JV_CONFIGS, JV_MESSAGES, JV_LINENUM, JV_VERBOSE
#define QUEUE_REPORT_VERBOSITY      QR_OFF                  // one of: QR_OFF, QR_SINGLE, QR_TRIPLE

#define STATUS_REPORT_VERBOSITY     SR_FILTERED             // one of: SR_OFF, SR_FILTERED, SR_VERBOSE

#define STATUS_REPORT_MIN_MS        100                     // milliseconds - enforces a viable minimum
#define STATUS_REPORT_INTERVAL_MS   250                     // milliseconds - set $SV=0 to disable

#define STATUS_REPORT_DEFAULTS "line","posx","posy","posz","feed","vel","unit","coor","dist","admo","frmo","momo","stat"
//#define STATUS_REPORT_DEFAULTS "line","posx","posy","posz","feed","vel","momo","stat"

// Alternate SRs that report in drawable units
//#define STATUS_REPORT_DEFAULTS "line","vel","mpox","mpoy","mpoz","mpoa","coor","ofsa","ofsx","ofsy","ofsz","dist","unit","stat","homz","homy","homx","momo"
//#define STATUS_REPORT_DEFAULTS "_ts1","_cs1","_es1","_xs1","_fe1","line","posx","posy","posz","vel","stat"

// Gcode startup defaults
#define GCODE_DEFAULT_UNITS         MILLIMETERS             // MILLIMETERS or INCHES
#define GCODE_DEFAULT_PLANE         CANON_PLANE_XY          // CANON_PLANE_XY, CANON_PLANE_XZ, or CANON_PLANE_YZ
#define GCODE_DEFAULT_COORD_SYSTEM  G54                     // G54, G55, G56, G57, G58 or G59
#define GCODE_DEFAULT_PATH_CONTROL  PATH_CONTINUOUS
#define GCODE_DEFAULT_DISTANCE_MODE ABSOLUTE_DISTANCE_MODE

// *** motor settings ************************************************************************************

#define MOTOR_POWER_MODE            MOTOR_POWERED_IN_CYCLE  // default motor power mode (see cmMotorPowerMode in stepper.h)
#define MOTOR_POWER_TIMEOUT         60.00                    // motor power timeout in seconds

#define M1_MOTOR_MAP                AXIS_X                  // 1ma
#define M1_STEP_ANGLE               1.8                     // 1sa
#define M1_TRAVEL_PER_REV           40.00                   // 1tr
#define M1_MICROSTEPS               8                       // 1mi  1,2,4,8,16,32
#define M1_POLARITY                 1                       // 1po  0=normal, 1=reversed
#define M1_POWER_MODE               MOTOR_POWER_MODE        // 1pm  TRUE=low power idle enabled
#define M1_POWER_LEVEL              0.500

[... 98 lines cut ...]
// Ymax
#define DI4_MODE                    NORMALLY_CLOSED
#define DI4_ACTION                  INPUT_ACTION_STOP
//#define DI4_ACTION                  INPUT_ACTION_NONE
#define DI4_FUNCTION                INPUT_FUNCTION_LIMIT

// Zmin
#define DI5_MODE                    NORMALLY_OPEN  // Z probe
#define DI5_ACTION                  INPUT_ACTION_NONE
#define DI5_FUNCTION                INPUT_FUNCTION_PROBE

// Zmax
#define DI6_MODE                    NORMALLY_CLOSED
#define DI6_ACTION                  INPUT_ACTION_NONE
#define DI6_FUNCTION                INPUT_FUNCTION_NONE

// Kill button
#define DI7_MODE                    NORMALLY_OPEN
#define DI7_ACTION                  INPUT_ACTION_HALT
#define DI7_FUNCTION                INPUT_FUNCTION_SHUTDOWN

// Hardware interlock input
#define DI8_MODE                    NORMALLY_CLOSED
#define DI8_ACTION                  INPUT_ACTION_NONE
#define DI8_FUNCTION                INPUT_FUNCTION_INTERLOCK

Dan211, 2017-03-28:
I did some more testing and I think i found the problem, while the signal is pretty noise free before and after the probe engages, the signal while the probe engages jumps around

[image]

Dan211, 2017-03-28:
Pressed the wrong button on moble...

aldenhart, 2017-03-28:
You are running a Due, so the inputs are not conditioned in any way. Try pulling the line up to 3.3v with a resistor - probably about 10K, and also putting an 0.1 uF cap to ground. That should clean it up.

Dan211, 2017-03-28:
Thanks, I actually did that already (as explained in the original post). Apparently that wasnt enough so i had to step up to a 4.7μF cap on the rising edge (signal line to ground) and a 0.1μF cap on the falling edge (signal line to 3.3v). This seems to have solved the issue but its not ideal since the rise time is so long, so the probe lifts ever so sightly above where it has to be, but this can be compensated for. I guess this solution will have to do, thanks.

closed 2017-03-28 by Dan211

hurr1k4ne, 2017-03-28:
I am facing the same Problem. I am very interested in a better solution. Thinking of opto-coupler or Buffer?

Dan211, 2017-03-28:
I have seen people use buffers somewhere but I can't find the thread, as for optoisolators, im not sure that they will work.... I just hooked one up on its own and it did nothing to help, Ill try to add some caps, see if they might help.

Dan211, 2017-04-06:
Im going to look into using a shmidt trigger for smoothing the probe signal, might work beter than a large cap.

mateoarv, 2017-10-25:
Just tried probing with the firmware from the edge branch, random errors everywhere, the caps didn't help either, it sometimes works, sometimes it doesn't.

mnesarco, 2019-10-27:
Hi @Dan211 did you finally found a workaround?