synthetos/g2 #260: how-to non-trivial kinematics

spiderdab, 2017-04-03:
Hi, I haven't found enough info on this, so I'm asking here.
I need to implement non-trivial kinematics using g2core on a arduino due, with self built step-dir drivers.
For what I understand kinematics calcs go into kinematics.cpp, but before I go studying all the files in the source, is there any other file to change you can point me to?
What I need is mechanically similar to a cable drawing machine, but with 4 winches, hanging a rotating head.
So i'm going to need 5 Joints (winch 1 to 4 plus the rotating head) and 3 (4?) axis ( X, Y, and C, with Z staying always to 0). 
I've already set these kinematics in LinuxCnc, but I really need to transfer this machine to Arduino Due.
Thanks for your interest, any help is of course appreciated.

giseburt, 2017-04-03:
Hello,

A few quick notes:
* In g2core all planning is in cartesian space. All limits are applied here as well. We need to find ways to allow the kinematics to interact with planning with as little of a performance impact as possible.
* g2core breaks moves that come from gcode into "blocks" which are point-to-point linear moves, then the blocks are broken into "sections" that represent the head (acceleration), body (cruise), and tail (deceleration) for each move, and finally each section is broken into brief - nominally 1ms for the current Due configuration - "segments" that are constant velocity linear moves. The kinematics are applied to each segment, which is where the system goes go from cartesian length-units (internally always mm) to joint-space fractional steps (floating point).

So we have identified a few things necessary for complete kinematics support:

- [x] Forward and inverse kinematics themselves. That is in place, with the [inverse](https://github.com/synthetos/g2/blob/dev-168-gquintic/g2core/kinematics.cpp#L116-L157) and [forward](https://github.com/synthetos/g2/blob/dev-168-gquintic/g2core/kinematics.cpp#L169-L214) both being in `kinematics.cpp`. Note that those are links to the code in the dev-168-gquintic branch.
  * The forward is rarely used, and at the moment only for probing and homing. Note that we can map an axis to multiple motors, and each motor may have it's own configuration.
  * These are applied on a segment-by-segment basis. 
  * The segments are converted to steps for each indicated motor by the kinematics layer.
  * We would like to later make this a run-time-selectable change, or at worse compile-time but cleaner - without the mess if `ifdef`s that's there now. (I put the CoreXY stuff in in a day, so it's a little sloppy.)
- [ ] Planner hooks - We need the ability for the kinematics layer to interact with the planner to identify limits that will need to be accounted for.
  * This also includes the ability for the planner to indicate a move is a traverse or not, in order to allow the kinematics to plan a move that is "direct" (traverse) or a move that is "linear" (feed) when the kinematics have a difference.
- [ ] Homing and probing hooks - for some kinematics homing is an elaborate procedure. Probing is sometimes almost the same as homing, and other times it's a hugely different process. So, we need to identify what kinematics needs in order to provide homing and probing procedures.

Is there anything I've missed? Perhaps for your particular case the current kinematics may be sufficient.

-Rob

spiderdab, 2017-04-04:
I understand. So I think you're suggesting to make my trials on the dev-168-gquintic branch?
I'm going to write my forward/inverse kinematics into the kinematics.cpp file, then, about homing. I think i'll have the need to send an homing signal, without touching any endstop. I mean i set the four winch-joints in a determinated position, then I press an homing button exchanging to inverse kins, with the joints in their corrected poistion to have X and Y in their homing position.
Is there already this possibility, or do I have to implement it?
Thank you for your explanations.

spiderdab, 2017-04-04:
Am I right to say that to add my special kins I should add my define into the settings_default in this way:
`// KINEMATICS (which may later change the following values)
#define KINE_CARTESIAN 0
#define KINE_CORE_XY 1
#define KINE_TETRAPOD 2

#ifndef KINEMATICS
#define KINEMATICS KINE_TETRAPOD
#endif`
 and than I should add a "settings.my_kins.h" file where to add all my special #define for machine constants?

mrarmyant, 2018-03-28:
Wait I' have searched all over, cane coreg2 do CoreXY?  Why is there so little information on it?

giseburt, 2018-03-28:
It's currently not in master or edge, and is not documented. It's in branch dev-168-gquintic. You can see the [Axidraw v3 settings](https://github.com/synthetos/g2/blob/dev-168-gquintic/g2core/settings/settings_axidraw_v3.h#L92) for an example. There's not a lot of configuration yet.

mrarmyant, 2018-03-28:
Cool, so it does in fact work! Now I just need to learn a lot about how to implement  that branch lol!