synthetos/g2 #261: Support Printrbot Simple Heated Bed

giseburt, 2017-04-05:
Some adjustments are necessary to support a heated bed on the Printrbot Simple 1608 (recently rebranded as the "Simple Pro").

This issue will track those changes.

closed 2017-04-05 by giseburt

justinclift, 2018-11-04:
@pr3sidentspence What software are you using to connect to the Simple Pro?

If it's something that can display a list of the commands sent and the responses received, then the version info should be in the first (roughly) response line.  It'll look something like:

```
{"r":{"sys":{"fb":100.26,"fbs":"100.26-10-g1ff7","fbc":"settings_default.h","fv":0.99, ...
```

That example line indicates firmware build 100.26.

Does that help? :smile:

giseburt, 2018-11-04:
Some firmwares have the heat bed disabled by default. Add this line to the
first few lines of the start.gcode to enable it:

M100.1 ({he3e:t}) ; enable heater 3 - the heat bed

From then the normal Marlin heater commands for the heat bed should start
working.

-Rob

pr3sidentspence, 2018-11-04:
@giseburt Yes, I've done that. It returns: Recv: {"r":{"he3e":true},"f":[1,0,18]}

@justinclift I've seen it return `g2core-100.21-81-gd770`, but I have no idea how that compares to Printrbot's final release binary: `G2core170411.bin`. And I'm not sure I am running that one, I have a sneaky feeling that I'm running a post-enabling-gcode but pre-heated-bed announcement firmware.

giseburt, 2018-11-04:
The latest release is 101.03, found here:
https://github.com/synthetos/g2/releases/tag/101.03

There was a bug where heater 3 (the heat bed) wasn’t getting the Marlin
commands properly. I’m not sure what version it was fixed in, but it is
certainly fixed in the latest version.

-Rob

justinclift, 2018-11-06:
Awesome. :smile:

PR #262 (2017-04-05): Work on PrintrBot simple heat bed support
This pull request is to fix #261
commit fd4f1d79 (2017-04-05): Merged dev-261-simple-heat-bed - fixes #261