synthetos/g2 #263: Error setting working coordinates G10 L20 commands not working

amx1, 2017-04-08:
I checked out the latest version 100.24. Setting the working coordinates doesn't work anymore. G10 L20 .. commands have no effect. I did a rollback. It was still ok in 100.21, but since 100.22 it is not working.

Configuration: ArduinoDue, Othermill - gShield

Best AMX1

amx1, 2017-04-09:
Sorry 100.22 is ok too, but 100.23 and 100.24 are not working. I compiled a small test case which simulates my typical start sequence for milling.

1) machine homing
2) set workpeace zero point of G54
3) start milling 
---------------------------------------
gcode:
(Simulate Homing Goto machine zero)
G21
G53 G0 X0 Y0 Z0
(clear also work zero - coordinate system 1 G54)
G10 L20 P1 X0 Y0 Z0
(wait 1 second)
G4 P1
(goto workpeace zero point: e.g. 50 50)
G0 X50 Y50
(set work zero - coordinate system 1)
G54 G10 L20 P1 X0 Y0 Z0
(wait 1 second)
G4 P1
(goto 50 50 in work coordinate system)
G1 X50 Y50 F500
(goto work home)
G0 X0 Y0 Z0
(wait 1 second)
G4 P1
(goto machine zero)
G53 G0 X0 Y0 Z0
-------------------------------------

amx1, 2017-04-10:
@aldenhart - Thank you very much for your quick response! I checked it and everything is working fine now.

aldenhart, 2017-04-10:
Thanks for the useful post and checking it out. This kind of stuff usually gets surfaced in our regression tests. I'm not sure what happened there, but wilI take a look at how this got through.

I will post a new build number and release for this later and close the issue once that's done.

closed 2017-04-10 by amx1