synthetos/g2 #267: Incorrect settings for older Printrbots

giseburt, 2017-04-15:
Some of the settings files for the older printrbots have become somewhat stale.

Following are what the new settings should be, in the form of gcode "start" files:

# Printrbot Plus
```gcode
M100.1 ({{xhi:4}})						; remap x axis endstop to input 4 in firmware
M100.1 ({{yhi:1}})						; remap y axis endstop to input 1 in firmware
M100.1 ({{5ma:4}}) 						; remap 5 to B
M100.1 ({{3ma:1}}) 						; map 3 to Y
M100.1 ({{3tr:40.64}}) 						; set the 3 travel-per-rev to 40.64
M100.1 ({{3po:1}})						; set motor 3 polarity to 1
M100.1 ({{4po:1}})						; set motor 4 polarity to 1
M100.1 ({{di5fn:4}})						; map z sensor to input 5
M100.1 ({{xtm:250}})						; change x axis to 250mm 
M100.1 ({{ytm:250}})						; change y axis to 250mm
M100.1 ({{ztm:196}})						; change z axis to 196mm
M100.1 ({{xjm:6000}}) 						; xjm is jerk maximum for x axis
M100.1 ({{xjh:6000}})						; xjh is jerk high speed for x axis
M100.1 ({{yjm:6000}})						; yjm is jerk maximum for y axis
M100.1 ({{yjh:6000}})						; yjh is jerk high speed for y axis
```

*Probing routine for `G29`*
```gcode
M100 ({{"_leds":3}})						; turn on LEDs to blue
G1 X0 Y245 Z6 F20000						; 1st probe location 
G38.2 Z-10 F200
G1 Z5 F20000
M100 ({{"_leds":5}})						; turn on LEDs to yellow
G1 X225 Y125 F20000						; 2nd probe location
G38.2 Z-10 F200
G1 Z5 F20000
M100 ({{"_leds":6}})						; turn on LEDs to magenta
G1 X0 Y5 F20000							; 3rd probe location
G38.2 Z-10 F200
G1 Z5 F20000
M100 ({{"_leds":3}})						; turn on LEDs to blue
M100 ({{"tram":1}})
```

# Printrbot Play
```gcode
M100.1 ({{xhi:4}})					; remap x axis endstop to input 4 in firmware
M100.1 ({{yhi:1}})					; remap y axis endstop to input 1 in firmware
M100.1 ({{5ma:4}}) 					; remap 5 to B
M100.1 ({{3ma:1}}) 					; map 3 to Y
M100.1 ({{3tr:40.64}}) 					; set the 3 travel-per-rev to 40.64
M100.1 ({{3po:1}}) 					; set motor 3 polarity to 1
M100.1 ({{4po:1}})					; set motor 4 polarity to 1
M100.1 ({{di5fn:4}})					; map z sensor to input 5
M100.1 ({{xjm:6000}}) 					; xjm is jerk maximum for x axis
M100.1 ({{xjh:6000}})					; xjh is jerk high speed for x axis
M100.1 ({{yjm:6000}})					; yjm is jerk maximum for y axis
M100.1 ({{yjh:6000}})					; yjh is jerk high speed for y axis
```

*Probing routine for `G29`*
```gcode
M100 ({{"_leds":3}})					; turn on LEDs to blue
G1 X0 Y95 Z6 F20000					; 1st probe location - if you have y axis upgrade change y value to Y200
G38.2 Z-10 F200
G1 Z5 F20000
M100 ({{"_leds":5}})					; turn on LEDs to yellow
G1 X70 Y45 F20000					; 2nd probe location - if you have y axis upgrade change y value to Y100
G38.2 Z-10 F200
G1 Z5 F20000
M100 ({{"_leds":6}})					; turn on LEDs to magenta
G1 X0 Y5 F20000						; 3rd probe location - remain the same value for y axis upgrade
G38.2 Z-10 F200
G1 Z5 F20000
M100 ({{"_leds":3}})					; turn on LEDs to blue
M100 ({{"tram":1}})
```

# Older Printrbot Simple
```gcode
M100.1 ({{di5fn:4}})						; map z sensor to input 5
M100.1 ({{xhi:4}})						; remap x axis endstop to input 4 in firmware
M100.1 ({{yhi:1}})						; remap y axis endstop to input 1in firmware
M100.1 ({{5ma:4}}) 						; remap 5 to B
M100.1 ({{3ma:1}}) 						; map 3 to Y
M100.1 ({{3tr:40.64}}) 						; set the 3 travel-per-rev to 40.64
M100.1 ({{xtm:152}})						; change x axis to 152mm - if you have x axis upgrade change this value to 254
M100.1 ({{ytm:152}})						; change y axis to 152mm
M100.1 ({{ztm:152}})						; change z axis to 152mm - if you have z axis upgrade change this value to 240
M100.1 ({{xjm:6000}})						; xjm is jerk maximum for x axis
M100.1 ({{xjh:6000}})						; xjh is jerk high speed for x axis
M100.1 ({{yjm:6000}})						; yjm is jerk maximum for y axis
M100.1 ({{yjh:6000}}) 						; yjh is jerk high speed for y axis
```

*Probing routine for `G29`*
```gcode
M100 ({{"_leds":3}})						; turn on LEDs to blue
G1 X0 Y145 Z6 F10000						; 1st probe location - remains the same value for x axis upgrade
G38.2 Z-10 F200
G1 Z5 F8000
M100 ({{"_leds":5}})						; turn on LEDs to yellow
G1 X140 Y65 F20000						; 2nd probe location - if you have x axis upgrade change y value to Y245
G38.2 Z-10 F200
G1 Z5 F8000
M100 ({{"_leds":6}})						; turn on LEDs to magenta
G1 X0 Y10 F20000						; 3rd probe location - remain the same value for x axis upgrade
G38.2 Z-10 F200
G1 Z5 F8000
M100 ({{"_leds":3}})						; turn on LEDs to blue
M100 ({{"tram":1}})
```

giseburt, 2017-04-18:
Notes:

* For the Simple with the extension kit: there's no need for the `xtm` and `ztm` - those are only needed if the soft limits are turned on (they aren't by default) or if a GUI reads those (so far, none for 3DP do)
* For any machine with a heat bed, `M100.1 ({he3e:t})` needs to be issued to enable the heat-bed temperature control.
  * If it's enabled and the heat-bed isn't connected it will cause an alarm. So the default is not enabled.
  * If it's **not enabled** and the heat bed is used (set to a temperature) it will cause an alarm.

closed 2017-04-19 by giseburt

Barafu, 2017-04-22:
In "Older printrbot simple" you have a comment "; 2nd probe location - if you have x axis upgrade change y value to Y245" That seems to be an error - shoul d not it be "change to X". And you can't probe next to the far X end on Printrbot simple, because there is a hole in the table and the probe will fall through and crush into the table. Y axis exafgerates mechanical impurities close to the Y0. Finally, oldest Printrbot simple metal needlessly strains Y belt to reach Y0. 
So, I'd probe upgraded X table at X10 Y135, X10, Y10, and X210 Y 10.

BDFife, 2018-04-06:
@Barafu, to fix the Y-slamming, try setting slowing down the y homing speed with the $ysv command. A value of 1000 seems to work well for me.

commit 0a0e392c (2017-04-19): Merge pull request #268 from synthetos/dev-267-pb-settings

Updates to all Printrbot configurations. Fixes #267.