synthetos/g2 #269: Win10 AS7 compling error

shczm, 2017-04-20:
Hi, 
when compiling g2, error info:
Shell Utils Path C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
		C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe -C "C:\github\g2\g2\g2core" -f "Makefile" BOARD=gShield CONFIG=ShapeokoDualY COLOR=0 VERBOSE=1 
		make: Entering directory 'C:/github/g2/g2/g2core'
		"Found that we're in Atmel Studio"
		process_begin: CreateProcess(NULL, git describe --abbrev=4 --dirty --always --tags, ...) failed.
		make: ../Motate/MotateProject/motate/Motate.mk:302: pipe: No such file or directory
		process_begin: CreateProcess(NULL, git describe --abbrev=0 --tags, ...) failed.
		make: ../Motate/MotateProject/motate/Motate.mk:303: pipe: No such file or directory

Can anyone help me？
Thanks

giseburt, 2017-04-20:
It appears that you are missing Git for Windows. Instructions [here](https://github.com/synthetos/g2/wiki/Compiling-G2-on-Windows-10-and-Atmel-Studio-7#git-for-windows).

If you use GitHub Desktop to clone the repo you still need to install git for Windows. Also, we don't support compiling from a zip-downloaded copy of the code. (It's possible, but just not a priority. We always welcome pull requests.) 

We have a note in our [troubleshooting page](https://github.com/synthetos/g2/wiki/Troubleshooting) for this.

shczm, 2017-04-20:
Yes, afte installed the Git  for windoes, compling is successful.
Thank you Rob  !

closed 2017-04-20 by shczm