synthetos/g2 #270: Error: trim_d compensation is not applied to first 20 (or so) movements

Barafu, 2017-04-23:
I use 3D printer Printrbot with G2 board and firmware 100.19 that was preinstalled by Printrbot. Sorry if this is already solved, but I can not install new firmawre because that board is not supported by `bossac`. This is part of my startup script that does tram compensation. 
```
G17 G21 G90
G28.3 A0
G28.2 X0 Y0
G1 X10 Y130 Z6 F10000
G38.2 Z-10 F200
G1 Z5 F8000
G1 X210 Y65 F20000
G38.2 Z-10 F200
G1 Z5 F8000
G1 X10 Y10 F20000
G38.2 Z-10 F200
G1 Z5 F8000
M100 ({"tram":1})
```
When I start a job, the first couple of dozens of lines are not affected by tram compensation, and tool howers above the Z0. After that, compensation starts to work and tool visibly shifts down to the real Z0  and the rest of the job is properly positioned. I did try to add a pause after M100 ({"tram":1}), but it had no effect. However, when I added something silly like 
```
G1 X20 Y10 Z10 F10000
G1 X20 Y20 Z5 F10000
G1 X10 Y20 Z10 F10000
G1 X10 Y10 Z5 F10000
G1 X20 Y10 Z10 F10000
G1 X20 Y20 Z5 F10000
G1 X10 Y20 Z10 F10000
G1 X10 Y10 Z5 F10000
G1 X20 Y10 Z10 F10000
G1 X20 Y20 Z5 F10000
G1 X10 Y20 Z10 F10000
G1 X10 Y10 Z5 F10000
G1 X20 Y10 Z10 F10000
G1 X20 Y20 Z5 F10000
G1 X10 Y20 Z10 F10000
G1 X10 Y10 Z5 F10000
G1 X20 Y10 Z10 F10000
G1 X20 Y20 Z5 F10000
G1 X10 Y20 Z10 F10000
G1 X10 Y10 Z5 F10000
```
Then trap compensations work fine right from the start of the job. It is the number of lines that are affected, not the time.

giseburt, 2017-04-24:
This is interesting. I'm looking into it. Thank you.

Barafu, 2017-04-24:
Tell me if you need some testing.

giseburt, 2017-04-24:
BOSSAC should work on that board. It's an Atmel SAM3X8C (also known as a AT91SAM3X8C) chip.

Also, as a workaround, you should be able to put another redundant probe after the tram command:

```gcode
G38.2 Z-10 F200
```

That will prevent it from planning the moves after the probe, and the tram rotation will be applied correct after that.

Background: g2core plans out future moves (up to 46 moves on that board). What's happening is that the tram command (as an M100) is getting planned and then moves after it are getting planned without the rotation matrix, and *then* the tram command gets executed and moves are applied.

This didn't happen in testing since we always had the temperature wait commands after the tram, which held up the planner until the wait (M140 or M190 in Marlin compatibilty, which internally issues an M101) was over.

I'm looking into a more permanent fix.

giseburt, 2017-04-24:
I forgot: the link to flashing g2core from Windows using BOSSAC is here: https://github.com/synthetos/g2/wiki/Flashing-g2core-with-Windows

For OS X it's here: https://github.com/synthetos/g2/wiki/Compiling-g2core-on-Linux-and-OS-X-(command-line)

And Linux is here: https://github.com/synthetos/g2/wiki/Flashing-g2core-with-Linux

We are working on an updated GUI, but those should work for now.

Barafu, 2017-04-25:
I have successfully compiled and installed the new firmware, thanks. If I may, a couple of questions. 

- What is the proper way to set up the difference between Z probe and nozzle? You know, what we did with M212 command. 
- Is it possible to change heat bed into bang_bang mode? I connect my heater with on/off relay for now. It works, but clicks excessively.

Barafu, 2017-04-25:
Update: 
```
G28.2 X0 Y0
G28.2 Z0
G29
G38.2 Z-10 F200
G92 Z0.8
```
seems to work properly.

giseburt, 2017-04-26:
Hi @Barafu ,

`G29` simply runs a built-in script, so if the settings file you used doesn't have that setting it will not do anything. (In Marlin compatibility mode you might not see the warning it kicks out.) See [here](https://github.com/synthetos/g2/blob/edge/g2core/settings/settings_Printrbot_Simple_1403.h#L83-L97) for the default `G29` "script" for the Printrbot Simple 1403.

Did `G29` do anything at all?

Barafu, 2017-04-26:
Yes, it does. Sorry if I was confusing. 
G29 runs the probing cycle and does apply the tram compensation. However, Z0 ends to be in the air , because I do not know how to set the offset between the nozzle and Z probe. All my attempts to do it did shift the print down, but did cancel out the tram compensation. 
My print bed is tilted like 1.5 mm along the Y axis, so the lack of tram compensation is always obvious. Two days ago I did the probing procedure manually and used Gcode build so as the first layer has Z-0.8 coordinates in it. It did work, but somehow it does not work now, after firmware update and switching to OctoPrint. 
Please advise me a proper startup script, and I will test it out.

giseburt, 2017-04-26:
Ok, no problem.

We recommend using the G55 offsets to handle this. This has a few advantages, some of which will be realized once we have the settings saving in flash.

So, at the top of the file, put an `M100.1` that sets the G55 Z offset. Then put a `G55` after the tram command (which is in the `G29`, so after that). From then on the `G55` offsets will be applied to all incoming commands.

So, to show what that looks like (in Marlin flavor gcode):

```gcode
M100.1 ({g55z:-0.8})
; temperature commands here
; homing commands here
G29        ; run auto bed leveling
G55        ; apply offsets
; temperature waits here
```

One additional note, if you're entering this into Cura (and maybe other senders, test to find out), you'll need to double the `{` and `}` so it keeps them intact. So a more complete example for Cura would look like this:

```gcode
M100.1 ({{g55z:-1.85}})
M140 S{material_bed_temperature}
M104 S{material_print_temperature}
G21              ; metric values
G90              ; absolute positioning
M82              ; set extruder to absolute mode
M106             ; start with the fan on for filament cooling
G28 X0 Y0        ; move X/Y to min endstops
G28 Z0           ; move Z to min endstops
G29              ; run auto bed leveling
G55              ; apply offsets
G1 Z15.0 F9000   ; move the platform to Z=15mm
M190 S{material_bed_temperature}
M109 S{material_print_temperature}
G92 E0           ; zero the extruded length
G1 F200 E10      ; extrude 10mm of feed stock
G92 E0           ; zero the extruded length again
```
The `{material_bed_temperature}` will be replaced by Cura to be that setting. The `M100.1 ({{g55z:-1.85}})` will be replaced to be `M100.1 ({g55z:-1.85})` in the output gcode.

Hopefully this helps.

-Rob

Barafu, 2017-04-28:
Thanks a lot. I had been printing for 2 days with the start script you provided and see no problem.