synthetos/g2 #272: 100.26 Complie Errors in Atmel Studio 7 using Github...

hublerdesignworks, 2017-04-27:
[image]

Help...

r89m, 2017-04-29:
Just had the same issue. Seems it was removed (not sure if intentionally) in c7c8728327ddff132712849f29cfef6784adfe0d 

Running `git revert c7c8728327ddff132712849f29cfef6784adfe0d`, cleaning and rebuilding the solution appeared to work

hublerdesignworks, 2017-04-30:
That did the job. Thanks!

giseburt, 2017-04-30:
That's my fault. It came in a pull request and I missed that that file got
deleted. I'll fix that in the next release.

Thank you both for finding this and figuring it out!

cmcgrath5035, 2017-11-20:
It appears that Riley fixed this issue in  Repository on 9/16/2017. I'll leave this open for Rob or Riley to verify

justinclift, 2018-04-12:
Closing this, as it was added back in a while ago by Riley with e9aba647b9ae0a59b048c14bb0d0fbe423137e44, as mentioned by @cmcgrath5035. :smile:

closed 2018-04-12 by justinclift