synthetos/g2 #274: Question: SAM3X8E UART instead of usb for Xio

n81641, 2017-05-02:
To use the UART0 on the SAM3X8E
Besides changing to
```
#define XIO_HAS_USB 0
#define XIO_HAS_UART 1
```
in <board>-pinout.h

What else needs to be done?

giseburt, 2017-05-03:
Hi @n81641 ,

I believe that your first instinct to put it here was correct. Motate offers the UART option, but the selection and usage is done here in the g2 repo.

You are also correct that In order to use the UART, XIO is the part that needs configured.

The board files have [`board_xio.h`](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/board_xio.h) and [`board_xio.cpp`](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/board_xio.cpp) whose job it is to provide the `Serial` object for when it is used ( [1](https://github.com/synthetos/g2/blob/edge/g2core/xio.cpp#L1289-L1299) [2](https://github.com/synthetos/g2/blob/edge/g2core/xio.cpp#L1311-L1313) [3](https://github.com/synthetos/g2/blob/edge/g2core/xio.cpp#L1341-L1343) ).

Ok, so the more direct answer to your question, now that we're done with the background portion of the answer 😄 , is that you need to provide the things used in [these lines](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/board_xio.h#L61-L64):

```c++
#if XIO_HAS_UART
#include "MotateUART.h"
extern Motate::UART<Motate::kSerial_RXPinNumber, Motate::kSerial_TXPinNumber, Motate::kSerial_RTSPinNumber, Motate::kSerial_CTSPinNumber> Serial;
#endif
```

So [these lines](https://github.com/synthetos/g2/blob/80bfa3f6c8d6fe9a18e9f4a58974de6dd274ae60/g2core/board/ArduinoDue/gShield-pinout.h#L92-L95) in the (currently deleted but soon to be returned, see #272 ) `gShield-pinout.h` file need to be corrected:

```c++
pin_number kSerial_RXPinNumber = 0;
pin_number kSerial_TXPinNumber = 1;
//    pin_number kSerial_RTSPinNumber                      =  8;   // added later
//    pin_number kSerial_CTSPinNumber                      =  9;   // added later
```

(*Note that in the Arduino Due files we handle the naming and numbering backward, to honor the numbers on the silk screen. That makes this more difficult than on our other boards.*)

You are in luck that we have support for soft RTS/CTS now, since the RTS line on the Due is wired to the reset line you don't want to use it. 

So, along with making `XIO_HAS_UART` have the value `1`, you need to uncomment the `kSerial_RTSPinNumber` and `kSerial_CTSPinNumber` definitions and assign them any two available pins. 

The firmware should then compile with the UART functioning. Remember you _must_ wire up and use the RTS and CTS lines. We use them for flow control as well as to determine when something is connected and interested in talking. If they're not hooked up and working properly, you will either see nothing happen at all or get erratic behavior as data is dropped when one side or the other has a full buffer.

n81641, 2017-05-03:
Thank you for the detailed answer. 

I added this to the pinout.h:
```
_MAKE_MOTATE_PIN(kSerial_RXPinNumber, 'A',  8);  			  //UART RX  
_MAKE_MOTATE_PIN(kSerial_TXPinNumber, 'A',  9);  			  //UART TX	 
_MAKE_MOTATE_PIN(kSerial_RTSPinNumber, 'C', 15);  			  //UART RTS  
_MAKE_MOTATE_PIN(kSerial_CTSPinNumber, 'C', 13);  			  //UART CTS  
``` 
and this is in motate_pin_assignments.h:

```
pin_number kSerial_RXPinNumber  = 0;
pin_number kSerial_TXPinNumber  = 1;
pin_number kSerial_RTSPinNumber = 8;  // added later
pin_number kSerial_CTSPinNumber = 9;  // added later
```
It was already there, and I don't care if the pin numbers match anything.

I get a couple warnings about the 'C' port: (67u)
```
../Motate/MotateProject/motate/Atmel_sam_common/SamPins.h: In member function 'constexpr Pio* const Motate::PortHardware<portLetter>::rawPort() const [with unsigned char portLetter = 67u]':

../Motate/MotateProject/motate/Atmel_sam_common/SamPins.h:177:9: warning: control reaches end of non-void function [-Wreturn-type]
```
Compiles OK.  Will test soon.

n81641, 2017-05-03:
OK, Warnings are not good, need to sort _MAKE_MOTATE_PIN pin assignments.

n81641, 2017-05-03:
Changed to these:
``
_MAKE_MOTATE_PIN(kSerial_RTSPinNumber, 'C', 15);  			  //UART RTS  
_MAKE_MOTATE_PIN(kSerial_CTSPinNumber, 'C', 13);  			  //UART CTS
``
Now compiles without warnings.  
I uploaded, but it still uses the USB for communication.  Am I mistaken in thinking the XIO would use this UART for the main command/report channel?   Have I simply created another auxiliary communication port?

giseburt, 2017-05-11:
You can use either in this configuration. UART can be the primary channel.

If you wish you can have this in your setting file:

```c++
#define XIO_UART_MUTES_WHEN_USB_CONNECTED 1 // Mute the UART when USB
connects
```

And when a USB connection is opened (not just the physical connection, but
the software opening the port), the UART will get a {"muted":true}, and
when it closes it'll send a {"muted":false}. When muted the UART will not
get any status reports, etc.

However, normally (without XIO_UART_MUTES_WHEN_USB_CONNECTED) both channels
will see all output, and both channels are valid inputs. So be careful not
to mix them up and have two things running protocols where they count
responses or such.

n81641, 2017-05-11:
Got it, Thank you very much!

closed 2017-05-11 by n81641