synthetos/g2 #277: Question: how is it possible to include coordinate system offsets in status report?

gilbertoverg, 2017-05-11:
Hi, i'm trying to include coordinate system offsets in status report, i've changed STATUS_REPORT_DEFAULTS to:
```
#define STATUS_REPORT_DEFAULTS "line","posx","posy","posz","posa","feed","vel","unit","coor","dist","stat","g54","g55","g56","g57","g58","g59"
```
but this is what i'm getting:
```
2017-05-11T14:24:49.795Z Opening: /dev/tty.usbmodem1411
2017-05-11T14:24:49.858Z Port opened
2017-05-11T14:24:49.869Z RX: {"r":{"fv":0.99,"fb":100.26,"fbs":"","fbc":"settings_my.h","hp":3,"hv":0,"id":"0084-d639-0084-08c","msg":"SYSTEM READY"},"f":[1,0,1]}
2017-05-11T14:24:50.859Z TX: G90
2017-05-11T14:24:50.859Z TX: F1000
2017-05-11T14:24:50.860Z RX: {"r":{},"f":[1,0,4]}
2017-05-11T14:24:50.861Z RX: {"r":{},"f":[1,0,6]}
2017-05-11T14:24:51.113Z RX: {"g59":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}}
2017-05-11T14:24:52.860Z TX: G1 X30
2017-05-11T14:24:52.861Z RX: {"r":{},"f":[1,0,7]}
2017-05-11T14:24:53.118Z RX: {"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}}
2017-05-11T14:24:53.369Z RX: {"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}}
2017-05-11T14:24:53.619Z RX: {"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}}
2017-05-11T14:24:53.870Z RX: {"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}}
2017-05-11T14:24:54.120Z RX: {"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}}
2017-05-11T14:24:54.370Z RX: {"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}}
2017-05-11T14:24:54.621Z RX: {"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}}
2017-05-11T14:24:54.868Z RX: {"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}}
2017-05-11T14:25:02.863Z Port closed
```
am i doing this right?

gilbertoverg, 2017-05-16:
I've tried changing STATUS_REPORT_DEFAULTS to:
```
#define STATUS_REPORT_DEFAULTS "line","posx","posy","posz","posa","ofsx","ofsy","ofsz","ofsa","feed","vel","unit","coor","dist","path","plan","stat","homx","homy","homz","g54x","g54y","g54z","g54a","g55x","g55y","g55z","g55a","g56x","g56y","g56z","g56a","g57x","g57y","g57z","g57a","g58x","g58y","g58z","g58a","g59x","g59y","g59z","g59a"
```
but i've received this error from the compiler:
```
Compiling cpp report.cpp
    -> build/Ebe-gShield/./report.o 
report.cpp: In function 'void sr_init_status_report()':
report.cpp:190:84: error: too many initializers for 'char [39][7]'
     char sr_defaults[NV_STATUS_REPORT_LEN][TOKEN_LEN+1] = { STATUS_REPORT_DEFAULTS };
                                                                                    ^
```
so i've increased NV_BODY_LEN in config.h from 40 to 50 and it compiles ok, but when i try to send gcode to the machine i'm getting a status report every millisecond and some reports are missing:
```
2017-05-16T06:22:47.652Z Opening: /dev/tty.usbmodem1411
2017-05-16T06:22:47.708Z Port opened
--> MISSING: {"r":{"fv":0.99,"fb":100.26,"fbs":"","fbc":"settings_ebe.h","hp":3,"hv":0,"id":"0084-d639-0084-08c","msg":"SYSTEM READY"},"f":[1,0,1]}
2017-05-16T06:22:48.709Z TX: G90
2017-05-16T06:22:48.709Z TX: F6000
--> MISSING: {"r":{},"f":[1,0,4]}
--> MISSING: {"r":{},"f":[1,0,6]}
2017-05-16T06:22:48.712Z RX: {"sr":{"line":0,"posx":0,"posy":0,"posz":0,"posa":0,"ofsx":0,"ofsy":0,"ofsz":0,"ofsa":0,"feed":0,"vel":0,"unit":0,"coor":0,"dist":0,"path":0,"plan":0,"stat":1,"homx":0,"homy":0,"homz":0,"g54x":0,"g54y":0,"g54z":0,"g54a":0,"g55x":0,"g55y":0,"g55z":0,"g55a":0,"g56x":0,"g56y":0,"g56z":0,"g56a":0,"g57x":0,"g57y":0,"g57z":0,"g57a":0,"g58x":0,"g58y":0,"g58z":0,"g58a":0}}
2017-05-16T06:22:48.714Z RX: {"sr":{"feed":6000}}
2017-05-16T06:22:50.710Z TX: G1 X10
--> MISSING: {"r":{},"f":[1,0,7]}
--> MISSING: {"qr":47}
2017-05-16T06:22:50.713Z RX: {"sr":{"feed":0,"stat":5}}
--> feed = 0 ?
2017-05-16T06:22:50.745Z RX: {"sr":{"feed":6000,"vel":0}}
2017-05-16T06:22:50.745Z RX: {"sr":{"vel":0.01}}
2017-05-16T06:22:50.746Z RX: {"sr":{"vel":0.01}}
2017-05-16T06:22:50.748Z RX: {"sr":{"vel":0.03}}
2017-05-16T06:22:50.749Z RX: {"sr":{"vel":0.05}}
2017-05-16T06:22:50.751Z RX: {"sr":{"vel":0.09}}
2017-05-16T06:22:50.752Z RX: {"sr":{"vel":0.13}}
2017-05-16T06:22:50.754Z RX: {"sr":{"vel":0.19}}
2017-05-16T06:22:50.755Z RX: {"sr":{"vel":0.26}}
2017-05-16T06:22:50.758Z RX: {"sr":{"vel":0.35}}
2017-05-16T06:22:50.758Z RX: {"sr":{"vel":0.46}}
2017-05-16T06:22:50.760Z RX: {"sr":{"vel":0.58}}
2017-05-16T06:22:50.761Z RX: {"sr":{"vel":0.73}}
2017-05-16T06:22:50.763Z RX: {"sr":{"vel":0.89}}
2017-05-16T06:22:50.764Z RX: {"sr":{"vel":1.08}}
2017-05-16T06:22:50.767Z RX: {"sr":{"vel":1.29}}
2017-05-16T06:22:50.768Z RX: {"sr":{"vel":1.53}}
2017-05-16T06:22:50.769Z RX: {"sr":{"vel":1.79}}
2017-05-16T06:22:50.771Z RX: {"sr":{"vel":2.08}}
2017-05-16T06:22:50.772Z RX: {"sr":{"vel":2.4}}
2017-05-16T06:22:50.774Z RX: {"sr":{"vel":2.74}}
2017-05-16T06:22:50.776Z RX: {"sr":{"vel":3.12}}
2017-05-16T06:22:50.777Z RX: {"sr":{"vel":3.52}}
2017-05-16T06:22:50.778Z RX: {"sr":{"vel":3.96}}
2017-05-16T06:22:50.780Z RX: {"sr":{"vel":4.43}}
2017-05-16T06:22:50.781Z RX: {"sr":{"vel":4.92}}
2017-05-16T06:22:50.782Z RX: {"sr":{"posx":0.001,"vel":5.46}}
2017-05-16T06:22:50.785Z RX: {"sr":{"vel":6.02}}
2017-05-16T06:22:50.786Z RX: {"sr":{"vel":6.62}}
2017-05-16T06:22:50.787Z RX: {"sr":{"vel":7.25}}
2017-05-16T06:22:50.789Z RX: {"sr":{"vel":7.92}}
2017-05-16T06:22:50.790Z RX: {"sr":{"vel":8.62}}
2017-05-16T06:22:50.793Z RX: {"sr":{"posx":0.002,"vel":9.36}}
2017-05-16T06:22:50.794Z RX: {"sr":{"vel":10.13}}
2017-05-16T06:22:50.795Z RX: {"sr":{"vel":10.94}}
2017-05-16T06:22:50.797Z RX: {"sr":{"vel":11.79}}
2017-05-16T06:22:50.798Z RX: {"sr":{"posx":0.003,"vel":12.67}}
2017-05-16T06:22:50.800Z RX: {"sr":{"vel":13.59}}
2017-05-16T06:22:50.801Z RX: {"sr":{"vel":14.54}}
2017-05-16T06:22:50.802Z RX: {"sr":{"posx":0.004,"vel":15.53}}
2017-05-16T06:22:50.803Z RX: {"sr":{"vel":16.56}}
[... 11 lines cut ...]
2017-05-16T06:22:50.822Z RX: {"sr":{"posx":0.012,"vel":31.71}}
2017-05-16T06:22:50.823Z RX: {"sr":{"vel":33.21}}
2017-05-16T06:22:50.826Z RX: {"sr":{"posx":0.014,"vel":34.73}}
2017-05-16T06:22:50.827Z RX: {"sr":{"vel":36.29}}
2017-05-16T06:22:50.828Z RX: {"sr":{"posx":0.016,"vel":37.88}}
2017-05-16T06:22:50.830Z RX: {"sr":{"vel":39.5}}
2017-05-16T06:22:50.831Z RX: {"sr":{"posx":0.018,"vel":41.15}}
2017-05-16T06:22:50.833Z RX: {"sr":{"posx":0.019,"vel":42.84}}
2017-05-16T06:22:50.834Z RX: {"sr":{"posx":0.02,"vel":44.56}}
2017-05-16T06:22:50.836Z RX: {"sr":{"posx":0.021,"vel":46.3}}
2017-05-16T06:22:50.837Z RX: {"sr":{"posx":0.022,"vel":48.08}}
2017-05-16T06:22:50.840Z RX: {"sr":{"posx":0.023,"vel":49.88}}
2017-05-16T06:22:50.841Z RX: {"sr":{"posx":0.025,"vel":51.71}}
2017-05-16T06:22:50.842Z RX: {"sr":{"posx":0.026,"vel":53.57}}
2017-05-16T06:22:50.843Z RX: {"sr":{"posx":0.027,"vel":55.46}}
2017-05-16T06:22:50.845Z RX: {"sr":{"posx":0.029,"vel":57.37}}
2017-05-16T06:22:50.846Z RX: {"sr":{"posx":0.03,"vel":59.31}}
2017-05-16T06:22:50.847Z RX: {"sr":{"posx":0.032,"vel":61.27}}
2017-05-16T06:22:50.850Z RX: {"sr":{"posx":0.033,"vel":63.25}}
2017-05-16T06:22:50.851Z RX: {"sr":{"posx":0.035,"vel":65.26}}
2017-05-16T06:22:50.853Z RX: {"sr":{"posx":0.037,"vel":67.29}}
...
...
```
What am I doing wrong?

justinclift, 2018-04-12:
Oops, this looks like it was well and truly missed.

@gilbertoverg Did you get it figured out?

gilbertoverg, 2018-04-14:
I'm still not able to get this working properly, I tried changing NV_STATUS_REPORT_LEN
```
#define NV_STATUS_REPORT_LEN 50
```
and adding lines in config_app.cpp
```
{ "","se40",_fp, 0, tx_print_nul, get_int, set_int,&sr.status_report_list[40],0 },
{ "","se41",_fp, 0, tx_print_nul, get_int, set_int,&sr.status_report_list[41],0 },
{ "","se42",_fp, 0, tx_print_nul, get_int, set_int,&sr.status_report_list[42],0 },
{ "","se43",_fp, 0, tx_print_nul, get_int, set_int,&sr.status_report_list[43],0 },
{ "","se44",_fp, 0, tx_print_nul, get_int, set_int,&sr.status_report_list[44],0 },
{ "","se45",_fp, 0, tx_print_nul, get_int, set_int,&sr.status_report_list[45],0 },
{ "","se46",_fp, 0, tx_print_nul, get_int, set_int,&sr.status_report_list[46],0 },
{ "","se47",_fp, 0, tx_print_nul, get_int, set_int,&sr.status_report_list[47],0 },
{ "","se48",_fp, 0, tx_print_nul, get_int, set_int,&sr.status_report_list[48],0 },
{ "","se49",_fp, 0, tx_print_nul, get_int, set_int,&sr.status_report_list[49],0 },
```
but with my STATUS_REPORT_DEFAULTS
```
#define STATUS_REPORT_DEFAULTS "line","posx","posy","posz","posa","ofsx","ofsy","ofsz","ofsa","feed","vel","unit","coor","dist","admo","path","plan","stat","homx","homy","homz","frmo","g54x","g54y","g54z","g54a","g55x","g55y","g55z","g55a","g56x","g56y","g56z","g56a","g57x","g57y","g57z","g57a","g58x","g58y","g58z","g58a","g59x","g59y","g59z","g59a"
```
sometimes some reports are missing, for example:
```
TX: F6000
RX: {"r":{},"f":[1,0,6]}
```