synthetos/g2 #281: "Arduino Due" doesn't change to "TinyG V2" after flashing

pufamuf, 2017-06-17:
Hi everyone, I just flashed my Arduino Due R3 with the "Due+gShield V078.03" bin found [here](http://synthetos.github.io/g2/binaries/TinyG2_Due-edge-078.03-default.bin).

My arduino was on COM10, and thus I did the following stuff in Cmd.exe, and the installation was sucessfull. 

> C:\Users\User\Desktop\arduino-flash-tools-master\arduino\bossac\bin>bossac.exe -
> -port=COM10 -U false -e -w -v -b binfile.bin -R
> Erase flash
> Write 125860 bytes to flash
> [==============================] 100% (492/492 pages)
> Verify 125860 bytes of flash
> [==============================] 100% (492/492 pages)
> Verify successful
> Set boot flash true
> CPU reset.

However, the device manager still says I have an "Arduino Due Programming Port (COM10)" connected.

Any advice?

Thank you

pufamuf, 2017-06-17:
Okay I have solved the issue now.
Something that the OSX guide mentions which the Windows guide does not, is that you need to disconnect the USB cable and plug it into the other port on Arduino due.

closed 2017-06-17 by pufamuf

irshadmgs011, 2018-09-15:
> Okay I have solved the issue now.
> Something that the OSX guide mentions which the Windows guide does not, is that you need to disconnect the USB cable and plug it into the other port on Arduino due.

do you compile g2core firmware???

justinclift, 2018-09-15:
@irshadmgs011 Out of curiosity, are you trying this too and hitting some kind of problem? :smile:

irshadmgs011, 2018-09-16:
My illusion was to select the board from the compiler or do it by editing the firmware

irshadmgs011, 2018-09-16:
It was my fault, I mean, I was compiling it with the Atmel studio on Windows 7,
and my english is so weak🤝😭

justinclift, 2018-09-16:
No worries.  Both will improve with continued practise. :smile:

irshadmgs011, 2018-09-16:
Can you give your email id???

justinclift, 2018-09-16:
Errr, what for? :smile: 

My email address shows on my profile page, if that helps?