synthetos/g2 #282: Homing sequence not halted by endstop input

JeppeIbsen, 2017-06-23:
Hi, I have experienced a couple of times where the endstop trigger failed to stop the homing movement in X and Y, causing a carriage crash. The problem disappears after a board reset.

The move starts by homing in the wrong direction, and after a few milimeters it changes direction and ignores endstop input. 

I have however not been able reproduce the fault, but it causes me to be very nervous every time I home the machine. I would just mention it here, perhaps other people have experienced the same.

Im using a arduino due with mechanical endstops and latest release of g2

lutorm, 2018-03-22:
I have also seen this exact behavior in my old 89.03 build that I've been running for the past two years on a Due. I've been able to feedhold and then home again and it works correctly. I wondered whether it was the homing switch being flaky, but maybe it's a bug. It's quite rare, maybe one time out of 25-50 homes.

Johnr24, 2020-01-24:
I've got the same issue, 
My Y-AXIS is fine, My X-AXIS and Z not.

justinclift, 2020-01-24:
@Johnr24 Which version of the software are you running, and on what hardware? :smile: