synthetos/g2 #283: Enhancement: Pause feature during 3D-printing

JeppeIbsen, 2017-06-23:
Regarding the compatibility with 3D-printing software, like OctoPrint:
The 'Pause print' button causes alarms to be triggered by the g2 firmware, requiring a complete reset of both OctoPrint and the controller. 

Being able to pause and move a FDM 3D-print is a critical feature, as simply using feedhold ( ! ) causes the stationary nozzle to ooze - creating a buldge which requires the entire print to be scrapped (to avoid subsequent collision with the nozzle).

A proper 'Pause' feature, which allows movement of the machine, while maintaining current g-code line, is really needed for g2 to be used for 3D-printing.

giseburt, 2017-06-23:
We have already started work in this direction. As far as OctoPrint, etc, goes, we need it to send a command that will inform the g2core that the host has paused.

Thank you for letting us know that this is wanted! You just gave it a 👍 .

abdrumm, 2018-04-05:
Keep us posted on this one! ;)
Brook