synthetos/g2 #288: Comments on latest dev-168-gquintic

amx1, 2017-07-28:
@aldenhart and @giseburt 

Dear Alden, dear Rob,
I saw that you are working on dev-168-gquintic quite a lot. I am happy with 100.26 on Due and my own Shield using it for milling on a CNC3040. It is really cool and stable, but I was curious and compiled the latest version of dev-168-gquintic with my settings. It is working stable, but I found there is at least one of the latest commits missing: https://github.com/synthetos/g2/pull/264. Merging fixed the problem.

I have some general questions, because you are changing some basic routines of the code quite a lot. Why are you testing a
1) new DDA algorithm?
2) new planner forward strategy?

BTW. The Trinamic Chip is really interesting! 

Some thoughts about Marlin compatibilty. I am using cncjs as frontend. I am a bit struggling with the permanent mode switching from g2 to marlin after some commands e.g. M105. One solution could be a g2core mode with Marlin gcode extensions not switching to marlin communication. slic3r has an export option CNC linux style with very few M codes and the extruder already translated to A-axis. If the controller returns to g2core mode after a "Marlin request" command, it will work out of the box. The other alternative I am thinking about is a pre-processor changing all Marlin codes to M100.1 / M101 syntax which gives you more flexibility. This could also be implemented as a widget in chilipeppr or cncjs.

Best

Axel

giseburt, 2017-07-28:
Hi @amx1 ,

There's a lot of questions in there. I see you're watching pretty closely. ;)

I'll look into that, but the G20 commands should either already be merged into dev-168-gquintic or will be when we merge it to edge.

We are testing a new DDA algorithm. There are actually two separate enhancements that we are testing. One is that we adjusted where in the DDA we encode the time, simplifying the cade, but otherwise it's mathematically identical. The second enhancement is much more experimental, but amounts to having segments (the part that the DDA executes over) go from being constant velocity to linear velocity, where it starts at one velocity and ends at another velocity.

The forward planning strategy that's in there is something that we changed a while back. Basically, instead of backward planning and forward planning the entire buffer, we plan backward until we hit "optimal" (we can't improve it anymore), and then we forward plan just-in-time ahead of the runtime. This way saves a ton of time with replanning forward when the plan changes (which it constantly is, as new moves come in).

We're pretty happy with the Trinamics. They're on our prototype gQuintics.

About Marlin compatibility, it's broken into two parts (outlined [here on our wiki](https://github.com/synthetos/g2/wiki/Marlin-Compatibility) )

- protocol - The M-codes that will never be seen in slicer output, but are what comes from the GUI. These include temperature polling, position polling, SD card setup, etc.
- gcode flavor - These are the M-codes that we get from a slicer, and we wouldn't otherwise see in CNC gcode. IOW, these are the rest of the Marlin-specific M-codes.

We switch those separately. Currently, no Marlin-based GUI is able to speak g2core protocol, so once g2core detects someone speaking Marlin protocol, it switches for the rest of that connection. We haven't built in a way to switch it back. If anyone anticipates needing that, it should be easy to add a JSON command to provide that.

In Marlin protocol mode, if you request something with JSON, it will respond in JSON, so you can still query the machine using g2core JSON. The only thing you really lose is the status reports.

Also note that every Marlin protocol command is a shunt to a JSON command or gcode (or, in some cases, more than one). For example, `M105` is asking for the temperatures. If nothing switches the machine to Marlin protocol you will get those automatically in a status report when they change, or you can request them explicitly with `{he1t:n}` for the extruder and `{he3t:n}` for the heat bed.

For Marlin gcode flavor, the only thing you really lose is distinction between `G0` and `G1`. (There are other subtleties, like how `T` is handled, but they rarely come up in 3D printing.)

-Rob

amx1, 2017-07-30:
Dear Rob,

thank you for the detailed answer. I am very happy with the software. Some feed back from my side why I am curious about the code changes. Up to now I am using it for milling only. I designed a simple break out board with a "Mach compatible pinout"  DB25 connector  for a 5-axis 4,5 A driver board (just some drivers and level shifting). The pitfall was, that I forgot that the driver board I am using, is active low logic, but I there was a hint in some other post. I changed it in step driver for all channels. The complete configuration is g2core on Due with my custom DB25 Due Shield, a driver board of a small german company, a PI 3 with 64bit linux running cncjs of Cheton Wu based on nodejs. The configration really rocks! Thank you for the great software.

Now I am playing a bit with a print head. I don't know yet if my mechanical setup is suffcient. (It is a router) But first of all, I dont want to change the controller and the GUI. The extruder stepper is working already using the A-axis. I am testing the heating functions now. I saw that you are working on a PT with MAX31865 instead of the NTC usually used. That is very nice option, to go beyond the PLA and ABS temperatures. 
 
Best

Axel

siy, 2018-03-04:
Hello,

In the list of commits for this branch I saw comment that this branch now by default points to Motate SAMS70 port. Which MCU it targeting specifically? And where I can find schematic of compatible hardware?

justinclift, 2018-04-12:
@amx1 It's nearly 2 years since you wrote that description up of your self built 5-axis setup.  Is it still going strong, and what have you changed / improved / etc since?

Also... any video of it in action?  Looking for inspiration here.  I'm working on adding a 4th (A) axis first to my Shapeoko 3, with a strong intention to follow it up with a 5th (B) axis not too long after.

My 4th axis (so far) is super dodgy (MDF, screws, and hot glue) but it's just for learning the concepts needed so I'm not really too horrified.  Up to the stage of getting CAD/CAM software to generate appropriate G/Mode now, and have an initial post for Fusion 360 that seems to work in very, very initial testing.

closed 2023-02-16 by ril3y