synthetos/g2 #289: Wrong steps

mrzk, 2017-08-06:
Hello
I've been testing my CNC device for last 2 weeks. I noticed that there are some incorrect movements on CNC device. At first I thought the problem could be the drivers, but after changing them the problem persists. After a few days of trying different things, I ended up counting step pulses generated by g2. I was shocked when I saw for different values of feedrate, there are different number of pulses generated by g2. As far as I know, total count of pulses show how much the stepper motor rotates and the frequency of pulses shows the speed of movement. With this in mind, I suppose that for a 200 step/revolution, microstep of 8 and ball screw of 5mm per revolution, I need 320 pulses to move 1mm but g2 generates 322 pulses for increasing and 320 for decreasing! (feedrate 1000 mm/min)
Any kind of help would be greatly appreciated.

giseburt, 2017-08-06:
What version and branch of the code are you using? There's some wildly experimental stuff in dev-168-gquintic involving steps.

mrzk, 2017-08-07:
The latest version of master branch

giseburt, 2017-08-07:
Try the latest commit of edge. We'll be pushing it to master soon. We only need to go though and clean up the readme and update the changelog.

mrzk, 2017-08-07:
Ok. I will try it and come back with the results

mrzk, 2017-08-07:
The problem still exists :/

giseburt, 2017-08-07:
And just to be thorough, you're counting positive to negative transitions on the step line *while* the corresponding enable line is *low*? And you're using the polarity of the direction line to detect which direction to count the pulse towards?

mrzk, 2017-08-08:
This is the test scenario I'm testing g2 on arduino due with:
1. Reset counter value (cnt = 0)
2. Move 1mm despite moving direction
3. Print and log both rising edge and falling edge counters (The program only counts pulses if enable pin is LOW)
4. Repeat above mentioned steps again in other direction (optional)

mrzk, 2017-08-10:
Does anyone have any idea how can I resolve this problem?

giseburt, 2017-08-10:
Hello again. I haven't been able to reproduce this issue but are still working on it.

I wanted to let you know that we take thi s very seriously. We have gone to great length to ensure that step integrity (both count and timing) is maintained. We even have a virtual encoder built into the code that is constantly checking and (if necessary) correcting the step counts. Only in the latest few commits of the experimental `dev-168-gquintic` branch do we see it actually having to make corrections, and even then [after a highly detailed 30+ hour print](https://twitter.com/giseburt/status/885175895719280644) with that experimental code there is no slippage or positional error. That error has been identified as a rounding bias in the new math. Before we release that branch it will be fixed.

You can monitor the following error detected by the encoders with status reports or poll them via JSON with the keys `_fe1` (for motor 1), `_fe2`, etc. (Note the `_` at the beginning of the name meaning it's for internal or debug use only.) There are a [few other keys related to the encoders](https://github.com/synthetos/g2/blob/dev-168-gquintic/g2core/config_app.cpp#L1242-L1247) as well. That number indicates the fractional portion of steps that were not taken as part of the last move portion. This will rarely be 0.0 but should stay below +-2.0 under normal and correct conditions, since the targets are computed in floating point steps but the step generator can only work in decimal steps. (And at this level we mean "step" as including microsteps, whatever the driver interprets one STEP pulse to indicate.) The reason that number is +-2.0 and not +-1.0 has to do with how our step generation is timed with the computation of the next segment to be stepped, so the error will temporarily be the combined error of the next segment and the current segment, but will end up below +-1.0.

I'll continue to research this, and see if I can reproduce it.

Thank you,
  -Rob

mrzk, 2017-08-11:
Thanks Rob for your hard work on this project. I will test it again tomorrow and record outputs using digital oscilloscope in case they are needed for diagnostics. I didn't know about those internal commands before, I will compare them with real-time output and report back the results

mrzk, 2017-08-12:
This is the result for a movement with feedrate of 1000, micro step of 16 and maximum jerk of 50 million mm/cm3.
`G1 F1000 Y3000  (Initial position is 0)`
Last few lines of status report:
`line:0,posx:0.000,posy:2986.901,posz:0.000,feed:1000.00,vel:1000.00,unit:1,coor:1,dist:0,stat:5,_ts2:1911616.75,_ps2:1911600.75,_cs2:1911584.88,_es2:1911585.00,_xs2:0.00,_fe2:0.12
line:0,posx:0.000,posy:2987.748,posz:0.000,feed:1000.00,vel:1000.00,unit:1,coor:1,dist:0,stat:5,_ts2:1912158.62,_ps2:1912142.62,_cs2:1912126.75,_es2:1912127.00,_xs2:0.00,_fe2:0.25
line:0,posx:0.000,posy:2988.594,posz:0.000,feed:1000.00,vel:1000.00,unit:1,coor:1,dist:0,stat:5,_ts2:1912700.50,_ps2:1912684.50,_cs2:1912668.62,_es2:1912669.00,_xs2:0.00,_fe2:0.38
line:0,posx:0.000,posy:2989.441,posz:0.000,feed:1000.00,vel:1000.00,unit:1,coor:1,dist:0,stat:5,_ts2:1913242.38,_ps2:1913226.38,_cs2:1913210.50,_es2:1913211.00,_xs2:0.00,_fe2:0.50
line:0,posx:0.000,posy:2990.288,posz:0.000,feed:1000.00,vel:1000.00,unit:1,coor:1,dist:0,stat:5,_ts2:1913784.25,_ps2:1913768.25,_cs2:1913752.38,_es2:1913753.00,_xs2:0.00,_fe2:0.62
line:0,posx:0.000,posy:2991.135,posz:0.000,feed:1000.00,vel:1000.00,unit:1,coor:1,dist:0,stat:5,_ts2:1914326.12,_ps2:1914310.12,_cs2:1914294.25,_es2:1914295.00,_xs2:0.00,_fe2:0.75
line:0,posx:0.000,posy:2991.981,posz:0.000,feed:1000.00,vel:1000.00,unit:1,coor:1,dist:0,stat:5,_ts2:1914868.00,_ps2:1914852.00,_cs2:1914836.12,_es2:1914837.00,_xs2:0.00,_fe2:0.88
line:0,posx:0.000,posy:2995.100,posz:0.000,feed:1000.00,vel:999.25,unit:1,coor:1,dist:0,stat:5,_ts2:1916864.25,_ps2:1916848.25,_cs2:1916832.38,_es2:1915364.00,_xs2:-1.80,_fe2:-1468.38
line:0,posx:0.000,posy:2995.945,posz:0.000,feed:1000.00,vel:984.97,unit:1,coor:1,dist:0,stat:5,_ts2:1917404.50,_ps2:1917388.75,_cs2:1917373.00,_es2:1915909.00,_xs2:-5.40,_fe2:-1464.00
line:0,posx:0.000,posy:2996.765,posz:0.000,feed:1000.00,vel:941.18,unit:1,coor:1,dist:0,stat:5,_ts2:1917929.50,_ps2:1917914.50,_cs2:1917899.38,_es2:1916438.00,_xs2:-9.00,_fe2:-1460.50
line:0,posx:0.000,posy:2997.554,posz:0.000,feed:1000.00,vel:859.22,unit:1,coor:1,dist:0,stat:5,_ts2:1918434.75,_ps2:1918421.00,_cs2:1918407.25,_es2:1916949.00,_xs2:-12.60,_fe2:-1458.25
line:0,posx:0.000,posy:2998.237,posz:0.000,feed:1000.00,vel:746.50,unit:1,coor:1,dist:0,stat:5,_ts2:1918871.88,_ps2:1918860.00,_cs2:1918848.00,_es2:1917394.00,_xs2:-15.60,_fe2:-1454.00
line:0,posx:0.000,posy:2998.813,posz:0.000,feed:1000.00,vel:610.15,unit:1,coor:1,dist:0,stat:5,_ts2:1919240.25,_ps2:1919230.62,_cs2:1919220.75,_es2:1917770.00,_xs2:-19.20,_fe2:-1450.75
line:0,posx:0.000,posy:2999.268,posz:0.000,feed:1000.00,vel:462.97,unit:1,coor:1,dist:0,stat:5,_ts2:1919531.25,_ps2:1919523.88,_cs2:1919516.38,_es2:1918069.00,_xs2:-22.80,_fe2:-1447.38
line:0,posx:0.000,posy:2999.598,posz:0.000,feed:1000.00,vel:319.45,unit:1,coor:1,dist:0,stat:5,_ts2:1919742.50,_ps2:1919737.38,_cs2:1919732.25,_es2:1918288.00,_xs2:-25.80,_fe2:-1444.25
line:0,posx:0.000,posy:2999.812,posz:0.000,feed:1000.00,vel:193.51,unit:1,coor:1,dist:0,stat:5,_ts2:1919879.75,_ps2:1919876.50,_cs2:1919873.50,_es2:1918433.00,_xs2:-29.40,_fe2:-1440.50
line:0,posx:0.000,posy:2999.931,posz:0.000,feed:1000.00,vel:96.29,unit:1,coor:1,dist:0,stat:5,_ts2:1919956.12,_ps2:1919954.50,_cs2:1919953.00,_es2:1918516.00,_xs2:-33.00,_fe2:-1437.00
line:0,posx:0.000,posy:2999.983,posz:0.000,feed:1000.00,vel:33.93,unit:1,coor:1,dist:0,stat:5,_ts2:1919989.25,_ps2:1919988.75,_cs2:1919988.12,_es2:1918554.00,_xs2:-36.00,_fe2:-1434.12
line:0,posx:0.000,posy:2999.998,posz:0.000,feed:1000.00,vel:5.35,unit:1,coor:1,dist:0,stat:5,_ts2:1919998.50,_ps2:1919998.25,_cs2:1919998.12,_es2:1918566.00,_xs2:-37.50,_fe2:-1432.12
line:0,posx:0.000,posy:2999.998,posz:0.000,feed:1000.00,vel:-0.02,unit:1,coor:1,dist:0,stat:5,_ts2:1919998.62,_ps2:1919998.62,_cs2:1919998.62,_es2:1918566.00,_xs2:-37.50,_fe2:-1432.62
line:0,posx:0.000,posy:3000.000,posz:0.000,feed:1000.00,vel:0.00,unit:1,coor:1,dist:0,stat:3,_ts2:1920000.00,_ps2:1919998.62,_cs2:1919998.62,_es2:1918566.00,_xs2:-37.50,_fe2:-1432.62`

giseburt, 2017-08-12:
Is this edge? Or dev-168-gquintic?

mrzk, 2017-08-12:
My bad! It was master branch. Let me program edge and check the results again

mrzk, 2017-08-12:
It seems to be working fine. I'll keep testing it to make sure it works well as I ruined one of my client's wooden work because of this problem :(