synthetos/g2 #290: Motor enable signal and first motion

Basti504, 2017-08-12:
Hello,

Currently I am using the TinyG2 (FW 100.26) on the Arduino Due to control a X,Y,Z CNC table. It works great, but today I started using the motor enable pins because I do not want the motors to be energized the whole time.

But I noticed in jogging mode that it looks like when I initiate a motion, the enable and step are set at the same time. The external stepper drive seems to not be able to handle this and will ignore the step and therefore not move while the TinyG2 thinks it does. Starting motion when the drive is enabled works without problem.

I have read trough the manual of my stepper drives (Leadshine AM882) and it states the following:

Remark:
**a) t1: ENA must be ahead of DIR by at least 5µs. Usually, ENA+ and ENA- are NC (not
connected). See “Connector P1 Configurations” for more information.**
b) t2: DIR must be ahead of PUL active edge by 5µs to ensure correct direction;
c) t3: Pulse width not less than 2.5µs;
d) t4: Low level width not less than 2.5µs.

I looked trough the configuration pages, but I cannot find something to delay an initial motion. (Start cycle/jogging etc.). 

What I expected is that there would be an delay between enabling the drive and sending the first step en dir pulses.

Is this a known issue? Can you please advice how to solve this.

cmcgrath5035, 2017-08-23:
While you wait for RobG to comment, you may want to add some info on your DUE to Leadshine interface.
Are you level translating DUE to at least 5V ("TTL Logic")?
I see on the Leadshine specs that the nominal input current for step, dir and enable is 10ma.
You will not get 10ma directly from a DUE (3.3V logic and no much current drive capability), but you may be seeing marginal signal turn-on of DUE is directly connected to Leadshine.
Your issue could be "soft" rising edges on the pulses.
Also, remember that in tinyG/G2core, ~~ENABLE is an active low signal.

Have you actually measured the timing relationship between ~~ENABLE and STEP?

Basti504, 2017-08-24:
Thanks for your reply.

To command the 5V drives from the 3.3V outputs I made the following circuit for DIR,PUL and ENA circuits. The 270 ohm resistor and LED are part of the Leadshine drive, the LED is actually an opto-coupler.

[image]

I do not have access to a scope to measure the actual outputs. If there is a software way of measuring this? I might be able to do this if I have some reference.

The last part of your reply I do not understand, are the times mentioned there the duration of the pulses or the time difference between setting ENA and (DIR & STEP)?

giseburt, 2017-08-24:
Hello. Sorry for the slow reply.

It should be possible to implement a delay after motor enable, but it will
be tricky. Currently, the motor enable is embedded into the step generator
preparation phase, so the enable and direction change happens between
segments and may be immediately before a step.

For the Due it's configured to have a 200khz step tick, so that should mean
a minimal 5us delay after direction and enable before a step pin goes HIGH,
and another 5us before it goes LOW.

That all said, like Carl pointed out if there is a chance that
electrically, due to the way the signal rises and falls through the circuit
(particularly with a signal converter) it may not actually be meeting those
requirements.

You may try a logic-level converter like this one from sparkfun
https://www.sparkfun.com/products/12009 or this one from adafruit (
https://www.adafruit.com/product/757). Double-check the specs and make sure
they can supply the amperage you need. The Adafruit one says it's good for
"low speed" such as up to 2mhz I2C, which might not be fast enough
considering how close we are getting already. It still wouldn't be bad to
have a well tested and proven design for the level shifting.

So, if your okay with using a "start file" you could enable the motors at
the start of the job:

M100 ({me:100})

That will enable all motors for 100 seconds, which should give the machine
time to do any homing or such and then start moving. Make sure you
configure the motors to stay enabled during motion and have a high timeout.
They will stay enabled during any motion in any configuration.

To disable all motors at the end of the job:

M100 ({md:0})

Replace 0 with the motor number to disable a specific motor.

Hopefully that helps.

cmcgrath5035, 2017-08-24:
Basti504 - The open collector schematic you show should work for step and dir, but you will need to invert the Enable signal, since the tinyG/G2core enable as active Low.
Does the Leadshine work OK if you leave enable disconnected?
PS, See https://plus.google.com/communities/103099267575816902412, someone else doing something similar. I used 2N7000 FETs rather than bipolars.

The set-up time for tinyG and tinyGV9 drivers was just data to indicate that these default development boards have a much shorter requirement than the Leadshine. But per Rob's explanation, G2core is a synchronous machine with a 200kHz cycle, so you should be getting 5us between Enable and Step(and Dir).

Basti504, 2017-08-25:
Thanks for both the replies.

I just re-checked the behaviour of the enable signal being connected to the G2. 

* Having the enable signals NOT connected makes the drive enabled by default. So if the enable signal is not connected it enables the drive. This seems to contradict the manual, but the manual is also not really clear in this part.

> Enable (ENA)
> signal: This signal is used for enabling/disabling the driver. High level
> (**NPN control signal, PNP and Differential control signals are on the contrary,
> namely Low level for enabling.**) for enabling the driver and low level for
> disabling the driver. **Usually left UNCONNECTED (ENABLED).**

* Also having the enable signals connected, when I jog the x-axis by 1, I hear the motor being enabled and the position of the axis changes in the software (cncjs) but the axis did not actually move. 

I might try to get hold of a scope or logic analyzer so see what the actual timing is. 

But in the meanwhile I made a switch to switch the main voltage supply of the drives and the router. Which also serves as an emergency stop. I can work with that as well.

cmcgrath5035, 2017-08-25:
I understand your confusion, but to me the statement 
> Usually left UNCONNECTED (ENABLED).
is clear and matches your reported behavior.
It is straightforward for the Driver design to be biased in such a way that if no connection is made to the ENA input, the state is ENABLED.

G2core (and tinyG) are designed to interface to on-board driver devices which expect ENABLE to be active low. The Leadshine and many similar external Stepper drivers are designed for ENABLE to be active high. To interface the DUE to the Leadshine, you need to logically invert the Enable lead before connecting it. You single transistor interface, drawn above, does **not** invert the logic sense as wired; current flows in the optoisolator input diode when the logic sate of DUE is high. That current flow is a logic 1 for the Driver.

> Also having the enable signals connected, when I jog the x-axis by 1, I hear the motor being enabled and the position of the axis changes in the software (cncjs) but the axis did not actually move.
I believe what you are hearing is the motor being energised after a $mt timeout (likely set for 2 seconds) after the Jog move is sent. That would be normal behavior with the Enable lead not inverted. The machine does not jog because Enable is low, disabling the Driver, when the Jog step is sent by G2core. After the Jog step is sent by G2core, Enable remains low for $mt seconds the goes back high.
I believe cncjs reads and displays what the Gcode tells G2core to implement, it does not monitor what G2core reports doing via status reports.

cmcgrath5035, 2017-09-14:
I am working on a wiki page covering this, perhaps it will help

https://github.com/synthetos/g2/wiki/G2core-on-DUE---External-Interfaces

cmcgrath5035, 2017-09-19:
I am working on a wiki page to help folks implement the DUE to external interface.
Perhaps it will help here