synthetos/g2 #292: Getting started with g2core for 3D printing.

pbannister, 2017-08-29:
This is a bit of a meta-issue, as I have not found a discussion forum (as yet) for g2core.

My interest is in using g2core for 3D printing. From the commit history, my impression is that support for 3D printing might not yet be quite entirely there ... though the Printrboard folk must be shipping something.

If some work/contribution on my part is needed to push things forward from almost-there, I am fine with that, as I have done the like before.

Got to the point in my printer-build where I need to choose the electronics. The Arduino Due, and ... some shield that supports stepper-drivers for XYZ, and (hopefully) two extruders ... presumably. Also the usual allowance for fan control, limit switches, etc. 

Not looking for an exhaustive reference. Rather some basic guidance on what is known/working hardware for use with g2core and 3D printing.

giseburt, 2017-08-29:
We indeed have 3D Printing working. The only currently shipping hardware is the [Printrbot Printrboard G2](https://printrbot.com/shop/printrboard-g2/).

We also have [Marlin-compatibility](https://github.com/synthetos/g2/wiki/Marlin-Compatibility) in place, allowing one to use g2core as a drop-in replacement for Marlin-compatible controllers.

If you have any more questions on how things work, please dig around on the wiki, and if you don't find answers there please feel free to ask (and we'll try to fill out that area of the wiki, once we answer your questions).

Thank you for your interest in our project!

-Rob

nitm07, 2017-10-06:
I am using the gshield v5b for controlling a stepper motor. The gshield is mounted on an Arduino Due. When I connect the arduino to the PC using the native USB port I am not able to see TinyG V2 on the device manager. Because of this I'm not able to install the drivers. How do I proceed? Urgent.

giseburt, 2017-10-22:
@nitm07 I’m sorry we didn’t see this. This is sorta in the wrong place, on an issue about 3D printing.

Could you open this into a new issue? We’ll need a few more details, such as which OS are your on (Windows 10, OS X)?, are you connecting via USB C?, how are you powering the Due and the gShield?, etc.

Thank you,
  -Rob

pbannister, 2017-11-17:
Reading through the wiki pages...

pbannister, 2017-11-17:
Reading through the wiki pages, and I have questions. :)

For the topic [Hardware Needed To Use g2core](https://github.com/synthetos/g2/wiki/Getting-Started-with-g2core#hardware-needed-to-use-g2core):

1. Some discussion of the hardware needs for 3D printing would seem order. 
2. I am assuming for XYZ and two extruders requires control of five steppers (or four with only one extruder). As the gShield seems to have support for three steppers, this *appears* to be a non-starter.
3. You have no mention of the Printrboard (which is now offered their website).
4. The planned release dates for the gQuadratic and gQuintic boards is a year in the past. :)

In my case, I am walking through the exercise of using an Arduino Due with a RADDS board - which should an addition here (once/if working).

Or ... I could just edit the wiki, to my understanding, and you can later fix what I got wrong. Your choice. (This would likely mean some placeholders to questions to which I do not have answers - but should save you some time.)

abdrumm, 2018-04-05:
We are working on a page that brings all the 3d printing info together in one place. We have combed thru the g2vore wiki and have a new volunteer rewriting it to be a little more human-readable. We have sent a few questions to read b and will post the remaining outstanding info

Our initial post is here: https://printrbot.com/2018/03/08/printrboard-g2-general-information/

Our g2 board is for sale on our site... Ultimachine is now making them and we should be able to keep them in stock. 

Brook

Brook

justinclift, 2018-04-11:
@abdrumm Looks nifty. :smile:

@pbannister Did the above cover all the bits you need, or are there still gaps you're needing help with? :smile:

justinclift, 2018-04-18:
@abdrumm As a thought, there's a start towards a page listing G2core (and TinyG) based products here:

&nbsp; &nbsp; https://github.com/synthetos/g2/wiki/g2core-in-use

And yeah, it's mostly Printrbot entries for now as they were the easiest to figure out the use TinyG/g2core.  Would you be ok to cast an eye over it and let me know if anything's wrong?  Or just edit the page directly if you want. :smile:

pbannister, 2018-04-20:
@justinclift I have not yet returned to this topic. Looks to be a week or two away.

justinclift, 2018-04-20:
Thanks @pbannister, no worries. :smile:

pbannister, 2018-05-17:
So ... off by a couple weeks. 

This is more of an FYI. Not making an ask.

Started by trying to trying to compile on a Raspberry Pi3 (will be on the printer running OctoPrint, so makes sense), but that still does not work, for reason you explained prior. (Hey. Had to check...)

Could create a Linux VM on my Macbook, but instead tried using Google's Cloud Shell (the advantage being a standard environment anyone could access ... if it works). 

```
$ make PLATFORM=DUE BOARD=gShield
"Found that we're in Linux"
Installing the necessary tools...
cd ../Motate/MotateProject/motate/../../Tools && make "ARCH=gcc-arm-none-eabi"
make[1]: Entering directory '/home/preston_bannister/build.g2core/g2/Motate/Tools'
The gcc-arm-none-eabi tools need installed. Downloading linux/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2 and installing now.

mkdir -p linux && \
cd linux && \
wget "https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2"
--2018-05-16 16:48:45--  https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
Resolving launchpad.net (launchpad.net)... 91.189.89.223, 91.189.89.222
Connecting to launchpad.net (launchpad.net)|91.189.89.223|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: https://launchpadlibrarian.net/268330503/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2 [following]
--2018-05-16 16:48:46--  https://launchpadlibrarian.net/268330503/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
Resolving launchpadlibrarian.net (launchpadlibrarian.net)... 91.189.89.228, 91.189.89.229
Connecting to launchpadlibrarian.net (launchpadlibrarian.net)|91.189.89.228|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 92600244 (88M) [application/x-tar]
Saving to: ‘gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2’
2018-05-16 16:51:54 (484 KB/s) - ‘gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2’ saved [92600244/92600244]

cd linux && \
rm -f "gcc-arm-none-eabi" && \
tar xjf gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
cd linux && \
ln -s "gcc-arm-none-eabi-5_4-2016q2" "gcc-arm-none-eabi"
make[1]: Leaving directory '/home/preston_bannister/build.g2core/g2/Motate/Tools'
Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.c
    -> build/gShield/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o 
bash: ../Motate/MotateProject/motate/../../Tools/linux/gcc-arm-none-eabi/bin/arm-none-eabi-gcc: No such file or directory
```

Hmm...
```
$ $PWD/arm-none-eabi-gcc
-bash: /home/preston_bannister/build.g2core/g2/Motate/Tools/linux/gcc-arm-none-eabi/bin/arm-none-eabi-gcc: No such file or directory

$ file arm-none-eabi-gcc
arm-none-eabi-gcc: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.8, stripped

$ uname -a
Linux cs-6000-devshell-vm-ace75d61-3ca7-4dc7-b57a-c5cdf4954aaf 3.16.0-6-amd64 #1 SMP Debian 3.16.56-1+deb8u1 (2018-05-08) x86_64 GNU/Linux

$ sudo apt-get install libstdc++6:i386
********************************************************************************
You are running apt-get inside of Cloud Shell. Note that your Cloud Shell
machine is ephemeral and no system-wide change will persist beyond session end.

Note that your home directory does persist across VM instantiations, so consider
installing your tools into your home directory whenever possible.

To suppress this warning, create an empty ~/.cloudshell/no-apt-get-warning file.
The command will automatically proceed in 5 seconds or on any key.

Visit https://cloud.google.com/shell/help for more information.
********************************************************************************

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libstdc++6:i386
E: Couldn't find any package by regex 'libstdc++6'
```

Right. Nevermind. Looks as though 32-bit binaries are not supported.

Again, point of information only. Guess a Linux VM is in order...

pbannister, 2018-05-17:
FYI, compiles fine on a 32-bit Ubuntu VM (no surprise): 
Ubuntu 16.04.4 LTS (GNU/Linux 4.4.0-116-generic i686)

```
$ time make PLATFORM=DUE BOARD=gShield -j4
...
Making binary bin/gShield/g2core.bin 
--- SIZE INFO ---
   text    data     bss     dec     hex filename
 185420       0   43584  229004   37e8c bin/gShield/g2core.elf
Build 101.02 101.02-15-g194c

real    0m10.119s
user    0m34.312s
sys     0m3.332s
```

Any current preference on the branch to use?
```
$ git branch -a
* edge
  remotes/origin/HEAD -> origin/edge
  remotes/origin/dev-168-gquintic
  remotes/origin/dev-212-common-enable
  remotes/origin/dev-307-spring-comp
  remotes/origin/dev-omco-issue-2
  remotes/origin/digital-dro
  remotes/origin/edge
  remotes/origin/edge-087
  remotes/origin/edge-090
  remotes/origin/edge-100
  remotes/origin/gh-pages
  remotes/origin/master
  remotes/origin/omc-072.65
```

justinclift, 2018-05-17:
> Any current preference on the branch to use?

`master` has had more people testing it over time, so might be better for getting started with. :smile:

`edge` is useful for creating PR's from, doing development work on, and that kind of thing.

As a thought, someone mentioned recently that Cura is known to work, though I'm not sure which branch they were running.  They did mention a few minor warnings were printed at some point.

Not sure about OctoPrint personally, as I've never tried it.  The OctoPrint project itself does look like it's going well though, and the maintainer seems responsive which is good. :smile:

justinclift, 2018-05-17:
@pbannister One small gotcha to be aware of (and we'll probably change things in the near-ish future), is that since you're compiling for the `gShield` the X/Y/Z axes for that start out disabled:

* https://github.com/synthetos/g2/blob/1ff7f600923c622fbf2b546e7d15a24498d1d1c1/g2core/settings/settings_default.h#L372
* https://github.com/synthetos/g2/blob/1ff7f600923c622fbf2b546e7d15a24498d1d1c1/g2core/settings/settings_default.h#L413
* https://github.com/synthetos/g2/blob/1ff7f600923c622fbf2b546e7d15a24498d1d1c1/g2core/settings/settings_default.h#L454

For me when starting out, that made things really confusing.  For example, the [CNCjs](https://github.com/cncjs/cncjs) user interface all looked fine, however "nothing actually moves".  You might find similar happening in OctoPrint.

Creating your own configuration is pretty easy though.  Here's an example from yesterday, for adding a Shapeoko3 specific setup:

* https://github.com/justinclift/g2/commit/23f58dbb0d5932668be4a6fe5fed65b5c4dd1253
* https://github.com/justinclift/g2/commit/b9158edc2fd9f918f83a1f67a7d42f0d88763ba8

The first commit makes a new `CONFIG=Shapeoko3` entry for the `make` command, and creates a new "settings_shapeoko3.h" file (manually copied from the Shapeoko 2 Dual Y one without changes).
The second commit makes the changes to that file which my Shapeoko3 needs. :smile:

The axes aren't disabled in those configs, so that's not a thing you'd have to worry about with those ones.  You will need to experiment and probably read lots of wiki pages to get things working for your 3D printer though.

pbannister, 2018-05-17:
Thanks for clarifying the branch choices. To clarify my interest...

I have two 3D printers at present, and am building a third of my design. The third printer is to run g2core. The two present printers each have an attached Raspberry Pi3 running OctoPrint. In effect, OctoPrint web-based control panel for the printer(s). OctoPrint is very, very nice - so much so that I entirely removed the physical control panel (with the horrid Marlin UI) from one printer. 

At present, I do designs in OnShape, render STL to GCode via Cura (on my Macbook), and use a Cura plugin to send the GCode (via OctoPrint) to the printer.  In theory I could use OctoPrint to render STL to GCode (via a Cura component), but this does not make sense in my usage, as I often want some manual control over the slicer. (Plus, my Macbook is much faster than the Pi, and my designs tend to burn up a bunch of CPU-cycles when rendering GCode.)

The electronics for the third printer is a Pi3 (to run OctoPrint), and an Arduino Due with a RADDS board and Trinamic 2208 stepper-drivers. (The Due/RADDS/Trinamic combo seemed to make sense at the time.)

Did the compile for gShield just as a test.

Today's exercise (hopefully) is to do the appropriate configuration to compile for Due/RADS/Trinamic.

FYI - misadventures: 
https://plus.google.com/u/0/collection/gUpBJE

justinclift, 2018-05-17:
Cool.  Your photo's look nifty. :smile:

Branch wise, definitely use the `master` branch for now, as you'll probably want something stable to work with.

If you do hit bugs or weirdness though, do check if the master branch has been updated since you last compiled and flashed the Due.  It doesn't happen super often, but it's worth keeping an awareness of. :smile:

pbannister, 2018-05-18:
Should have mentioned, the printer is using CoreXY, which seems to have not yet landed in the master branch: #215 and/or #285 . So ... maybe not the **master** branch. (Got as far as defining the **settings/settings_spaceage3d.h** file, and **M1_MOTOR_MAP**.)

Seems I have to use branch **dev-168-gquintic** for now? Or...??

justinclift, 2018-05-18:
Oops.  Sorry @pbannister, I didn't realise that CoreXY meant it needs different Kinematics handling.

You're right, only the `dev-168-gquintic` branch seems to have the patches in it.

Looking at the commit by @giseburt which adds the support (based upon code by @ewidance), it looks like the **Mn_MOTOR_MAP** value should be set to `AXIS_COREXY_A` or `AXIS_COREXY_B` (depending on which axis it's supposed to be):

* https://github.com/synthetos/g2/pull/285/commits/91f7793f86a8b3a72a9b9b82677746bb9a42c6d0#diff-951b9b1dd559251607afcc5001c5b24fL93
* https://github.com/synthetos/g2/pull/285/commits/91f7793f86a8b3a72a9b9b82677746bb9a42c6d0#diff-951b9b1dd559251607afcc5001c5b24fL105

There is an example settings file in that branch set up for CoreXY too:

* https://github.com/synthetos/g2/blob/91f7793f86a8b3a72a9b9b82677746bb9a42c6d0/g2core/settings/settings_axidraw_v3.h

Does that help? :smile:

justinclift, 2018-05-18:
Also note that the dev branch there is more experimental than the stable master branch.  So... well... definitely report back any bugs or weirdness you hit.  Wouldn't be too surprising to find unexpected gotchas in parts of it. (!) :wink:

ril3y, 2018-05-18:
> Just tried it. Doesn't cleanly apply. 😦
>
> I think we'll need some direction from @giseburt
> <https://github.com/giseburt> on this one. 😄

Rob is sick with the flu (his whole family is down) so it might be a bit of
a delayed response.

pbannister, 2018-05-18:
The last bug that went around took me out for a week, about three weeks
back, and still lingers. :/

Did find the example CoreXY usage. Going to look at patching the change to
kinematics.cpp into master ... maybe (looked isolated?). Have to get a kid
off to school, first.

justinclift, 2018-05-18:
Thanks for the update @ril3y. :smile:

ril3y, 2018-05-18:
I could be wrong but dev-168 is the only branch I know that is capable of
3d printing.  I have been printing with it for a few months now.  I am
using cura to slice / send print job.

aldenhart, 2018-05-18:
Notes:
* The edge branch is capable of 3d printing. The Printrbot uses an M3 ARM core and uses that code.
* The Ultimaker settings are there for retrofitting the Ultimaker with the edge branch, which we have also done.
* The dev-168 branch is the M7 ARM core branch, and also has 3d printing functions, but works only on some pre-release HW we are currently testing. 

Hope that clarifies things.

pbannister, 2018-05-18:
What branch are @abdrumm and the Printrbot folk using?

pbannister, 2018-05-18:
FYI, when switching between branches, I needed to delete:
`rm -rf g2/Motate/Tools/linux`
Otherwise I got a link(?) error when compiling on **master**. Not sure the exact cause.

justinclift, 2018-05-19:
> Might want a realclean rule in Motate.mk?

Make a PR? (it'd against edge branch) :smile:

As a data point, I generally run `git clean -dffx; git reset --hard HEAD`.  That restores the tree to pristine "just cloned" condition, keeping in whichever branch your in at the time.

To make it faster, I added an alias for it in my bash shell config (.bash_profile):

```
alias regit="git clean -dffx; git reset --hard HEAD"
```

Of course, there are other ways to do the same thing. :smile: