synthetos/g2 #293: A Axis not moving during G2/G3 moves

somethingneweveryday, 2017-09-25:
Greetings all,
I've taken on the challenge of building a tangential knife machine, and I was hoping to use g2 firmware for control. Over the course of the day I've been able to get a reasonable grasp of g2, and have a setup that allows A axis rotation during G1 moves. What I haven't yet been able to achieve, is A axis rotation during G2/G3 moves. The following code sees the XY arc move completed first, followed by the A axis move. Obviously for a tangential knife, both the arc move and the A axis move need to happen simultaneously - with both moves arriving at their destination at the same time.

G1 X0 Y100 F1000
G2 X100 Y0 I0 J-100 A25

I'm unsure if this is supported by g2, or perhaps I'm missing something. I've spent a couple of hours online looking for answers, but no luck as yet. This is my first time using g2, though I do have a reasonable amount of experience using Grbl, Marlin and Smoothie. Surprisingly, Marlin is the only firmware I've been able to instruct to move the A axis (E axis) during an arc move.

Any help will be greatly appreciated.

thx Dan

aldenhart, 2017-09-26:
That's an interesting one. Let me check this out and report back.

somethingneweveryday, 2017-09-27:
thanks @aldenhart... would really like to be able to use G2. The more I play with it, the more I realize how awesome it is.

aldenhart, 2017-09-27:
A lot of work has gone into it. I'm glad you are enjoying it. I have some tests to run. The number of edge cases in Gcode is enormous.

somethingneweveryday, 2017-09-27:
I can only imagine the amount of work that must have gone into G2. And the fact that it is open source, really is amazing.
I've spent a reasonable percentage of my working life building CNC machines commercially. Typically we would spend many thousands of dollars for controllers that don't have all the features offered by G2.
Well done to everyone involved.