synthetos/g2 #294: missing definitions HOT_FUNC, HOT_DATA

vaughanatworld, 2017-10-12:
Compile problems: missing definitions for "HOT_FUNC" and "HOT_DATA"

Development environment: win10, AtmelStudio 7, gcc-arm-none-eabi-5_4-2016q2-20160622-win32, git branch dev_168-gquintic, d377b16.

If I were using Xcode would I avoid these problems?
Any help is welcome.
Many thanks.
Brian

ril3y, 2017-10-12:
Yes,  This is a issue with compiling in windows I have ran into this.  I
think you can remove those "decorators" or
whatever and it will build.  But it does come at the cost of optimizations
i think.

@rob what do you think?

vaughanatworld, 2017-10-13:
Here are all the changes I made to get the solution to compile without errors. The  //bav are my markers for changes. It would be nice to have a proper definition for HOT_DATA.

  C:\Users\Brian\Documents\GitHub\g2\g2core\plan_line.cpp(740):                delta = std::max(delta, (float)fabs(bf->junction_unit[axis] - bf->nx->unit[axis])); // formula (1) //bav
  C:\Users\Brian\Documents\GitHub\g2\g2core\stepper.cpp(872):                correction_steps = std::min(std::min(correction_steps, (float)fabs(travel_steps[motor])), STEP_CORRECTION_MAX); //bav
  C:\Users\Brian\Documents\GitHub\g2\g2core\stepper.cpp(874):                correction_steps = std::max(std::max(correction_steps, (float)-fabs(travel_steps[motor])), -STEP_CORRECTION_MAX); //bav
  C:\Users\Brian\Documents\GitHub\g2\Motate\MotateProject\motate\Atmel_sam_common\SamPins.h(81):        // For use on ADC pins only!    //bav
  C:\Users\Brian\Documents\GitHub\g2\Motate\MotateProject\motate\Atmel_sam_common\SamPins.h(82):        kDifferentialPair = 1<<9,       //bav
  C:\Users\Brian\Documents\GitHub\g2\Motate\MotateProject\motate\MotateUtilities.h(37): #define HOT_FUNC __attribute__((hot)) //bav
  C:\Users\Brian\Documents\GitHub\g2\Motate\MotateProject\motate\MotateUtilities.h(38): #define HOT_DATA //bav

giseburt, 2017-10-22:
I thought the correction for this had been pushed already. I’m sorry you’re
seeing this. I’ll look into it shortly.

justinclift, 2018-07-29:
We should probably check if this is still an issue, and close (or fix first) as appropriate. :smile: