synthetos/g2 #295: Planner not doing its job with G1 commands?

mateoarv, 2017-10-21:
Hello, i recently flashed my arduino due with the edge branch firmware. However, i noticed that even though i send multiple lines at a time, the planner doesn't seem to "plan". For example, if I send the lines:

G1 F1000
X10
X20 Y10

The X axis shouldn't stop between lines 2 and 3, but it does. This is a problem when trying to send PCB gcodes which are mostly composed by G1 commands, the result is a jerky movement.

Is this intentional? I believe this didn't happen with previous version i was using.

Thanks in advance!

ril3y, 2017-10-22:
This is most likely because tinyg has planned the movement and you are not
sending the next line fast enough.  AKA the input buffer is starving.

Riley

giseburt, 2017-10-22:
There are several things this could be.

1) You’ll need to be more specific about “stop between lines.” Is it going
through the 90º turn slowly? That would be adjusted with {jt:n} and the X
and Y jerk settings. It should only be coming to a complete stop if:

2) How quickly is “multiple lines at a time” sending the lines? Chilipepr
and Coolterm both introduce delays between lines. These are rarely a
problem in a job since the planner can still fill up faster than the lines
are sent. However, with only a few lines, it’s possible for it to plan and
start to execute the first line before the second line comes in. Once a
move is planned *and starts to execute* then the move is not replanned.
Again, in a job this isn’t a problem.

What version of g2core are you using? You should be using the code from the
edge branch. It has a 30ms delay from an empy buffer to fill and plan
before it starts moving, which eliminates these type of “cold start” issues.

-Rob

mateoarv, 2017-10-22:
Hi, thank you both for the answer. About the input buffer starving, I think this may not be the case since this happens when using Chillipeppr too. Also, the serial terminal I'm using doesn't put delays between lines. I'm using the code from the edge branch with the gshield configuration.

Is there a way to change that 30ms delay to something higher to test if this is the source of the problem?