synthetos/g2 #296: wpos precision loss in inches mode

MitchBradley, 2017-10-23:
In inches mode, the work position report precision is 3 decimal places, which is not enough for some machines.  That results in imprecise position displays, loss of reports (when position changes are filtered out because of changes smaller than EPSILON3), and jog granularity problems.

I have a fix for this problem and will submit a pull request after I have cleaned up the changes.

giseburt, 2017-10-23:
Sounds good. Thank you for catching this.

justinclift, 2018-04-11:
@MitchBradley Guessing you've been using a version of g2core with your patch included, for a few months now.  If that's the case, are the extra decimal places working out as hoped? :smile:

Just thinking it might be useful to get some feedback on the change from someone's who's probably actually been using it in reality for a while now, just in case you feel your patch could do with some subsequent tweaking. :grin:

justinclift, 2018-04-11:
Awesome. :smile:

ril3y, 2018-04-13:
Mitch What are you making with g2?  We are always curious to see what
people make with it.  Any videos?

MitchBradley, 2018-04-13:
I make golf putters from bamboo.  There is a time-lapse video at
https://www.youtube.com/watch?v=yoLG7eB8SuI

In that video, the milling is done with Mach3 but since then I have built a new shop that is specialized for putters, converted my Taig mill to a g2core/Due/RaspberryPi/cncjs controller, and added several cnc processes to replace manual shaping.

ril3y, 2018-04-13:
Pretty awesome!  Thanks for sharing.

justinclift, 2018-04-13:
Excellent!  We'd better make a page on the wiki that showcases stuff being created with g2core.

This should go on there. :smile:

justinclift, 2018-04-13:
New "g2core in use" page created here:

&nbsp; &nbsp; https://github.com/synthetos/g2/wiki/g2core-in-use

Not 100% sure of the wording, layout, etc, but it seems like a reasonable start. :smile: