synthetos/g2 #300: CAN GPIO

mawildoer, 2017-11-11:
Less an issue rather than the best forum I can find for this information, but I have recently developed a CAN GPIO network that I've implemented on my machine using the controller built into the DUE.

CAN is a fantastically noise resistant and capable of handling thousands of inputs using very little wire. We're currently working on boards with standard RJ45 jacks to use CAT6 Ethernet Cables to join the nodes together. These cables also carry the power for the peripheral network.

So far, I've implemented digital inputs which may be configured the same way as any other di pin.

Currently I'm working on adding heartbeats for the bus to guarantee it's running during operation.

Later we also plan to use the bus to communicate with our wireless probe and control the ATC.

I'm planning to commit the additions here soon, but it'd be great to gauge everyone's interest.

Here's a video of my machine using a probe (crudely) connected over CAN and a laser tool height setter connected the same way:
[Video of it probing - Youtube](https://youtu.be/kZxRn_lXfGc)

Up next... closed loop, 240v servo control over SPI. I'll keep you posted ;)
[Photos of the PCB](https://user-images.githubusercontent.com/4241578/32687712-af3e15b8-c716-11e7-96a8-12c237c3d402.jpg)

justinclift, 2018-04-12:
Excellent!  Personally, I've been meaning to take a look at CAN stuff (in general), to get a better understanding of how it could be useful for us.

With the CAN GPIO bit you've been working on, is that using the g2core code as a basis or something, or is it new code which might one day be kinda useful to incorporate? :smile:

Went looking through your GitHub repositories (https://github.com/mawildoer?tab=repositories) but not seeing anything obvious there. :wink:

giseburt, 2018-04-12:
Wow! Somehow I missed this issue coming in. I'd love to hear more. In particular how you're incorporating the CAN into the system.

hurr1k4ne, 2018-04-12:
Both are fantastic news.
This would get g2core to a whole new Level.

Keep it Up!

mawildoer, 2018-04-13:
Thanks guys! I'm glad there's some interest in it. I will have to make the code public now. 

I actually rebuilt the GPIO section of the g2core code. It's now more OO based which is convenient given the possibility for SO many IOs that CAN opens. 
I've added a CAN header and modified the compiler scripts accordingly. It was based on an arduino DUE CAN library which I gutted and re-referenced the headers.

Thankfully, the CAN network seems to be extremely robust. The PSoCs we've been using as nodes have some really nice built in noise filtering making for really reliable noise free limits and laser tool height setter (as you can see in the background of the video).

Unfortunately, the project has effectively been scraped as a friend and I deffered uni to start a company based on a machining center we were developing, however I'm happy to post most of the code. 

As for the servos, they worked quite well actually. They're based on Ti's Instaspin-Motion packages and the drivers we made were able to run the standard cheap Chinese servo motors pulg-n-play with awesome accuracy. We never got to the point of actually putting them on a machine though.

giseburt, 2018-04-13:
That would be very cool. Even if it doesn’t merge with current code then we
could at least utilize some of the CAN code. I’m in the middle of a GPIO
refactor as well, so there may be some work merging them in but it’ll be
god to see it done from a different perspective.

And whenever possible we’ll use actual commits so you get credit in the git
log. Failing that accreditation in the changelog.

Thank you!
-Rob

justinclift, 2018-04-13:
>I'm glad there's some interest in it. I will have to make the code public now.

Definitely, please do. :smile:

justinclift, 2018-04-13:
> They're based on Ti's Instaspin-Motion packages and the drivers we made were able to run the standard cheap Chinese servo motors pulg-n-play with awesome accuracy. We never got to the point of actually putting them on a machine though.

As a thought, it there any chance you'd throw the design for the drivers you made, online somewhere under an Open Source license of some variety?  Just thinking it might be useful, so the next person to wanting to do stuff in this area has a potentially easier time of it. :smile:

justinclift, 2018-04-22:
That's Awesome @mawildoer, thanks heaps! :smile:

No worries at all about the state it's in.  Some of my dev code is **all kinds** of interesting (not the good sort) while I'm trying to figure something out.  Shouldn't be a problem. :wink:

> A few commits ago, there was work on a major revision and it may be wise to go a few back for the most stable work.

With the branches there, the two main ones with work in them appear to be the [can_ths](https://github.com/geratech/controller/commits/can_ths) one and [master](https://github.com/geratech/controller/commits/master).  Err... could you point out the commit where the (potentially unstable) work on a major revision started?  Just to make things easier to know where to draw an initial imaginary line. :smile:

mawildoer, 2018-04-23:
I reckon the first working (obviously the simplest) is called "CAN fucking works now. woop woop. THS on pin 9" -- It was around 3am I believe

Tracking through those master commits, it looks like I did a better job than I remembered keeping master clean enough.

justinclift, 2018-04-23:
Thanks @mawildoer. :smile:

justinclift, 2018-04-26:
Just as potentially useful reference info, for anyone else looking into the CAN side of things with the Arduino Due this site has plenty of exactly relevant info:

* http://copperhilltech.com/blog/arduino-due-can-bus-controller-area-network-interfaces/

Saying that after reading **way** too much about CAN today.  Most CAN stuff around isn't Due specific, and the above website (copperhilltech.com) has lots that is.  Including CAN transceiver shields for the Due.

mawildoer, 2018-04-26:
Thanks for the reference Justin. That's actually where I got started with it as well. Great reference and useful for debugging as well