synthetos/g2 #301: Compiling g2core on a Raspberry Pi 3

pbannister, 2017-11-13:
I want to build g2core on a Raspberry Pi 3. There is a rationale...

I have a lashup for a 3D printer that consists of a Raspberry Pi 3, an Arduino Due, a RADDS board, and Trinamic 2208 stepper drivers. The Pi3 will run Octoprint. The Due will run g2core.

[image]

The Pi3 is connected to the programming post of the Due. Yes, it does seem faintly ridiculous to use a $35 board for development, but ... this does make some sense. (By my estimate, the Pi3 is roughly equivalent to an early-2000s desktop ... which is not bad.)

I could fall back to building on my Macbook, but then I would have to bring the either the printer electronics or the Macbook within reach of a USB cable. Also, I have a couple of Chromebooks, and would like to use them for development at times.

The Pi is going in the printer for Octoprint - in any case - so using it as an always-connected development host makes some sense. I can use Secure Shell (a Chrome extension from Google) to SSH into the Pi from the Macbook or any of my Chromebooks (the connection settings are replicated via Google Drive).

(Also, I am extremely comfortable with Unix as a development environment.)

So I can sit down with any of my laptops, use SSH into the Pi, and program the Due. The long explanation above is to explain how I got here.

Attempting to compile:

$ make CONFIG=TestV9
"Found that we're in Linux"
Installing the necessary tools...
cd ../Motate/MotateProject/motate/../../Tools && make "ARCH=gcc-arm-none-eabi"
make[1]: Entering directory '/home/pi/projects/g2/Motate/Tools'
The gcc-arm-none-eabi tools need installed. Downloading linux/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2 and installing now.

mkdir -p linux && \
cd linux && \
wget "https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2"
--2017-11-12 23:49:57--  https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
Resolving launchpad.net (launchpad.net)... 91.189.89.222, 91.189.89.223
Connecting to launchpad.net (launchpad.net)|91.189.89.222|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: https://launchpadlibrarian.net/268330503/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2 [following]
--2017-11-12 23:49:58--  https://launchpadlibrarian.net/268330503/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
Resolving launchpadlibrarian.net (launchpadlibrarian.net)... 91.189.89.228, 91.189.89.229
Connecting to launchpadlibrarian.net (launchpadlibrarian.net)|91.189.89.228|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 92600244 (88M) [application/x-tar]
Saving to: ‘gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2’

gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2 100%[==================================================================================================================>]  88.31M   608KB/s    in 2m 3s   

2017-11-12 23:52:02 (733 KB/s) - ‘gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2’ saved [92600244/92600244]

cd linux && \
rm -f "gcc-arm-none-eabi" && \
tar xjf gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
cd linux && \
ln -s "gcc-arm-none-eabi-5_4-2016q2" "gcc-arm-none-eabi"
make[1]: Leaving directory '/home/pi/projects/g2/Motate/Tools'
Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.c
    -> build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o 
bash: ../Motate/MotateProject/motate/../../Tools/linux/gcc-arm-none-eabi/bin/arm-none-eabi-gcc: cannot execute binary file: Exec format error
../Motate/MotateProject/motate/Motate.mk:475: recipe for target 'build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o' failed
make: *** [build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o] Error 126

(Yes, I know "TestV9" is not my board.)

Not looking for someone else to solve the problem for me (though hints would be welcome). Looks like "Motate" is downloading executables that are not in usable on the Pi. As a guess, the native GCC on the Pi might be able to target the Due (as yet unchecked). So I have some work.

Posting this issue to track my efforts.

giseburt, 2017-11-13:
I agree, being able to compile on an ARM SBC would be nice.

So, a little history: We use C++17 features, and we don't have a lot of time for tech support for compilers, so we "locked down" what compiler gets used in Motate by having it download the toolchain from ARM and install it locally in the repo. That way we know we're getting the compiler we tested with. (There's still, rarely, the OS difference. Windows is one that shines there.)

So, if you look at [Motate/Tools/Makefile](https://github.com/synthetos/Motate/blob/sams70-port/Tools/Makefile#L19-L21) you'll see where we determine what OS is being used. We use ARM's [precompiled toolchains](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads) which, oddly, doesn't actually have a build that *runs* on ARM. (I would expect to see an ARM Linux there.)

So, there are two problems to solve:
- (Easy) In the Makefile split the `OS = LINUX` (and all the variables that contain `LINUX` where `${OS}` is used to refer to) into `OS = LINUX_X86` and `OS = LINUX_ARMHF` (or similar)
- (Not so easy) Build from the source on that page *or* locate pre-built from a reliable source that version of the toolchain. Keep in mind that this is a more-or-less complete toolchain, including GDB and other supporting tools.

We have had trouble in the past with OS package repos having older GCCs and them taking a long time to update. You're welcome to try using them, but if there are compilation issues they may be tricky to hunt down.

Hopefully that's helpful.

Thank you for your interest in our project and the offer to help out!
  -Rob

pbannister, 2017-11-15:
FWIW ... misadventures between ...

C++17(!?) ... punt on compiling on the Pi3, at least for now.

Made the mistake of accepting the latest MacOS update ... which hung, and
left my MacbookPro unable to boot. The restore from TimeMachine offered an
18 hour estimate. Started some experiments in the time between.

Wanted to spend the time with Google's cloud computing offering in any
case. Maybe do builds in a Google Compute VM instance? Maybe do builds in a
Google container instance (Docker)?

Oh wait, Google offers "Cloud Shell <https://cloud.google.com/shell/>"  ...
interesting. (Did not know this existed.) This looks to be a small Docker
container, but fairly decent set of tools, and free(!).

Cloned your repository on Github. Followed your directions...

preston_bannister@angelic-hexagon-186016:~$ *git clone
https://github.com/pbannister/g2.git
<https://github.com/pbannister/g2.git> --recursive*
Cloning into 'g2'...
remote: Counting objects: 23126, done.
remote: Total 23126 (delta 0), reused 0 (delta 0), pack-reused 23126
Receiving objects: 100% (23126/23126), 111.04 MiB | 31.49 MiB/s, done.
Resolving deltas: 100% (15155/15155), done.
Submodule 'Motate' (https://github.com/synthetos/Motate.git)
registered for path 'Motate'
Cloning into '/home/preston_bannister/g2/Motate'...
remote: Counting objects: 12451, done.
remote: Total 12451 (delta 0), reused 0 (delta 0), pack-reused 12451
Receiving objects: 100% (12451/12451), 11.63 MiB | 20.39 MiB/s, done.
Resolving deltas: 100% (9383/9383), done.
Submodule path 'Motate': checked out '41e5b92a98de4b268d1804bf6eadf3333298fc75'
preston_bannister@angelic-hexagon-186016:~$ cd g2/g2core
preston_bannister@angelic-hexagon-186016:~/g2/g2core$ *make CONFIG=TestV9*
"Found that we're in Linux"
Installing the necessary tools...
cd ../Motate/MotateProject/motate/../../Tools && make "ARCH=gcc-arm-none-eabi"
make[1]: Entering directory '/home/preston_bannister/g2/Motate/Tools'
The gcc-arm-none-eabi tools need installed. Downloading
linux/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2 and
installing now.

mkdir -p linux && \
cd linux && \
wget "https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2"
--2017-11-14 19:04:51--
https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
Resolving launchpad.net (launchpad.net)... 91.189.89.222, 91.189.89.223
Connecting to launchpad.net (launchpad.net)|91.189.89.222|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: https://launchpadlibrarian.net/268330503/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
[following]
--2017-11-14 19:04:52--
https://launchpadlibrarian.net/268330503/gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
Resolving launchpadlibrarian.net (launchpadlibrarian.net)...
91.189.89.229, 91.189.89.228
Connecting to launchpadlibrarian.net
(launchpadlibrarian.net)|91.189.89.229|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 92600244 (88M) [application/x-tar]
Saving to: ‘gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2’

gcc-arm-none-eabi-5_4-2016q2-20160622-linux.ta
100%[===================================================================================================>]
 88.31M  2.59MB/s   in 15s

2017-11-14 19:05:08 (5.96 MB/s) -
‘gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2’ saved
[92600244/92600244]

cd linux && \
rm -f "gcc-arm-none-eabi" && \
tar xjf gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
cd linux && \
ln -s "gcc-arm-none-eabi-5_4-2016q2" "gcc-arm-none-eabi"
make[1]: Leaving directory '/home/preston_bannister/g2/Motate/Tools'
Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.c
    -> build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o
*bash: ../Motate/MotateProject/motate/../../Tools/linux/gcc-arm-none-eabi/bin/arm-none-eabi-gcc:
No such file or directory*
../Motate/MotateProject/motate/Motate.mk:475: recipe for target
'build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o'
failed
make: *** [build/TestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o]
Error 127

Well, something went wrong. Checking....

preston_bannister@angelic-hexagon-186016:~/g2/g2core$ ls -l
../Motate/MotateProject/motate/../../Tools/linux/gcc-arm-none-eabi/bin/arm-none-eabi-gcc-rwxr-xr-x
2 preston_bannister preston_bannister 776368 Jun 22 2016
../Motate/MotateProject/motate/../../Tools/linux/gcc-arm-none-eabi/bin/arm-none-eabi-gcc

Right. The file is there, and appears to have the correct permissions.
But...

preston_bannister@angelic-hexagon-186016:~/g2/g2core$
../Motate/MotateProject/motate/../../Tools/linux/gcc-arm-none-eabi/bin/arm-none-eabi-gcc-bash:
../Motate/MotateProject/motate/../../Tools/linux/gcc-arm-none-eabi/bin/arm-none-eabi-gcc:
No such file or directory

... it does not execute. Right. Something about Google Cloud Shell?

(Again, not asking you to solve my problem. Just a note.)

pbannister, 2017-11-15:
Ah. Builds on MacOS, not on Ubuntu. Checking the makefiles for Linux...

giseburt, 2017-11-15:
Ah, I know this one! The tools used from the branch you are using are 32-bit tools. So, you have three options:

1. Switch to the more experimental [`dev-168-gquintic`](https://github.com/synthetos/g2/tree/dev-168-gquintic) branch (which of course still supports all the other boards, but is occasionally broken since it's in active development), which uses the [`sams70-port`](https://github.com/synthetos/Motate/tree/sams70-port) branch of Motate, which uses the next-to-latest tools from ARM. The newer tools from ARM are compiled for 64-bit on Linux.
2. Install 32-bit compatibility libraries for your flavor of linux. Each flavor is wildly different as to how that's achieved, but they all support it one way or another.
3. Use a 32-bit Linux VM.

I have built using Docker before, and it should work fine. We do our CI builds on TravisCI, which uses a 64-bit linux. In order to get the compatibility there [we use](https://github.com/synthetos/g2/blob/edge/.travis.yml#L7):

```bash
sudo apt-get install lib32z1 lib32ncurses5 lib32bz2-1.0
```

Hopefully that saves you some searching time.
  -Rob

pbannister, 2017-11-15:
Thanks. Much better...

Did checkout on dev-168-qquintic (g2) and sams70-port (Motate). Ran (twice) the command:
```bash
$ time make clean all CONFIG=TestV9
```

### On MacOS
```
$ time make -j 4 clean all CONFIG=TestV9
....
Making binary bin/TestV9-g2v9k/g2core.bin 
--- SIZE INFO ---
   text	   data	    bss	    dec	    hex	filename
 177040	      0	  19968	 197008	  30190	bin/TestV9-g2v9k/g2core.elf
Build 100.22 100.22-214-gcfc3

real	0m9.431s
user	0m30.066s
sys	0m4.419s
```
### On Ubuntu (VM on Google Compute)
```
Making binary bin/TestV9-g2v9k/g2core.bin 
--- SIZE INFO ---
   text	   data	    bss	    dec	    hex	filename
 177040	      0	  19968	 197008	  30190	bin/TestV9-g2v9k/g2core.elf
Build 100.22 100.22-214-gcfc3

real	0m38.110s
user	0m33.944s
sys	0m3.368s
```
### On Google Cloud Shell:
```
Making binary bin/TestV9-g2v9k/g2core.bin 
--- SIZE INFO ---
   text    data     bss     dec     hex filename
 177040       0   19968  197008   30190 bin/TestV9-g2v9k/g2core.elf
Build 100.22 100.22-214-gcfc3

real    0m30.431s
user    0m26.784s
sys     0m2.420s
```

Figure I am better off using whatever is in active development.

closed 2017-11-15 by pbannister