synthetos/g2 #302: Build on OSX broken?

knollja, 2017-11-22:
I'm attempting to build g2core 100.26 with the gShield pinout on my Macintosh.  I've carefully followed the instructions here: https://github.com/synthetos/g2/wiki/Getting-Started-with-g2core-Development#compiling-the-code
and:
https://github.com/synthetos/g2/wiki/Compiling-g2core-on-OS-X-(with-Xcode)

I clone the GitHub repository to my local drive, open the g2core.xcworkspace project in Xcode, select gShield as the target, and hit make.  The build succeeds.  I flash it to the Due which succeeds, but the result doesn't work correctly.

Talking to the Due over the serial port works, and it responds to query commands, but any movement commands do nothing.  Checking the step, direction or enable pins, they're all low all the time.

Carl McGrath sent me a build he did on a linux box, and that works driving the motors, etc.

Something about the build process on the Macintosh is resulting in a non-fiunctional result, even though the build is reporting success.

Anyone successfully building on OSX who can offer advice on how to debug this?

Thanks!

-John Knoll

giseburt, 2017-11-22:
It appears from what you describe that it’s building fine. It’s likely
you’re using the that the default “test” configuration, which has all the
motors disabled.

Start with another configuration and see if you have better results.

knollja, 2017-11-22:
OK, I'll give that a try.  Any suggestions which one to use?  What should I look for to know if it's a test configuration with disabled motors?

Ultimately I'm looking for a gShield pinout.

knollja, 2017-11-22:
OK, that was it.  Thank you.

For the record, I do think that having an Xcode build target called "gShield" that compiles into a binary that won't actually drive a gShield is confusing.  Shouldn't that target be called "gShield_motors_disabled" or something?  

Furthermore, I think it would be useful to have a target that actually does build a functional gShield firmware.

ril3y, 2017-11-22:
Good point. Do you have a pull request?

knollja, 2017-11-22:
I don't.  I'm pretty new to Git and haven't read up on how to do that.

cmcgrath5035, 2017-11-24:
I suggest we proceed with a bit of caution here. 
I generally agree with knollja's suggestion, but worry a bit about folks who might blindly build a particular configuration with an improper parameter set and jump to trying to jog their machine. Slamming a gantry into a side rail is usually not a permanent damage event with belt machines, but may be with popular screw machines. I have no idea what might happen to a $$ 3D printer setup.
In knollja's case, his build did  produce a gShield pinout controller(I assume), but the settings were blocking the motors, we need to ensure users understand the significance of the parameter set
Getting new users to fully understand what the settings mean and how to set them is always a challenge. 
This is further complicated for Chilipeppr users where the Settings widget is out of date relative to the G2core parameter set and I know the Archive and Restore is broken for G2core. Having to use CLI, and lack of parameter persistence will be hurdles for new users.

knollja, 2017-11-24:
Perhaps the working build configuration could have very conservative motor speeds, but for the love of god please let it be a working configuration.  I burned many hours to this confusion.  

If you take a look on Youtube, you'll see that apparently its common practice to get the motor control working before the motors even get installed into the machine.  That's also what I did in this case, but conservative speed settings would be a reasonable compromise.

Certainly it's pretty gross that _settings_ currently need to be compiled into the binary, but perhaps the Chilipeppr widget will be extended to work for this case.

Ril3y, if you're still there, would you be willing to act on this pull request:  https://github.com/synthetos/g2/pull/118

The DueFromOSX.sh script that's still part of the source tree is obsolete and no longer works.  I burned a few hours to that error only to find it had been fixed nearly two years ago but the fix had never been merged.

ril3y, 2017-11-25:
@knollja I did not see this sorry let me take a look at this.

ril3y, 2017-11-25:
@knollja I merged the pull request #118 but I wanted you guys to know about the new nodejs flashing utility.  It works well on linux and osx so far.
https://github.com/synthetos/g2/wiki/Flashing-with-Node-SAM-Ba-Utility-in-OSX

closed 2017-11-25 by ril3y

knollja, 2017-11-25:
Thanks Ril3y!  

Reading that page, it appears that to use the node.js method I need an AVR Atmel ICE programmer, but the DueFromOSX.sh does not require one.

giseburt, 2017-11-25:
No the node-sam-ba uses the serial port, and doesn’t require any other
hardware.

knollja, 2017-11-25:
Huh.  The Usage: paragraph starts with: 

> To use the node-sam-ba programming utility you need to make sure your AVR Atmel ICE programmer is hooked up via usb and your g2core board is also hooked up via usb and is in bootloader mode.

Is this not the case?

giseburt, 2017-11-25:
I can’t check the instructions you are referring to at the moment, but I
can say that that’s wrong. You do not need any additional hardware. It
won’t work with an external programmer anyway.

knollja, 2017-11-25:
OK, good to know.  Seems like the usage instructions should be revised then.

knollja, 2017-11-25:
Page is updated, thanks Ril3y!