synthetos/g2 #303: No steps generated

benlaurie, 2017-12-02:
I am using current master, i.e. commit 92e455e965e2a68741218ca9b10162b70e45ffa2. If I jog the X axis (the only one I am currently monitoring), I don't see any steps at all (using the TestV9 config - I've also tried with Othermill).

However, if I use the prebuilt version, TinyG2_Due-edge-078.03-default.elf, I do see steps.

This is using a bare Due with no attached drivers at all.

I guess I'm doing something obviously wrong, but I've spent several days banging my head against it, so not _that_ obvious.

hurr1k4ne, 2017-12-03:
The Axis are disabled / Not configured in the config Files. Please have a
Look at them. You will find the Motors disabled and Not assigned the the
Axis.

Greetings, André

benlaurie, 2017-12-03:
Thanks for that - but what do you mean by the config files? If I look in settings_othermill.h, for example, the motors do appear to be enabled and assigned.

Also, if I look at the motor/axis settings from chillpepper, they seem OK...

giseburt, 2017-12-03:
Another thing to look for is if the machine is in alarm state. Depending on
the input configuration (and how you have the machine wired), it may go
into alarm or feedhold state before the usb comes up and you can get the
alarm message.

You can check by querying the stat with {stat:n}. If you get 2 (Alarm), 6
(Hold), 11 (Interlock), 12 (Shutdown), or 13 (Panic), then it won’t
generate steps when you issue a gcode. (Full stat list here:
https://github.com/synthetos/g2/blob/master/g2core/canonical_machine.h#L72-L85
)

You probably seeing Interlock or Shutdown. For the Othermill config
specifically, it has:

DI2 set to cause Shutdown after Halt (
https://github.com/synthetos/g2/blob/master/g2core/settings/settings_othermill.h#L220-L222
)
and
DI4 set to cause Interlock (
https://github.com/synthetos/g2/blob/master/g2core/settings/settings_othermill.h#L230-L232
)

And with a Due those pins won’t have pull-downs on board, and the internal
pull-down isn’t enabled at boot in Edge. (It is in dev-168-gquintic which
will be merged in next.) So, you either need to add pull-down resistors to
those inputs (10k or so will do fine).

You can see that DI1 is Due pin 14 (
https://github.com/synthetos/g2/blob/master/g2core/board/ArduinoDue/gShield-pinout.h#L196
)
and
DI4 is Due pin 17 (
https://github.com/synthetos/g2/blob/master/g2core/board/ArduinoDue/gShield-pinout.h#L199
).

(On Due we align the numbers to the silk screen. Verified in the pin
assignments file here:
https://github.com/synthetos/g2/blob/master/g2core/board/ArduinoDue/motate_pin_assignments.h
).

-Rob

giseburt, 2017-12-03:
I didn’t first my thought. Sorry. You need to either add the pull-down
resistors or change the configuration.

-Rob

benlaurie, 2017-12-13:
Thanks, I'll try this when I get home...

benlaurie, 2017-12-16:
Unfortunately, that does not seem to be the issue - from ChilliPeppr:

{stat:n}
{"r":{"stat":1},"f":[1,0,9]}

i.e. READY.

I tried building dev-168-gquintic instead but it has lots of errors.

benlaurie, 2017-12-16:
By the way, it might be helpful to know what commit TinyG2_Due-edge-078.03-default.elf was built from.

benlaurie, 2017-12-16:
Solved!

Problem was I should have been specifying BOARD=gShield instead of using the default g2v9k.

I am now getting steps. Hoorah!

That said, I found this out by trying to connect an Atmel ICE, and realised eventually that the wrong chip was selected. However, even with the right chip selected, I don't end up connected:
```
...
SAM3X8E
./board/ArduinoDue.gdb:2: Error in sourced command file:
Remote connection closed
"monitor" command not supported by this target.
(gdb) quit
```

Any ideas?

BTW, is there any way to talk to you guys without raising a Github issue? IRC channel? Mailing list?

justinclift, 2018-04-11:
Hey Ben, is this solved now?

I'm just going through closing out issues that aren't really needing to be open any more.  This kind of seems like one, but I'm not 100% sure. :smile:

benlaurie, 2018-04-11:
The original issue is indeed solved - as I said, I was using the wrong board. The default build is not for a Due, right?

Still haven't made the ICE work, though!

justinclift, 2018-04-11:
Wasn't sure of the default board, so just jumped on my CentOS dev board and tried `make` without any arguments.  That barfed with "BOARD not defined - please provide a project name".

So went looking on the wiki for the most likely "How to compile..." type of page.  The OSX and Linux one seems closest (are you on FreeBSD?):

* https://github.com/synthetos/g2/wiki/Compiling-g2core-on-Linux-and-OS-X-(command-line)#build-the-sources

That has:

  For Arduino Due:

  cd g2/gcore
  make PLATFORM=DUE BOARD=gShield

  For TinyG V9 board:

  cd g2/g2core
  make CONFIG=TestV9

Which lines up with what you were hitting.  Looks like you went with the TestV9 one instead of the `PLATFORM=DUE BOARD=gShield` combo when doing it.

I think I did the same thing when trying to figure out how to get things working (at all).  I'm just doing stuff with a Due + some stepper drivers on a breadboard (for now) + wiring furball hooked up to motors, switches, etc. :smile:  For me it's working (now), but jeez... it was frustrating to get through the initial stages.

Have you gotten things to the point of making your own settings file yet?  Or well past that now? :smile:

justinclift, 2018-04-11:
With the ICE... I haven't bought one yet.  I'll likely get to it later, but it's probably not something I need to do right away.  My focus is more on a different layer of things.  eg creating CAM post(s) so g2core can be driven by CAM software.  Not sure if I need an ICE for that just yet. :wink:

justinclift, 2018-04-12:
... and I forgot to close this yesterday.  Oops. :wink:

closed 2018-04-12 by justinclift