synthetos/g2 #309: Weird final step?

benlaurie, 2017-12-16:
While debugging #303, I noticed the end of a jog looks a little odd:

[image]

Why is the very last step so close?

justinclift, 2018-04-11:
Ditto with this.  Is this issue closable, or should it remain open? :smile:

benlaurie, 2018-04-11:
Well, is there an explanation of the final step? Not a serious issue, I don't think but doesn't seem quite right...

giseburt, 2018-04-11:
That shouldn’t be there. It’s definitely a bug. However, that section has
been refactored somewhat on edge and further refined in Dev-168-gQuintic
(which should work for all boards btw), so hopefully it’s already fixed.

justinclift, 2018-04-11:
Excellent!

@benlaurie Would you be ok to try either of those two branches then, to check if it's indeed now working properly there? :smile: