synthetos/g2 #310: arcs are interpreted wrong

martin2250, 2017-12-17:
Hi,

I'm using Fusion360 (with a custom postprocessor, worked fine with GRBL for years) to generate GCode. This particular file was a test to try out 3D machining. I originally ran this file on grbl without issues. The toolpath should look like this:
[image]
(This is a screenshot from OpenCNCPilot, grbl followed the path exactly)

I ultimately want to switch to TinyG G2 and today I tried the same file on TinyG G2. I used a modified version of OpenCNCPilot to send the file and my machine crashed into the end of travel in positive Z direction (really hope nothing broke).
I just tried opening the same file with ChiliPeppr and this is what it looks like:
[image]
Comparing this visualization to the movement of the toolhead in my preview window (after the machine crashed and I cut power to the servos) it looks very similar. I guess TinyG G2 also interpreted the file like this.

The issue starts with the '3D Parallel' operation (around line 3500).
I don't know why those arcs are not interpreted correctly. Before this part of the file, there are arcs in all three planes and those were interpreted correctly.

The code is here: [https://pastebin.com/JrcRhv2Z](https://pastebin.com/JrcRhv2Z)

Cheers!

justinclift, 2018-04-12:
Sorry for the long delay with this, people have been snowed under some things. :frowning:

With the issue you're showing here, did you get a solution worked out?

martin2250, 2018-04-13:
Hi Justin,

I switched back to grbl / grbl on STM32, so I didn't investigate this any further

justinclift, 2018-04-13:
No worries @martin2250, the definitely makes sense.  Needing a working solution, and not getting a response -> revert to what worked. :smile:

Do you happen to have the custom Fusion 360 post processor around still, that you'd put together?  Just asking because I've been mucking around with making Fusion 360 post's for g2core recently too (various branches [here](https://github.com/justinclift/Mill-Post-Processors-Maker-Machines)).

The stuff I've been doing is for enabling 4th and 5th axis control via g2core, but I'm definitely curious to see what changes other people have tried. :smile:

martin2250, 2018-04-14:
I've just stripped the existing grbl PP to the bare minimum (no spindle enable/speed, no coolant, no M30 at the end, just plain motion commands): https://github.com/martin2250/RandomBits/blob/master/BetterGRBL.cps

giseburt, 2018-04-14:
Did you try the built-in TinyG post in Fusion? I’ve been cutting a fair bit
using arcs in all three plains and having no issues.

-Rob

martin2250, 2018-04-14:
I can't find the part anymore, so I can't test this easily. The file worked on grbl, so there shoudn't be any issues with the post. Just take a look at this G2 in ChiliPeppr vs OpenCNCPilot (or test it on your machine with the steppers powered down):
```
G90 G21 G18 
G0 X-9.865 Z-4.518
G2 X-17.852 Z-4.548 I-3.569 K-112.957
```

amx1, 2018-05-28:
@giseburt 

I am using the built in TinyG post proc of Fusion with arcs enabled. It runs really fine!

justinclift, 2018-05-30:
Tried out the example G-Code ([breaksTinyGG2.nc](https://pastebin.com/JrcRhv2Z)) in the OP with `master` branch (that's the stable code)... and yep that errors out immediately:

```
> G2 X28.49 Y0.002 I-1.436 J1.159
{"err":{"code":159,"msg":"Arc specification error - endpoint is starting point","line":9,"data":"G2 X28.49 Y0.002 I-1.436 J1.159"}}
```

Then tried using the `edge` branch (our leading edge code), compiled just now.  Same problem:

```
> G2 X28.49 Y0.002 I-1.436 J1.159
{"err":{"code":159,"msg":"Arc specification error - endpoint is starting point","line":9,"data":"G2 X28.49 Y0.002 I-1.436 J1.159"}}
```

As a data point, the visual display of the G-Code in CNCjs "looks ok" though:
[image]

Completely guessing here, but it kind of seems like the G-Code that was generated might be a bit dodgy from g2core's point of view.  Using the TinyG post in Fusion360 should (in theory) generate G-Code that doesn't barf. :smile:

@martin2250 Would you have time to try things out again, but using the TinyG Fusion post instead?

justinclift, 2018-05-30:
@martin2250 And yeah, you've mentioned not having the part any more.  This is more to get you using what **should** be a 100% working post, and see if the issue crops up again.  In theory (!), it shouldn't.

aldenhart, 2018-05-31:
I have run [this file](https://pastebin.com/JrcRhv2Z) successfully using the latest edge release, fb 101.03. The offending arc `G2 X28.49 Y0.002 I-1.436 J1.159` was in line 9, which ran almost immediately. 

I'm still waiting for the full file to complete - if there's a failure I'll edit this comment, but I don't expect there will be.

justinclift, 2018-05-31:
That's weird.  My follow up post yesterday has disappeared.

It wasn't super important - just some further info and a screenshot - but that's kind of unsettling.  I know GitHub have been updating their comment handling code on the backend over the last few days, but ugh.  Hopefully it doesn't keep happening.

aldenhart, 2018-06-01:
I'm going to close this issue. I've run this thing multipel times with the latest edge (101.03) and it's fine. If any errors with arcs pop up we'll open another issue.

closed 2018-06-01 by aldenhart

martin2250, 2018-06-04:
Hi Justin,

it didn't generate an error message for me then and I can't see why it would complain about end- and startpoints being the same. The line you posted occurs twice (L9 and L793) and both times the X and Y coordinates are different from the previous lines.
Also I know the preview looks fine in most programs (see op), it's just Chilipeppr that behaves similarly to  the version of G2 that I was using.
Anyways, good to know that this is fixed in the latest edge. I probably won't test it anymore as I've switched to GRBL on STM32 which works with my existing software.

Martin