synthetos/g2 #312: Using External motor drivers

Hallulla, 2017-12-20:
Hi!
I'm looking to use g2core on Arduino due to drive a small 5 axis machine. I am thinking of using 5 [Big EasyDriver](https://www.sparkfun.com/products/12859?_ga=2.116567494.1129919016.1513789857-1738791622.1506469362) boards, since tinyG and Gshield can't drive 5 independent motors. 
So far my research indicates this can be done: pins are equivalent, microstepping can be configured to g2core defaults and enable pin is active low on the drivers.
However, I wanted to know your opinion before proceeding.
I looked around online and could not find similar issues.

benlaurie, 2017-12-21:
FYI, I am using a Due with external drivers, no problem - note that you need to set the board to gShield to get the right CPU/pinouts.

Also, don't forget the Due is a 3.3V system!

Curious what 5 axis machine?

hurr1k4ne, 2017-12-21:
I am using my due too with 8 AMP Nema23.
Works flawless.
Hint: use shielded Data cables ;-)

cmcgrath5035, 2017-12-24:
Have a look at this product, it might provide some helpful hints (schematics available)
https://www.djuke.nl/index.php?page=shop.product_details&category_id=1&flypage=flypage.tpl&product_id=406&vmcchk=1&option=com_virtuemart&Itemid=8

pburkli, 2018-01-23:
I'm using an Arduino Due with the above board (G2core shield by djuke.nl). It translates all outputs to 5V and protects them with photocouplers. Works great.

ril3y, 2018-01-23:
I would ask anyone using g2 with bigger motors was is willing to share some
pictures / build logs please do so.  I would love them for the site.

Hallulla, 2018-01-26:
Thanks everyone for your comments. Sadly my CNC build is being delayed for personal reasons.
Hope to resume this later this year :( Love g2core/tinyg

kitwn, 2018-01-27:
I'm planning to change my 3-axis machine from LinuxCNC to g2core controlled by Goko or CNCjs if I can persuade the team to keep the pre-compiled .bin page up to date ;-) . I'm better with bits of wood and fast moving sharp steel than I am with software and haven't yet mastered the art of turning Github gobbledygook into something I can squeeze onto my shiny new Due board.

hurr1k4ne, 2018-01-27:
Please See your other issue for an answer.
The generic release files are built automatically in every Release.
Have a Look at githubs Release Page.

Greetings 
André

kitwn, 2018-01-28:
André
Thanks to your answer to the other thread I have now found the Releases folder. As I said there, Releases doesnt immediately show up on the synthetos/g2 github front page and I hadn't found any links to it in my search through the wiki and other sources of information. This is not meant as a criticism of people who  give a great deal of their time and effort free of charge, but as someone who has spent much of his professional career teaching high level technical subjects I am very aware of the danger of assuming that your newest students already have some of the knowledge that you yourself feel as if you were born with. Knowing that release files are built automatically and where to find them is an example of this.

I now have everything I need (I hope!) to produce a working machine using Goko, g2core and (more relevant to this thread) a bespoke breakout board of my own devising or the one from djuke.nl mentioned above . Once I have it sorted I will be documenting the whole process, probably on one of the CNC forums.

Regards
Kit

scotthz, 2018-03-03:
FWIW I am running g2 on a Due with 4 external Sainsmart ST-M5045 drivers. Motors are middle-of-the road NEMA 23s. I have one board which buffers the control signals between the Due and the drivers. Probably not necessary, since the drivers have their own optoisolaters, but it made me feel good that I wasn't overloading the Due pins. Also a simple filter board with RC circuits for home/limit switches to address noise concerns. I have a branch of synthetos/g2 where I have been committing my changes. So far only configuration, board, and makefile stuff. Happy to share if anyone's interested. Have run into one thing that I think is a bug but haven't investigated.

justinclift, 2018-04-13:
Reading through this, it seems well and truly answered by now. :smile:

So, going to close this issue.  If I've somehow missed something and it should be kept open instead, please feel welcome to re-open it. :smile:

closed 2018-04-13 by justinclift