synthetos/g2 #313: Show settings view for di10 - di12 missing in console

AVRsteffen, 2018-01-25:
I use Arduino Due with master branch build 100.26.
I  increase in gpio.h amount of useable digital Inputs from 9 to 12. It works well but I wonder cause can't see settings of digital inputs by send $di for all. Software show only settings for digital inputs from 1 to 9. I am confused cause physical all Inputs from 1-12 exist and can change settings with command like "$di10mo=0" or "$di12fn=2"..
Software show me too if send request of state them like "$in12" or "$in11"..

But why software don't show me all settings together for command like "$di" ?

I am scrolling through codes but can't understand why is not perform all of my input setting request.

Is it a bug?
Who can help me to fix it?

Steffen H.

cmcgrath5035, 2018-02-20:
Perhaps try {"di":n} ?
G2core starts up in JSON mode

cmcgrath5035, 2018-02-24:
(I am an interested bystander, learning from your question and possible fixes. This is significantly new feature, perhaps the set-up requirements are not clearly documented)

Did you add default definitions for your additional di pins in your settings.h file?
Perhaps something needs to exist at compile time to properly set up query parsers.
From your description above, it is apparent that you can modify values, such as "$di10mo=0", but perhaps things don't work correctly if those are not initiated at compile time.
It will likely be helpful to the developers to have a link to the settings.h file you used when compiling.

AVRsteffen, 2018-02-24:
Yes, I can modify and use pins for DI10 - DI12. They work correctly.
Sorry, have no link for my setting.h file
Can only give you an extract of them here.
**settings.h**:
`//*** Input / output settings ***
/*
    IO_MODE_DISABLED
    IO_ACTIVE_LOW    aka NORMALLY_OPEN
    IO_ACTIVE_HIGH   aka NORMALLY_CLOSED

    INPUT_ACTION_NONE
    INPUT_ACTION_STOP
    INPUT_ACTION_FAST_STOP
    INPUT_ACTION_HALT
    INPUT_ACTION_RESET

    INPUT_FUNCTION_NONE
    INPUT_FUNCTION_LIMIT
    INPUT_FUNCTION_INTERLOCK
    INPUT_FUNCTION_SHUTDOWN
    INPUT_FUNCTION_PANIC
*/
// Xmin 
#define DI1_MODE                    NORMALLY_CLOSED
#define DI1_ACTION                  INPUT_ACTION_NONE
#define DI1_FUNCTION                INPUT_FUNCTION_LIMIT

// Xmax
#define DI2_MODE                    NORMALLY_CLOSED
#define DI2_ACTION                  INPUT_ACTION_NONE
#define DI2_FUNCTION                INPUT_FUNCTION_LIMIT

// Ymin
#define DI3_MODE                    NORMALLY_CLOSED
#define DI3_ACTION                  INPUT_ACTION_NONE
#define DI3_FUNCTION                INPUT_FUNCTION_LIMIT

// Ymax
#define DI4_MODE                    NORMALLY_CLOSED
#define DI4_ACTION                  INPUT_ACTION_NONE
#define DI4_FUNCTION                INPUT_FUNCTION_LIMIT

// Zmin
#define DI5_MODE                    NORMALLY_CLOSED
#define DI5_ACTION                  INPUT_ACTION_NONE
#define DI5_FUNCTION                INPUT_FUNCTION_LIMIT

// Zmax
#define DI6_MODE                    NORMALLY_CLOSED
#define DI6_ACTION                  INPUT_ACTION_NONE
#define DI6_FUNCTION                INPUT_FUNCTION_LIMIT

// free
#define DI7_MODE                    IO_MODE_DISABLED
#define DI7_ACTION                  INPUT_ACTION_NONE
#define DI7_FUNCTION                INPUT_FUNCTION_NONE

//free
#define DI8_MODE                    IO_MODE_DISABLED
#define DI8_ACTION                  INPUT_ACTION_NONE
#define DI8_FUNCTION                INPUT_FUNCTION_NONE

// free
#define DI9_MODE                    IO_MODE_DISABLED
#define DI9_ACTION                  INPUT_ACTION_NONE
#define DI9_FUNCTION                INPUT_FUNCTION_NONE

// free
#define DI10_MODE                    IO_MODE_DISABLED
#define DI10_ACTION                  INPUT_ACTION_NONE
#define DI10_FUNCTION                INPUT_FUNCTION_NONE

// Hardware interlock input
#define DI11_MODE                    NORMALLY_OPEN
#define DI11_ACTION                  INPUT_ACTION_NONE
#define DI11_FUNCTION                INPUT_FUNCTION_INTERLOCK

// Hardware Emergency Stop input
#define DI12_MODE                    IO_MODE_DISABLED
#define DI12_ACTION                  INPUT_ACTION_STOP
#define DI12_FUNCTION                INPUT_FUNCTION_NONE`

**gpio.h**:
`#ifndef GPIO_H_ONCE
#define GPIO_H_ONCE

/*
 * GPIO defines
 */
//--- change as required for board and switch hardware ---//

#define D_IN_CHANNELS       12  // 9    // number of digital inputs supported
//#define D_OUT_CHANNELS    13          // number of digital outputs supported
#define D_OUT_CHANNELS	    8           // number of digital outputs supported
#define A_IN_CHANNELS	    0           // number of analog inputs supported
#define A_OUT_CHANNELS	    0           // number of analog outputs supported

//#define INPUT_LOCKOUT_MS    50        // milliseconds to go dead after input firing
#define INPUT_LOCKOUT_MS    10          // milliseconds to go dead after input firing`

cmcgrath5035, 2018-02-24:
Hmm, looks OK to me. You'll have to wait for a Dev to stop by.

justinclift, 2018-04-11:
Hmmm, at a rough guess from this and https://github.com/synthetos/g2/issues/315#issuecomment-366478127, it sounds like there might be some kind of bug with digital input pins > 10 not being displayed when `$di` (and it's JSON equivalent) are run.

AVRsteffen, 2018-04-11:
Yes it's not displayed in Text mode or JSON too.

But I am to stupid for understand C++
😣

justinclift, 2018-04-11:
> But I am to stupid for understand C++

Yeah, I don't program with C++ much either any more. :wink: