synthetos/g2 #314: Please help an 'ordinary' user!

kitwn, 2018-01-27:
I'm very interested in trying g2core as a replacement for my current LinuxCNC setup but am not quite enough of a computer geek to start from scratch compiling the latest master branch myself. The pre-compiled Windows .zip file available at http://synthetos.github.io/g2/ referenced in the wiki is still on v0.97 which doesn't match the 0.99 requirement of Goko 0.3.5 which is the software I'm hoping to use for controlling it.

I'm sure there are many other people like me who are more interested in building a working CNC machine than mastering all the arts of software development who would appreciate  an "easy steps for the ordinary user" wiki page with the latest pre-compiled files attached. This would then expand the user base which can only be a good thing for getting more useful feedback and cash donations if requested.

Thanks

Kit

hurr1k4ne, 2018-01-27:
The Problems are the machine specific settings. They are precompiled inside the bin.
So it is impossible to create a genuine firmware.
I used linux to compile mine. Which ist just editing the config and running a lenghty Command. 

But you may have a go with the release files in GitHub. There are machine specific firmwares which you can use to Start from.
The downside is you have to overwrite the settings in every startup with your own, because g2core can not save the settings persistent in EEprom.

https://github.com/synthetos/g2/releases

Greetings André.

kitwn, 2018-01-27:
André,
Thanks for your reply. LinuxCNC includes the 'stepperconfig' application which is a GUI used to set up the machine specific parameters and test the movement of the machine. Once you are happy with the settings a configuration file is saved which is automatically read by the main application when started. I'm assuming that the Due can, like every other Arduino board I've used, read a file from EEPROM during startup.

I've written a fair amount of software for earlier Arduino boards using the simple Arduino language but have so far balked at taking my programming skills any further. I have enough half-finished projects as it is!

Looks like I'm talking myself into adding learning Java or some other GUI writing language to my ToDo list so that I can create the g2core version of 'stepperconfig' and have it write a file to the Due. But I have enough unfinished projects as it is! This would make a great widget to add to CNCjs or Goko. We just need g2core modified to read a file on startup. Can anyone recommend an idiots guide ( a REAL idiots guide beginning right at the beginning) to compiling software on Github using Windows?

Kit

kitwn, 2018-01-27:
André,
I've just followed the link you put in your post and there are a pile of binaries! The 'releases' folder doesn't appear on the front page of Github and I haven't noticed any links to it elsewhere. The 'Getting Started with G2core' page in the wiki sent me off to  http://synthetos.github.io/g2/ which is v0.97. The page has no "last updated on..." information so I had no idea that it's a fossil.

This is the kind of very low level information that those of us who know more about CNC machines than we do about managing software online need to help us get up off the floor and take the first step.

Kit

cmcgrath5035, 2018-02-17:
Kit
Be aware that none of the releases  current available at https://github.com/synthetos/g2/releases are built for DUE hardware, they are for other custom machines. Not all pins may function as you need.

kitwn, 2018-02-18:
cmcgrath5035,
That's interesting. I thought the ONLY hardware it worked on at the moment was the Due! I'm sure the wiki says that somewhere. Thanks for that vital bit of information. 

I have loaded a binary onto my Due and sucessfully communicated with it from the serial interface of the Arduino IDE and from cncgoko but have not yet made a breakout board for testing it's ability to control any motors etc.
There is a nice design for a Due shield at https://github.com/labsud/GTwo but I was planning on making a veroboard version for intial testing. The combination of a suitable binary file and the above shield would be a great way to encourage more people to start using g2core.

Would anyone like to compile a binary of the latest stable release intended to load onto a Due using the standard pinout as given in the Wiki and/or suited to the above shield? A mutually agreed donation is available if you can save me having to learn how to do it myself. I have enough unfinished projects as it is!

Kit

cmcgrath5035, 2018-02-18:
The only "openly available" hardware is the DUE. The v9k is an early development prototype and the other builds are for 3rd party machine specific hardware that runs a version of G2.
You need a build for DUE with gShield to get functionality matching the (draft) Pinout chart at https://github.com/synthetos/g2/wiki/Arduino-DUE-Pinout-for-g2core.

I am on the road at the moment, don't have access to my test build. Check back later in the week.

I highly recommend you invest the time to set up a G2 build environment. 
G2 is in continuous development, you may want to get quick access to new features.
An important side benefit is that you can easily modify an appropriate settings.h file and build a G2core that has your most recent parameters as defaults.

kitwn, 2018-02-19:
Carl,
Thanks for your advice, I agree that I need to get to grips with being able to build G2 myself. My problem is where to begin? It can't be that difficult but all the explanations seem to assume an existing level of knowledge I don't have and I'm confused as to the very first steps I need to take in order to make progress. I need the equivalent of the first driving lesson entitled "First, unlock the doors" but haven't had any luck finding it yet.

kitwn, 2018-02-19:
OK, it turns out that "unlocking the doors" includes clicking on the 'pages' header to open up a whole new world of wiki menu pages including instructions on how to compile g2.  Please tell me I'm not the only first-time visitor to GitHub who didn't spot that one?

Kit

hurr1k4ne, 2018-02-19:
Hmm it cost me some minutes the First time, but found it in the right Menu.

An advise: If you are used to use linux, i recommend you to compile the
Source Code with linux. It Is much easier to Setup, understand and handle.

If you setup your build environment in windows, please follow the
Instructions very careful. The compiler ist very picky! So often
first-timers miss something and end with a broken compile environment.
But it should Work, as i guess ;-).

kitwn, 2018-02-19:
I've been trying to make it work in Windows. No luck on my W7 desktop machine: I've been following the instructions on "https://github.com/synthetos/g2/wiki/Compiling-and-configuring-G2-on-Windows" but cannot get past the "'make' is not recognized as an internal or external command, operable program or batch file." notice no matter what I try.

I've had a go at following "https://github.com/synthetos/g2/wiki/Compiling-G2-on-Windows-10-and-Atmel-Studio-7" on my W10 laptop but there is no "gear menu" in v1.0.13 of the GitHub Desktop application so I'm stuck at "Cloning the Git Repository - Github Windows App" section 5. I've tried going straight to the "Compiling the Project" section but the dropdown menu for selecting 'gshield' is greyed out... At least it was, it's just opened up after the program has been open for about 10 minutes! Must have been downloading something.

It worked! I now have a new 'g2core.elf' file, presumably based on the gshield pinout. That's enough for tonight. Tomorrow I can load it onto my Due board and see if it will talk to 'goko' and throw some pulses out of the pins I expect it to.

Kit

cmcgrath5035, 2018-02-19:
After loading your new .elf file and restarting the DUE, you should see a slow "heart beat" from one of the LEDs - that is normal

alfredanil, 2018-03-06:
I wish someone built a g2core configuration tool like marlin config tool
 https://github.com/akaJes/marlin-config
This would attract more users to g2core.
Also adding new shield to Due is a bit tedious with Motoate. I did manage to partially build RADDS shield motate pins but still need more tweeking and research to get it right. FYI RADDS V1.5 shield is for Arduino Due and can run upto 6 steppers pololu type (8 or 9 with exp), 6 mosfets, 5 thermistor inputs, sd card and more.

justinclift, 2018-04-11:
@kitwn How'd this go??  Did you get it figured out? :smile:

kitwn, 2018-04-13:
Justin,
I managed to get Atmel-Studio 7 up and running an have compiled a binary with it which is now moving motors with the aid of a breakout board from djuke.nl. The link is to a thread on MYCNCUK which includes a video of where I'm up to.
Next time I have a few days off work I will be sorting out limit switches and producing my own bespoke config file to enable them.
http://www.mycncuk.com/threads/11833-G2Core-GOKO

Kit

justinclift, 2018-04-13:
Thanks @kitwn, looks like solid progress. :smile:

Your video is nifty and is the first time I've seen Goko "in action".  Goko seems to have really good support for g2core.

In the part where you're showing the motor settings, only 4 motors were shown.  Any idea if that's a hard coded thing in Goko, or if it was adapting to the number of motors defined in the g2core config?

eg if someone with 6 motors tried using Goko, do you reckon that all 6 motor would have settings available to be adjusted?

justinclift, 2018-04-13:
Oh, that g2core shield looks nifty too.  When I get past my own "make it work on a breadboard" stage, that might be a useful upgrade for me too.

We should probably create a page on the wiki that lists g2core compatible boards (with links, info, etc) to make things easier for future people.  How did you come across that board? :smile:

justinclift, 2018-04-13:
@alfredanil Looking at the RADDS v1.5 page [here](https://www.reprap.me/radds-v15.html), it does seem like a potentially good alternative as well.

Did you get the updates (eg motate, etc) finished to make it "just work" with g2core? :smile:

justinclift, 2018-04-13:
@alfredanil
> I wish someone built a g2core configuration tool like marlin config tool
> https://github.com/akaJes/marlin-config

Interesting thought.  Might be useful to look into it later. :smile:

kitwn, 2018-04-13:
Justin,
At the price I decided to just buy the Djuke.nl board rather than bother making even the simplest prototype breakout myself. The site sugests using the Shapeoko config which is what I have done. The next step is adding limit switches and confirming their configuration on my test bed. You cannot configure that via the Goko GUI, though you could write a dedicated text file and upload it.

The ability to read and write a configuration from/to the Due is useful, especially if you can write a macro to upload the file every time serial communication is established, but a bespoke version of G2Core will be more reliable once you have used Goko to establish the appropriate values for your specific machine. I'd really like to see someone who understands Studio-7 better than I do produce a simple guide to producing a config file and compiling the result.

I'm not sure if Goko will show more motors if they are configured but suspect it will. The website and the wiki have been quiet since about September last year. With luck, this renewed interest will spark some further input from the development team. My first impressions are that the combination of g2Core, Goko, a Due board and the djuke.nl breakout can offer a very low-cost entry point for building CNC machines without the need for a dedicated, ex-junkbox PC with a parallel port running LinuxCNC.

justinclift, 2018-04-13:
> I'd really like to see someone who understands Studio-7 better than I do produce a simple guide to producing a config file and compiling the result.

Good suggestion. :smile:

aldenhart, 2018-05-14:
I will close this comment as this conversation appears to be over. Please feel free to re-open if you want to continue the conversation.

closed 2018-05-14 by aldenhart