synthetos/g2 #315: Safety Connections

timothy-godfrey, 2018-02-10:
Hello! I am working on using an Arduino Due with external stepper drivers (ie no driver shield, just level adapters and screw terminals) in place of the factory controller on a laser cutter. I would like to connect 
 - Emergency Stop
 - Door Interlock
 - Cooling Alarm
signals to my controller. As far as I can tell, the GRBL-compatible pins for feedhold, reset, and cycle start signals in the gShield pin layout are not used in the G2core code at the moment. Is there any reason not to configure those pins as digital input pins for the three functions I mentioned above? Are there any functions in the code (in addition to the IO and Alarm Handling documentation) or upcoming feature implementations that I should be aware of while I'm considering and planning how to connect those signals? 

Thanks!

hurr1k4ne, 2018-02-10:
In the Source Files in machine configuration you can edit the i/o Pin
functions. E.g. Alarm, Panic.... Normal Open, normal closed and so on.

hurr1k4ne, 2018-02-11:
> HI! Thanks for the reply. I'm familiar with the options for adding pins
> and configuring the alarm modes. This is what I had in mind to do with the
> GRBL pins. I'm just wondering if there are any factors I should be aware of
> or have in mind when choosing which physical pins to use, or anything else.
>
> Thanks!

Hm, for my two cents Just keep an eye on noise. The due Board is very picky
with it.

I use external stepper drivers and they Ran only Well after changing to
shielded cables.

pburkli, 2018-02-14:
This issue is related to this one: https://github.com/synthetos/g2/issues/145

timothy-godfrey, 2018-02-17:
I thought I write what I ended up deciding, in case it is useful to anyone else.

I had a look at using the GRBL pins, but it looked like it was going to be more work than just configuring the pin outs and the settings. Because there were spare digital inputs already available I figured it would be simpler to use those. After a quick trial I got a response from pin 10, but not pins 11 and 12 (could be a board problem?). Also, as described in #313, digital inputs 10 - 12 don't show up in $in reports, and I didn't get to the bottom of that after a quick look, so I preferred pins <= 9. Because my machine is only a 3 axis machine I have no A axis, so I decided to reconfigure the default A axis limit switches for my inputs. All of this meant that in the end of I have my interlock signal connected to pin 7, the emergency stop signal connected to pin 8, and cooling alarm connected to pin 9.

AVRsteffen, 2018-02-24:
How you configurate this Pins?

closed 2018-04-01 by timothy-godfrey