synthetos/g2 #316: Compiling on Texas Instruments development board

Brac24, 2018-02-22:
Would I be able to compile this for my tm4c mcu on Keil? I see there is support for different boards in the motate project. Would I be able to get it to work fairly easily or would it be too much of a hassle?

justinclift, 2018-04-08:
Is there a link for further info on the processor, and/or dev board you're using it on?

Brac24, 2018-04-09:
Hi there @justinclift  thank you for your help. Sorry about that I probably should have put something.
[http://www.ti.com/tool/ek-tm4c123gxl](url) this should have a lot of info on the dev board

justinclift, 2018-04-10:
Thanks @Brac24.  Those look interesting.  Fairly cheap, and seems like they do CAN, which is on my personal ToDo list to get some hands on learning done with.  I haven't touched the motate stuff yet myself either... but that's also on my personal ToDo list.  I'll probably look at it for the RISC-V side of things though, as that platform (eg HiFive1) appeals to me more. :smile: