synthetos/g2 #319: fusion 360 post config for g2?

scotthz, 2018-03-03:
Wondering if anyone has a Fusion 360 post config for g2. It comes with a tinyg post, but it doesn't support some things that may be possible with g2. For example, it doesn't allow helical moves.

I'm no expert with Fusion 360 or CAD/CAM in general, just wondering if anyone else has thought about a postprocessor for g2.

giseburt, 2018-03-03:
This is what I've been using, and is the default setting for the TinyG post:
(You can ignore everything above "Properties".)
[image]

giseburt, 2018-03-03:
I also recommend enabling smoothing on any operation that allows it, and set it to what's acceptable for your job. For example:
[image]

scotthz, 2018-03-05:
Thanks. Allow Helical Moves isn't enabled by default in the TInyG post in my environment, good to know it should work.

closed 2018-03-05 by scotthz