synthetos/g2 #322: Please create a mailing list / IRC / something for g2core

justinclift, 2018-03-20:
Hiyas,

It's incredibly difficult for new users and people starting out with g2core to communicate with each other to get going.  The Syntheos TinyG forum doesn't meet that need, nor is GitHub issues suited for this purpose.

Is it feasible to create a mailing list (and maybe an IRC channel) for users to communiate with each other?  That'd significantly help new users.

hurr1k4ne, 2018-03-20:
https://plus.google.com/communities/103099267575816902412

Have a go Here.

Greetings

justinclift, 2018-03-20:
Thanks, but that's a G+ / social-media thing.  Not exactly a mailing list, IRC channel, or similar. :smile:

eg For people not into social media, that's not going to work.

hurr1k4ne, 2018-03-20:
I use the g+ group as a Mailinglist. Because i am Not into social Media.

 i got your Point. But thats working for today. Other Channels need a
decision of the official representatives which need some sort of manpower
to maintain.

justinclift, 2018-03-20:
No worries. :smile:

giseburt, 2018-03-20:
Hi guys,

We have been thinking of doing a Synthetos public Slack. Does that sound
like it would work in the capacity you’re thinking?

It would be exciting to be able to talk more one-on-one with those that are
interested in g2core.

-Rob

justinclift, 2018-03-20:
Well... personally I've avoided Slack because it seems to need people to create a brand new account for everything, and the client has a terrible reputation as being a resource hog (eg 1.5GB resident ram usage).

But, if that's the only option then sure I could be convinced.  It's at least real-time (like IRC), so would help newbies get over the initial teething bits. :wink:

justinclift, 2018-04-12:
Going to close this for now, as it's not something that has a clear answer for implementing for away.

We'll probably revisit this later on anyway, after the Community grows in size again. :smile:

closed 2018-04-12 by justinclift