synthetos/g2 #323: Text mode switch not working?

lutorm, 2018-03-22:
I just updated my two-year old edge (89.03 with my added tool length offsets etc) to current edge, and now I don't get any feedback when running in chilipeppr any more. When connected directly with a terminal and type "$", I get this:

```
Terminal ready
{"r":{"fv":0.99,"fb":100.26,"fbs":"100.26-9-g84507","fbc":"settings_minimill.h","hp":3,"hv":0,"id":"0084-d639-18c6-4ad","msg":"SYSTEM READY"},"f":[1,0,1]}
$
{"r":{"sys":{"fb":100.26,"fbs":"100.26-9-g84507","fbc":"settings_minimill.h","fv":0.99,"hp":3,"hv":0,"id":"0084-d639-18c6-4ad","jt":0.75,"ct":0.001,"sl":1,"lim":0,"saf":0,"m48e":1,"mfoe":0,"mfo":1,"mtoe":0,"mto":1,"mt":5,"spep":1,"spdp":0,"spph":1,"spdw":2,"ssoe":0,"sso":1,"cofp":1,"comp":1,"coph":1,"tv":1,"ej":1,"jv":2,"qv":0,"sv":1,"si":250,"gpl":0,"gun":1,"gco":1,"gpa":2,"gdi":0}},"f":[1,0,2]}

```
So it seems to be echoing back the settings in JSON mode rather than text mode. With my old build, I get this:
```
Terminal ready
{"r":{"fv":0.98,"fb":89.03,"hp":3,"hv":0,"id":"0213-0215-d443","msg":"SYSTEM READY"},"f":[1,0,1]}
tinyg [mm] ok> 
$
[fb]  firmware build             89.03
[fbs] firmware build "                           build"
[fv]  firmware version            0.98
[cv]  configuration version       7.00
[hp]  hardware platform           3.00
[hv]  hardware version            0.00
[id]  TinyG ID       0213-0215-d443
```
In both cases, COMM_MODE is JSON_MODE. I thought it was supposed to echo back text mode queries in text mode. Has this behavior changed or is something not right with my new build? I also note that it used to provide a prompt, which the new build doesn't. 

If it doesn't respond in kind any more it's hard to work with the settings from Chilipeppr since it filters JSON responses but relies on JSON to work.

hurr1k4ne, 2018-03-22:
Please try $ej=2 for Auto Mode. I use it with a equal scenario.

lutorm, 2018-03-23:
Ah, excellent. Thanks. This option is not mentioned under COMM_MODE in settings_default.h.

ril3y, 2018-03-23:
Thanks for the help on this hurr1k4ne

closed 2018-04-11 by justinclift