synthetos/g2 #324: Auto tool change request

nesquik011, 2018-03-28:
Hi , is there option to use auto tool changor spindle ? And how to connect the axis with arduino is there some text i edit so i can set pins ?

justinclift, 2018-04-11:
Thinking about this a bit, it would probably depend on the autochanger.  What model is yours, and how is it normally driven?

nesquik011, 2018-04-12:
Check this , nsk
http://www.artcotools.com/nsk-nakanishi-nr40-5100-atc-spindle-50-000rpm.html

justinclift, 2018-04-12:
Ok, that's the spindle part of an autochanger.  What autochanger do you use?

As a thought... you seem to be asking lots of questions but aren't very specific on your machine.

To make this simpler, what kind of CNC machine do you have, or are you wanting to build your own? :smile:

justinclift, 2018-04-12:
Cool.  With the controller for it, what kind of wiring does it use and what signals does it need to receive?

nesquik011, 2018-04-12:
i dont know there is no much info about it , you need to contact them in Japan

justinclift, 2018-04-12:
Probably not, in this specific instance.  If I had the exact hardware here to work with myself, then I probably could.  Want to buy two ATC's and send one to me? :smile:

nesquik011, 2018-04-12:
lol i wish , i hardly trying to save money for mine hhhhhhh

justinclift, 2018-04-12:
No worries.  I'll close this issue for now then, and later on if uou have the hardware working we can re-open this to investigate.  :smile:

closed 2018-04-12 by justinclift