synthetos/g2 #325: g2core only works on chilipeppr and doesnt save config

goboldgraphics, 2018-03-31:
g2core only works on chilipeppr and doesnt save config. It does not work with any other g code sender including coolterm, UGS or any.

hurr1k4ne, 2018-04-05:
Hey,

If you dig through the Wiki, you can find information that g2core can not
save config changes persistent to eeprom at the Moment. This is why
chilipeppr has the ability to send config changes (saved in chilipeppr) on
each connect of the Controller.

To your gcode sender Problem:
I sometimes use standard Linux or Windows serial console programs to send
gcode commands to g2core (Like # or ## or config changes #ej=2). This might
be a configuration Problem.

justinclift, 2018-04-05:
As a data point, g2core definitely works with CoolTerm.

At least, the "master" branch from git does.  Saying that as I've been using CoolTerm with g2core pretty much every day for the last few weeks.

If CoolTerm isn't working for you, then there's probably a configuration problem needing to be fixed as @hurr1k4ne mentioned.

justinclift, 2018-04-12:
@goboldgraphics Did you get manual communication - for example through CoolTerm - working?

aldenhart, 2018-05-14:
Closing this for now.

closed 2018-05-14 by aldenhart