synthetos/g2 #326: UART instead of USB (Arduino DUE)

rkoe, 2018-03-31:
Hi,
is it possible to use g2core on an Arduino DUE and communicate with it with UART instead of USB?

I've already modified gShield-pinout.h: I've disabled USB, enabled UART and set/shifted some pins (for details: see below). But when I try to compile I get the following error-messages:

```
../Motate/MotateProject/motate/Atmel_sam_common/SamUART.h:334:20: error: cannot call member function 'constexpr const Motate::DMA<Usart*, periph_num>* Motate::_USARTHardware<uartPeripheralNumber>::dma() [with unsigned char uartPeripheralNumber = 1u]' without object
             if (dma()->inTxBufferEmptyInterrupt())
                    ^
../Motate/MotateProject/motate/Atmel_sam_common/SamUART.h:342:20: error: cannot call member function 'constexpr const Motate::DMA<Usart*, periph_num>* Motate::_USARTHardware<uartPeripheralNumber>::dma() [with unsigned char uartPeripheralNumber = 1u]' without object
             if (dma()->inRxBufferFullInterrupt())
                    ^
```

How can this be fixed?
I would appreciate any hint.

My gShield-pinout.h modifications:
```
#define XIO_HAS_USB 0
#define XIO_HAS_UART 1
pin_number kSerial_RXPinNumber = 17;
pin_number kSerial_TXPinNumber = 16;
pin_number kSerial_RTSPinNumber =  23;
pin_number kSerial_CTSPinNumber =  24;

pin_number kSerial0_RX = 17;
pin_number kSerial0_TX = 16;
pin_number kSerial0_RTS =  23;
pin_number kSerial0_CTS =  24;

pin_number kSocket1_Microstep_0PinNumber    = -1;
pin_number kSocket1_Microstep_1PinNumber    = -1;
pin_number kInput4_PinNumber = 51;
pin_number kInput5_PinNumber = 65;
pin_number kInput11_PinNumber = -1;
pin_number kInput12_PinNumber = -1;
```

giseburt, 2018-03-31:
Which branch are you using for this? I can’t test this this weekend but you
could try this against edge or dev-168-gquintic branches. Note that you
will likely need to run `git submodule update` after switching branches
unless you are using a GUI git client like Sourcetree that does that
automatically.

rkoe, 2018-03-31:
I've used the stable branch (100.26).
I've now also tried the g2core-edge-branch, but I got the same error.
(I've also tried to switch Motate to the edge branch, but this results in a lot of errors, since e.g. the number of parameters for _MAKE_MOTATE_PWM_PIN seems to have changed.)

rkoe, 2018-04-09:
Any news or more hints about this?
I would like to make this work, and I hope that I don't have to read the whole code, so any hint would be appreciated.

Is it "only" a bug, and how can it be fixed, or where can I read more? Or is the non-USB-UART-communication not supported at all, and so I would have to (nearly) completely implement it?

giseburt, 2018-04-09:
It looks correct afaict. I can’t test this configuration now but it looks
like it’s not auto detecting the DMA correctly.

I believe that chip uses PDC DMA for the USART. If you put a #warning
pragma here:
https://github.com/synthetos/Motate/blob/master/MotateProject/motate/Atmel_sam_common/SamDMA.h

You should see the message you put after #warning when you compile. If you
don’t, then it’s not correctly auto detecting it. That is configured here:
https://github.com/synthetos/Motate/blob/master/MotateProject/motate/Atmel_sam_common/SamCommon.h

(It only checks for 0 and assumes it will have 1 as well. It doesn’t appear
to support 2, that is another bug.)

Hopefully this helps if you can get to it before I can.

-Rob

rkoe, 2018-04-13:
I tried some more, but without success.

The DUE-ARM has 5 uart/usart-interfaces:
- UART: ARM-pins URXD/UTXD, DUE-pins RX/TX, no RTS/CTS; used for programming the ARM
- USART0..3: ARM-pins RXD0..3, TXD0..3, RTS0..3, CTS0..3

Results:
- UART: When I select the UART (URXD/UTXD, DUE-pins RX/TX, DUE-pinnumbers 0/1), and select some other pins for RTS/CTS, compiling works.
  BUT: I don't think this would work correctly, since
  - this uart is used for programming the ARM (via the AtMega)
  - this uart does not have RTS/CTS pins, but uart-communication requires hardware-handshake via RTS/CTS according to the documentation
  - setting some other pins as RTS/CTS probably won't work

- USART: When I select USART0 (RXD0/TXD0/RTS0/CTS0, DUE-pins 19, 18, 2, 20) or USART1 (RXD1/TXD1/RTS0/CTS0, DUE-pins 17, 16, 23, 24), compiling fails with the aboce "dma-error".

FranciscoPoloChavez, 2020-04-08:
Did you got the solution rkoe?

jam-sys, 2020-05-07:
I have exactly the same error.
Is there a solution for this problem?